realsenseai/librealsense #7338: Potentail Bug in the Python Wrapper Generating Access Violation Error

amitelle1337, 2020-09-14:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D4435i & L515 | 
| Firmware Version                   | 05.12.06.00 & 01.04.01.02 respectivly | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  v2.36.0 }                          | 
| Language                            |  python }                          | 

### Issue Description
I found a potential bug in the python wrapper.
I've created a simple code to demonstrate it and recreate the problem:
```python
import numpy as np
import pyrealsense2 as rs2

def capture_frame() -> rs2.depth_frame:
    pipe: rs2.pipeline = rs2.pipeline()

    config: rs2 = rs2.config()
    config.enable_stream(rs2.stream.depth)

    pipe.start(config)
    try:
        for _ in range(30):
            pipe.wait_for_frames()

        frame: rs2.composite_frame = pipe.wait_for_frames()
        return frame.get_depth_frame().as_depth_frame()
    finally:
        pipe.stop()

def capture_to_pointcloud() -> np.ndarray:
    frame = capture_frame()
    thresh: rs2.threshold_filter = rs2.threshold_filter(max_dist=0.9)
    frame = thresh.process(frame)
    pc: rs2.pointcloud = rs2.pointcloud()
    points: np.ndarray = np.array(pc.calculate(frame).get_vertices())
    points = np.array([(x, y, z) for (x, y, z) in points if not x == y == z == 0])
    return points

if __name__ == '__main__':
    points = capture_to_pointcloud()
    print(points)
```

Running this code will generate the following error: `Process finished with exit code -1073741819 (0xC0000005)`.
This error number represents an access violation error (as much as I know).
Running in debug mode reveals that the code crushes in line 27:
'''python
points: np.ndarray = np.array(pc.calculate(frame).get_vertices())
'''
To be more exact, the crushing code is:
'''python
pc.calculate(frame)
'''
After trying to fix the code I was able to fix it by moving the creation of the config object outside the capturing function:
```python
def capture_frame(config: rs2.config) -> rs2.depth_frame:
   # Same code

   # This is no longer necessary
   # config: rs2 = rs2.config()
   # config.enable_stream(rs2.stream.depth)

def capture_to_pointcloud() -> np.ndarray:
   config = rs2.config()
   config.enable_stream(rs2.stream.depth)
   frame = capture_frame(config)
   # Same code
```
Now the code will work correctly.
My speculation is that because the python garbage collector is based on reference counting, the config object is being deleted while another real-sense object is pointing to it (Or some data inside it) and thus an access violation error is created, as the error number suggests.
To be more exact: the problem (I assume) is the wrapper not accounting for native pointers because I am thinking some native code in another real-sense object (probably the point-cloud object) is using native data in the config object but the python GC deleted it already.
I hope this helps,
If I said anything wrong please correct me.

RealSenseSupport, 2020-09-27:
@amitelle1337 Sorry that I couldn't reproduce the issue with your code. Does this issue happen for every execution or just sometimes? Looking forward to your reply. Thanks!

amitelle1337, 2020-09-27:
I've done a couple more tests, just to make sure, and it seems the I can reproduce the issue only when running it from pycharm; Running it from the terminal is fine.
I guess the problem is with pycharm and not the library itself.
Also, I've updated everything on my machine, so the new specs would be:
|      Specs                   |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.00.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.38.1.2223                         | 
| Language                            |  python (numpy v1.19.2, pyrealsense2 v2.38.1.2225, pycharm v2020.2.2)                          |

Yet the problem stays but only when running it from pycharm.
I guess I misinterpreted the issue and it's a problem with pycharm, not sure why though.
Just to be clear, I am not getting a python exception, I am only getting the following message: `Process finished with exit code -1073741819 (0xC0000005)`, as I said in the issue description.
This means (I think) Pycharm/OS killed the process because of access violation.

RealSenseSupport, 2020-09-27:
@amitelle1337 Thank you for your update! So I understood it's not librealsense issue. Can we close this? Thanks!

amitelle1337, 2020-09-27:
Yes, I think we can close this one. 😄

RealSenseSupport, 2020-09-28:
@amitelle1337 Thanks for your confirmation! Close it accordingly.

closed 2020-09-28 by RealSenseSupport