realsenseai/librealsense #7340: Green Dead pixels in right corner of depth steam of D435 camera

Ashutosh-Badave, 2020-09-15:
Hi all,
System :
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|013222073966|
|**Firmware Version**|05.12.07.100|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|009223021035|
|**Firmware Update Id**|009223021035|

I have getting strange 3D depth stream from D435 camera after connected with realsense viewer as shown below:
[image]

I searched for this issue and I found it might be mis-calibration because of physical shock or if  camera dropped. As mentioned in [here](https://support.intelrealsense.com/hc/en-us/community/posts/360048022393-Dead-green-pixels-in-stream-) 
So, I did the intrinsic, extrinsic calibration from viwer as well as calibration from dynamic calibrator v2.11.0.0 but still issue remained.
Please let me know what I need to do in this case.

Kind regards,
Ashutosh Badave

MartyG-RealSense, 2020-09-15:
Hi @Ashutosh-Badave   There have been a couple of cases of this green corner effect before, though to a lesser extent than the prominent green area that you are experiencing.  The case that you linked to was one of them.  In the other case, a solution that corrected the problem was to install librealsense via the **RSUSB backend** method.

https://github.com/IntelRealSense/librealsense/issues/6845

Ashutosh-Badave, 2020-09-15:
Hi @MartyG-RealSense , Thank you for providing a quick response. I started building the code as mentioned above on my Linux system but the more interesting thing is the camera is working perfectly fine with JetsonTx2 board without any dead pixels and I didn't get the reason.

MartyG-RealSense, 2020-09-15:
If you built librealsense on TX2 from packages (like in the link below) then I would speculate that this is what protects it against the problem, and that it is an issue related to the kernel when building from source (since the RSUSB installation method does not require patching).  That is theory though, not certainty.

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

Ashutosh-Badave, 2020-09-15:
Hi @MartyG-RealSense , Thanks for sharing the information, and yes for Tx2, I used the above method.
About my issue, I have downloaded, extracted, and built the librealsense-2.38.1. Now how should I launch realsense viewer so that these files will be used and not previously installed? and how should I use them in my application?

MartyG-RealSense, 2020-09-15:
Hi @Ashutosh-Badave   If you use the command ** realsense-viewer** to launch the Viewer (or find it in a file search by searching for the name ** realsense-viewer**), does the Viewer have the title **Intel RealSense Viewer v2.38.1** at the top of its window, please?

Ashutosh-Badave, 2020-09-15:
Yes. It is showing with same  error. 
[image]

MartyG-RealSense, 2020-09-15:
And your first installation (the one where you first encountered the green pixels) was built from source code using the patching-based installation method?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Ashutosh-Badave, 2020-09-15:
I have installed as per this https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

I would like to point out that I am using realsense V2.35.2. I don't know this information is relevant or not.

MartyG-RealSense, 2020-09-15:
I think what you are saying is that on the computer that you have the problem on, you installed with the package distribution method and you also installed from packages on TX2, so you do not know why the TX2 build works fine and the other has the problem with the green box.  Is that correct, please?

Ashutosh-Badave, 2020-09-15:
Yes. After doing the backend method then also it has the same issue.

MartyG-RealSense, 2020-09-15:
A common factor in the other two cases of a green area in the corner are that the green appears in the bottom corner of the RGB image.  Does the green appear in the bottom corner of your RGB image too, please?

Ashutosh-Badave, 2020-09-15:
Hi @MartyG-RealSense,  it is resolved I have uninstalled the realsense2 package given as per this https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md.

As I already asked how should I launch the realsense after installing using backend method. 
Before uninstall librealsense2 package, when I am launching realsense viewer directly from command line, it is using that package. But now to launch realsense viewer, I used below command,

`librealsense-2.38.1/build$ ./tools/realsense-viewer/realsense-viewer` 
And the issue is resolved.

But it shows ´librealsense2-udev-rules´ are missing. and also my application could not find realsense cmake.

MartyG-RealSense, 2020-09-15:
I'm very pleased to hear that you made progress.

If you have missing udev rules, please try inputting the command below:

**sudo apt-get upgrade/install librealsense2-udev-rules**

Ashutosh-Badave, 2020-09-15:
Thank you so much @MartyG-RealSense for your support. Everything is working correctly.

closed 2020-09-15 by Ashutosh-Badave

MartyG-RealSense, 2020-09-15:
Great news - good luck with the camera!