realsenseai/librealsense #7341: Raspi3 protobuf python3 build fail error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

prem48science, 2020-09-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentation](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open a new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Raspbian or Raspberry pi OS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description

google/protobuf/pyext/descriptor_containers.cc: In function ‘bool google::protobuf::python::descriptor::_GetItemByKey(google::protobuf::python::PyContainer*, PyObject*, const void**)’:
google/protobuf/pyext/descriptor_containers.cc:69:45: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
        ((*(charpp) = PyUnicode_AsUTF8AndSize(ob, (sizep))) == NULL? -1: 0): \
                      ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
google/protobuf/pyext/descriptor_containers.cc:172:13: note: in expansion of macro ‘PyString_AsStringAndSize’
         if (PyString_AsStringAndSize(key, &name, &name_size) < 0) {
             ^~~~~~~~~~~~~~~~~~~~~~~~
google/protobuf/pyext/descriptor_containers.cc:69:45: error: invalid conversion from ‘const char*’ to ‘char*’ [-fpermissive]
        ((*(charpp) = PyUnicode_AsUTF8AndSize(ob, (sizep))) == NULL? -1: 0): \
                      ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~
google/protobuf/pyext/descriptor_containers.cc:189:13: note: in expansion of macro ‘PyString_AsStringAndSize’
         if (PyString_AsStringAndSize(key, &camelcase_name, &name_size) < 0) {
             ^~~~~~~~~~~~~~~~~~~~~~~~
error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1

Hi guys, I am following documentation as mentioned in [docs](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md) I am not able to move forward as the above mentioned error is recurring no matter what I tried to fix it. 

this error can be reproduced while installing protobuf in raspi

MartyG-RealSense, 2020-09-15:
Hi @prem48science   I see that you linked today to another case about this error that contains advice for fixing problems, and also contributed a pull.

https://github.com/IntelRealSense/librealsense/issues/5092
https://github.com/IntelRealSense/librealsense/pull/7339

May I ask whether you have solved the problem and are highlighting changes to the official Raspberry Pi instructions that you believe are necessary?  Or do you still have the problem and require further assistance with it?   Thanks!

prem48science, 2020-09-16:
> Hi @prem48science I see that you linked today to another case about this error that contains advice for fixing problems, and also contributed a pull.
>
> #5092
> #7339
>
> May I ask whether you have solved [...]

yes, I am able to solve some of the problems but not complete ones. since raspi foundation has changed its OS from stretch to raspberry pi os and some parts of it are currently under development I suspect some errors may arise because of that. I am working to solve this "error: command 'arm-linux-gnueabihf-gcc' failed with exit status 1" error so far no luck. so If you have encountered it  please guide me to the solution.

MartyG-RealSense, 2020-09-16:
The only reference on the librealsense GitHub about this error that I could find was the one that you had already linked to. 
 https://github.com/IntelRealSense/librealsense/issues/5092

Expanding the scope of my research out to Raspberry Pi in general though, I located the links below about this error:

https://github.com/jgarff/rpi_ws281x/issues/294
https://www.raspberrypi.org/forums/viewtopic.php?t=211194
https://github.com/JHUISI/charm/issues/207
https://stackoverflow.com/questions/61667166/pillow-installation-fails-command-arm-linux-gnueabihf-gcc-failed-with-exit-st

All of these cases have a connection to Python.

prem48science, 2020-09-17:
@MartyG-RealSense , thanks for the references. 
I looked up and [tried these solutions in thread](https://stackoverflow.com/questions/61667166/pillow-installation-fails-command-arm-linux-gnueabihf-gcc-failed-with-exit-st )
```
sudo apt-get install portaudio19-dev
sudo apt install -y python python-pip libffi-dev python-backports.ssl-match-hostname
sudo pip install rpi_ws281x
sudo apt install -y libffi-dev
sudo apt-get install automake libtool vim cmake libusb-1.0-0-dev libx11-dev xorg-dev libglu1-mesa-dev
```

on fresh installation of raspberry pi os on raspi 4. and I also tried new version of protobuf i.e

`git clone --depth=1 -b v3.13.0 https://github.com/google/protobuf.git 
`
instead of git clone --depth=1 -b v3.5.1 https://github.com/google/protobuf.git

and rest of the commands I followed as specified. the error got eliminated.

closed 2020-09-17 by prem48science

MartyG-RealSense, 2020-09-17:
Thanks very much for the update and the sharing of the technical details!