realsenseai/librealsense #7343: GLFW Driver Error: GLX: GLX version 1.3 is required

prem48science, 2020-09-15:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentation](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   { Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi.  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hi, I tried to build librealsense and was successful but while I am trying to read the camera feed. I ran realsense-viewer and it is returning this error GLFW Driver Error: GLX: GLX version 1.3 is required Could not initialize offscreen context!

please suggest a solution. tried different methods but unsuccessful.

MartyG-RealSense, 2020-09-15:
Hi @prem48science   Could you try updating the graphics drivers on your Pi, please to ensure that the OpenGL drivers are up to date?

prem48science, 2020-09-16:
> Hi @prem48science If you have not done so already could you try updating the graphics drivers on your Pi please to ensure that the OpenGL drivers are up to date?

I am not aware of updating OpenGL drivers and didn't find any documentation for the same. please suggest me if any documentation exists for doing so. I am using sudo apt-get for update and upgrade but no luck!

prem48science, 2020-09-16:
@MartyG-RealSense  
I want to close this issue as I found source of error. 
the problem is in the RealVNC viewer x64 EXE for windows 10 machine. 
I tried connecting my pi to a monitor with an HDMI port and to my surprise, the command realsense-viewer started working without any problems.  
after a little bit of research, I found that the GLX driver is a virtual drive on the pi machine shipped by VMware. so it depends on the type of display that it is performing graphics for ~_~. (probably not so good explanation) 

anyways it worked when I ran it on the original display. so no worries. 
[image]

closed 2020-09-16 by prem48science

MartyG-RealSense, 2020-09-16:
Great - thanks so much for the update and technical explanation.   :)