realsenseai/librealsense #7345: How to use the SDK to get a list of distances?

benntowner, 2020-09-15:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 

### Issue Description
I assume the camera is pointing vertically downwards, parallel to the ground. I want to get the X-Y-Z matrix in mm/cm. Which SDK methods do I need to use? 

Example (the values ​​in the table are the distance along the Z axis.)
||x1|x2|x3|x4|x5|x6|x7|x8|x9|x10|
|-|-|-|-|-|-|-|-|-|-|-|
|y1|0|0|0|0|0|0|0|0|0|0|
|y2|0|0|0|0|0|0|0|0|0|0|
|y3|0|0|0|0|0|0|0|0|0|0|
|y4|0|0|0|1|1|1|1|1|0|0|
|y5|0|0|0|1|2|2|2|1|0|0|
|y6|0|0|0|1|2|3|2|1|0|0|
|y7|0|0|0|1|2|2|2|1|0|0|
|y8|0|0|0|1|1|1|1|1|0|0|
|y9|0|0|0|0|0|0|0|0|0|0|
|y10|0|0|0|0|0|0|0|0|0|0|

I want to see a list of the distance to the ground (Z axis) such a table as above in the example.
How to calculate the real distance in the XY plane (between x4, y4 and x8, y8) in mm/cm?

RealSenseSupport, 2020-09-18:
Hi @benntowner 

We have quite a lot of examples as part of the RealSense SDK.  Couple items I want to point out:

- L515 Users Guide:  https://www.intelrealsense.com/download/13061/
- Optimize for Range:  https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/
- SDK Examples:  https://github.com/IntelRealSense/librealsense/tree/master/examples, https://github.com/IntelRealSense/librealsense/blob/master/examples/measure, also the RealSense Viewer has a measure feature

The available DIM Weight SW may also be of assistance:  https://www.intelrealsense.com/dimensional-weight-software/

RealSenseSupport, 2020-09-30:
Hi @benntowner 

Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-10-07 by RealSenseSupport