realsenseai/librealsense #7348: Increase max distance with threshold_filter?

s-trinh, 2020-09-15:
It looks like the `max_dist` for [rs2::threshold_filter](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1threshold__filter.html) is [16](https://github.com/IntelRealSense/librealsense/blob/0adceb9dc6fce63c348346e1aef1b63c052a1db9/src/proc/threshold.cpp#L24).

Would it make sense to allow bigger value? By default with 16bit depth and 1000µm depth unit, the max range is ~65m, see [this guide](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance).

MartyG-RealSense, 2020-09-16:
Hi @s-trinh  Although in theory you can set a much longer range, in practice depth measuring error will increase linearly over distance (a phenomenon called **RMS Error**).  So you likely would not get any depth data beyond 10 meters depth sensing range that is accurate enough to be useful for depth measurement, even if the camera could pick up detail at that range.  And the **Threshold Filter** in the RealSense Viewer enforces a maximum distance of 16 meters when inputting a manual value greater than 16.

s-trinh, 2020-09-20:
I agree that RMS error increases quadratically with distance.

In my opinion, there should not be hard-coded max distance limit for the threshold_filter class. But instead let the developer choose this value for whatever he wants to do with the sensor / use case (for instance, I am not interested with (accurate) measurement applications).

> And the Threshold Filter in the RealSense Viewer enforces a maximum distance of 16 meters when inputting a manual value greater than 16.

To be clear, the RealSense viewer is a great tool. They can keep the 16 m limit if they want. But I am talking about the C++ API, not the RealSense viewer.

Closing since the [ implemented code](https://github.com/IntelRealSense/librealsense/blob/0adceb9dc6fce63c348346e1aef1b63c052a1db9/src/proc/threshold.cpp#L56-L60) is trivial and thus can be easily reimplemented by the developer.

closed 2020-09-20 by s-trinh

MartyG-RealSense, 2020-09-20:
Thanks very much for the detailed update @s-trinh