realsenseai/librealsense #7349: Accessing device after advanced mode toggle.

Chandler, 2020-09-16:
Hello, I am trying to reliably set advanced mode on a d415. I am doing it like this

```
        dev = profile.get_device()
        depth_sensor = dev.first_depth_sensor()

        advnc_mode = rs.rs400_advanced_mode(dev)
        while not advnc_mode.is_enabled():
            print("Trying to enable advanced mode...")
            advnc_mode.toggle_advanced_mode(True)
            # At this point the device will disconnect and re-connect.
            print("Sleeping for 5 seconds...")
            time.sleep(5)
            # The 'dev' object will become invalid and we need to initialize it again
            breakpoint()
            dev = profile.get_device()
            advnc_mode = rs.rs400_advanced_mode(dev)
            print("Advanced mode is", "enabled" if advnc_mode.is_enabled() else "disabled")
```

However, it seems like `profile.get_device()` does not work, since I get these errors on the second loop.

```
RuntimeError("Cannot open '/dev/video4 Last Error: No such file or directory")
RuntimeError('profile not found')
```

I have seen the device is accessed in the advanced example, however I cannot make sense of it.
https://github.com/IntelRealSense/librealsense/blob/0adceb9dc6fce63c348346e1aef1b63c052a1db9/wrappers/python/examples/python-rs400-advanced-mode-example.py#L15

```
def find_device_that_supports_advanced_mode() :
    ctx = rs.context()
    ds5_dev = rs.device()
    devices = ctx.query_devices();
    for dev in devices:
        if dev.supports(rs.camera_info.product_id) and str(dev.get_info(rs.camera_info.product_id)) in DS5_product_ids:
            if dev.supports(rs.camera_info.name):
                print("Found device that supports advanced mode:", dev.get_info(rs.camera_info.name))
            return dev
    raise Exception("No device that supports advanced mode was found")
```

This seems to search a global list of all devices, **is there a way to get the new device instance for the specific profile that I started with?** I may have multiple devices so this search that returns the first advanced mode device doesn't seem like it would work for me.

Chandler, 2020-09-16:
Should I loop through all the devices and check for the serial number that matches the original? to be sure that I have the same device I started with, after toggling advanced mode.

MartyG-RealSense, 2020-09-16:
Hi @Chandler   There is an excellent and very long discussion on using **ctx.query_devices()** to query multiple devices that ends with the script in the link below.

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412899468

The entire discussion is well worth reading from the start though to understand the mechanics of how multi-device querying works and how to apply them to the requirements of your particular project.

MartyG-RealSense, 2020-09-26:
Hi @Chandler   Do you require further assistance with this case, please?  Thanks!

Chandler, 2020-10-01:
I found a solution, thanks @MartyG-RealSense. For future readers here's what I did, after each `advnc_mode.toggle_advanced_mode(True)` I requery the device by looking at all devices and finding the one with the correct serial number.

This is run in a loop in case it takes a while for the device to reappear.

```
def query_device_by_serial_number(desired_serial_number):
	for device in rs.context().devices:
		serial_number = str(device.get_info(rs.camera_info.serial_number))
		if serial_number == desired_serial_number:
			return device
```

closed 2020-10-01 by Chandler

MartyG-RealSense, 2020-10-01:
Great news, thanks so much for sharing your solution!