realsenseai/librealsense #7350: Export functionality MacOS

seanngpack, 2020-09-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-09-16:
Hi @seanngpack   The json export button should be on the toolbar at the top of the Viewer's options side-panel.

[image]

seanngpack, 2020-09-16:
Hi @MartyG-RealSense, thanks for the response! My viewers (v.2.38.1) on MacOS and windows definitely don't have that button

[image]

MartyG-RealSense, 2020-09-16:
Thanks very much for the update that you are using an SR305.  I can confirm that the json export option is not available in the Viewer for SR305 but is for the 400 Series cameras.  The SR305 model does not have an **Advanced Mode** and therefore does not have the Visual Presets (also known as json) that the 400 Series cameras support.

seanngpack, 2020-09-16:
@MartyG-RealSense Thank you for the information! 

This lead to stumble upon this thread: https://github.com/IntelRealSense/realsense-ros/issues/1168

I think I'll try adjusting the settings dynamically using the API

closed 2020-09-16 by seanngpack