realsenseai/librealsense #7351: About convention used in rs-motion example

kuwt, 2020-09-16:
I am using the imu in L515 and I am studying the rs-motion example.  However, I fail to figure out the convention used in the example. 

Firstly, in draw_axes(), the x-axis is drawn from (0,0,0) to (-1,0,0). Why is it not the common convention from (0,0,0) to (1,0,0).
y-axis has similar problem while z-axis follows the common convention.
```
void draw_axes()
{
    glLineWidth(2);
    glBegin(GL_LINES);
    // Draw x, y, z axes
    glColor3f(1, 0, 0); glVertex3f(0, 0, 0);  glVertex3f(-1, 0, 0);
    glColor3f(0, 1, 0); glVertex3f(0, 0, 0);  glVertex3f(0, -1, 0);
    glColor3f(0, 0, 1); glVertex3f(0, 0, 0);  glVertex3f(0, 0, 1);
    glEnd();

    glLineWidth(1);
}
```

Secondly, in the function render_camera(float3 theta) from class camera_renderer, the rotation direction are in different form for different axes. For theta.x, it is rotated by the z axis. For theta.z, there is an extra -pi/2. What are the reasons behind these implementation?
```
glRotatef(theta.x * 180 / PI, 0, 0, -1);
glRotatef(theta.y * 180 / PI, 0, -1, 0);
glRotatef((theta.z - PI / 2) * 180 / PI, -1, 0, 0);
```

Thirdly, in function process_gyro(), the gyro_angles are added to theta in the order of z y x. What is the reason behind?
```
theta.add(-gyro_angle.z, -gyro_angle.y, gyro_angle.x);
``` 

Lastly, in function process_accel(), how is the acceleration angle derived?  
```
accel_angle.z = atan2(accel_data.y, accel_data.z);
accel_angle.x = atan2(accel_data.x, sqrt(accel_data.y * accel_data.y + accel_data.z * accel_data.z));
```

RealSenseSupport, 2020-09-30:
Hi @kuwt 

The convention should be the same amongst all our products that have IMU similar to what is outlined in this blog:  https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

We do need to make modifications to the blog there as well as the readme.md of rs-motion to take into consideration the RealSense cameras that include IMUs.  The convention should be the same amongst the cameras.

RealSenseSupport, 2020-11-12:
Hi @kuwt 

Is there anything else on this thread needed?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-12-01 by RealSenseSupport