realsenseai/librealsense #7352: arm64 ubuntu18.04 python3.7 error

siddharthcb, 2020-09-16:


MartyG-RealSense, 2020-09-16:
Hi @siddharthcb   The pyrealsense2 pip packages are for x86 / x64 processors and will not work with Arm processors.  On a computing device that uses Arm, the pyrealsense2 wrapper should be built from source code instead.

siddharthcb, 2020-09-16:
hi @MartyG-RealSense 
thanks for the reply but it is installing with python3.6 on the same arm64 processor.
you meant it is not compatible with python3.7 in arm64 ? or not compatible for any version of python in arm64?

MartyG-RealSense, 2020-09-16:
The packages should not be compatible with Arm processors.  The terms "Arm64" and "AMD64" (an x64 processor) are not the same.

siddharthcb, 2020-09-16:
thanks a lot. i will try building it from source.

siddharthcb, 2020-09-17:
hi @MartyG-RealSense .
i built from source using python 3.6.
suppose say if i have upgraded version of python3.7. How do i change the one that i have built from source?

MartyG-RealSense, 2020-09-17:
As far as I know, you would have to build the pyrealsense2 wrapper from source with CMake again and target it at Python 3.7 using the CMake instruction **-DPYTHON_EXECUTABLE=[full path to the exact python executable]**

siddharthcb, 2020-09-18:
is it `cmake -DPYTHON_EXECUTABLE=[full path to the exact python executable`?

siddharthcb, 2020-09-18:
when i run this `cmake .. -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=true -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/local/lib/python3.7`
i get following error: 
`CMake Error at wrappers/python/third_party/pybind11/tools/FindPythonLibsNew.cmake:95 (message):
  Python config failure:

Call Stack (most recent call first):
  wrappers/python/third_party/pybind11/tools/pybind11Tools.cmake:16 (find_package)
  wrappers/python/third_party/pybind11/CMakeLists.txt:33 (include)

-- Configuring incomplete, errors occurred!
See also "/home/smartagri/librealsense/build/CMakeFiles/CMakeOutput.log".`

MartyG-RealSense, 2020-09-18:
Please try the **usr/bin** path instead of lib:

 - DPYTHON_EXECUTABLE=/usr/bin/python3.7

Variations of the -DPYTHON_EXECUTABLE instruction that you can try are:

-DPYTHON_EXECUTABLE=...
(i.e just 3 full stops)

-DPYTHON_EXECUTABLE=/usr/bin/python3

siddharthcb, 2020-09-23:
thank you. issue solved.

closed 2020-09-23 by siddharthcb