realsenseai/librealsense #7356: Error in using CMake to build Times

hebowu, 2020-09-16:
I used CMake to build "librealsense/wrappers/python". The Times was wrong. 
Error message: 

> CMake Error at CMakeLists.txt: 34 (set_target_properties):

> set_target_properties called with incorrect number of arguments.

> CMake Error at CMakeLists.txt: 77 (if):

> if given arguments:

> "STREQUAL"  "RS2_USE_V4L2_BACKEND"

> Unknown arguments specified.

Can you help me?

MartyG-RealSense, 2020-09-16:
Hi @hebowu   Did you launch the CMake command from the top-level librealsense root directory please?  As listed in point 3 of the Python wrapper's intructions, you should go to the **librealsense** directory and use the command **mkdir build && cd build**.  This creates a sub-folder in your librealsense root directory called **build** and automatically moves you into that new folder.

Once inside the **build** folder, you then input the CMake build command.

https://github.com/IntelRealSense/librealsense/issues/4335

hebowu, 2020-09-18:
Yes, I built it from the librealsense root directory. 
（My friend, please forgive me for not seeing your news in time. Thank you very much.）

MartyG-RealSense, 2020-09-18:
No problem, thank you very much for your update.  Has your error been solved or do you still require assistance please?  Thanks!

closed 2020-09-26 by MartyG-RealSense