realsenseai/librealsense #7359: Frame didn't arrived within 5000 when datetime is changed

kamadhenu87, 2020-09-16:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 |
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Raspbian GNU/Linux 9.6 (stretch)  | 
| Kernel Version (Linux Only)    |  4.14.79-v7+                                         | 
| Platform                                 | Raspberry Pi.  |
| SDK Version                            |  2.12      | 
| Language                            |  C                          | 
| Segment			|  others                           | 

### Issue Description
Frame didn't arrived within 5000 when datetime is changed.
when we start the sensor, eg rs-distance and update the RPI datetime, if datetime is drifted and updated, the program crashes.

kamadhenu87, 2020-09-18:
some one facing same?

RealSenseSupport, 2020-09-18:
for Raspberry Pi, when doing cmake, you need to add the -DFORCE_RSUSB_BACKEND=ON
by the way, the SDK  is pretty old, would suggest you to upgrade SDK and firmware
the latest SDK version is 2.38.1 firmware is 5.12.07.100

kamadhenu87, 2020-09-21:
Its still crashing when date is changed during streaming with the flag compiled.

RealSenseSupport, 2020-09-21:
@kamadhenu87 How do you change the date? Could you please provide an example? And it seems not normal case to change date time during streaming? Could you please elaborate more why you need such operation during streaming? Thanks!

kamadhenu87, 2020-09-21:
We have object detection product(device) processing the received frames, but the device has RTC that can be changed? If the device CMOS battery fails the RTC will goto to wrong date, so when we set RTC(date and time) during streaming, the application stops with exception

kamadhenu87, 2020-09-21:
Same observation with latest SDK librealsense-2.38.1 also on
RPI4 with kernel 5.4.51-v7l+and Raspbian GNU/Linux 10 (buster)

RealSenseSupport, 2020-09-22:
@kamadhenu87 There're error checks in rs-distance such as https://github.com/IntelRealSense/librealsense/blob/master/examples/C/distance/rs-distance.c#L84. When the date changed, error detected and the program crashed.  Is it possible for you to stop the streaming before change the date?  Or you can comment those error checks to get the program running but not suggested as all the errors check also ignored in this case.

closed 2020-09-22 by kamadhenu87