realsenseai/librealsense #7360: CMake error

harshmittal2210, 2020-09-16:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 2.38.1 | 
| Operating System & Version |   Ubuntu 18   | 
| Kernel Version (Linux Only)    |                                         | 
| Platform                                 |          PC             |
| SDK Version                            |                        | 
| Language                            | C++                     | 
| Segment			|          Other                | 

### Issue Description
Can you please share a similar example like example/cmake to compile for OpenGL as well. Right now when I am compiling my code a lot errors are coming. I have also tried by adding realsense2-gl in target_link_libraries but still errors are there.

#### Error
CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o: In function `set_viewport(rect const&)':
rs-pointcloud.cpp:(.text+0x650): undefined reference to `glViewport'
rs-pointcloud.cpp:(.text+0x655): undefined reference to `glLoadIdentity'
rs-pointcloud.cpp:(.text+0x65f): undefined reference to `glMatrixMode'
rs-pointcloud.cpp:(.text+0x6a6): undefined reference to `glOrtho'
CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o: In function `draw_pointcloud(float, float, glfw_state&, rs2::points&)':
rs-pointcloud.cpp:(.text+0x6ef): undefined reference to `glLoadIdentity'
rs-pointcloud.cpp:(.text+0x6f9): undefined reference to `glPushAttrib'
rs-pointcloud.cpp:(.text+0x71e): undefined reference to `glClearColor'
rs-pointcloud.cpp:(.text+0x728): undefined reference to `glClear'
rs-pointcloud.cpp:(.text+0x732): undefined reference to `glMatrixMode'
rs-pointcloud.cpp:(.text+0x737): undefined reference to `glPushMatrix'
rs-pointcloud.cpp:(.text+0x776): undefined reference to `gluPerspective'
rs-pointcloud.cpp:(.text+0x780): undefined reference to `glMatrixMode'
rs-pointcloud.cpp:(.text+0x785): undefined reference to `glPushMatrix'
rs-pointcloud.cpp:(.text+0x7cc): undefined reference to `gluLookAt'
rs-pointcloud.cpp:(.text+0x801): undefined reference to `glTranslatef'
rs-pointcloud.cpp:(.text+0x82b): undefined reference to `glRotated'
rs-pointcloud.cpp:(.text+0x854): undefined reference to `glRotated'
rs-pointcloud.cpp:(.text+0x869): undefined reference to `glTranslatef'
rs-pointcloud.cpp:(.text+0x87f): undefined reference to `glPointSize'
rs-pointcloud.cpp:(.text+0x889): undefined reference to `glEnable'
rs-pointcloud.cpp:(.text+0x893): undefined reference to `glEnable'
rs-pointcloud.cpp:(.text+0x8af): undefined reference to `glBindTexture'
rs-pointcloud.cpp:(.text+0x8f9): undefined reference to `glTexParameterfv'
rs-pointcloud.cpp:(.text+0x90d): undefined reference to `glTexParameteri'
rs-pointcloud.cpp:(.text+0x921): undefined reference to `glTexParameteri'
rs-pointcloud.cpp:(.text+0x92b): undefined reference to `glBegin'
rs-pointcloud.cpp:(.text+0x9d3): undefined reference to `glVertex3fv'
rs-pointcloud.cpp:(.text+0x9f7): undefined reference to `glTexCoord2fv'
rs-pointcloud.cpp:(.text+0xa05): undefined reference to `glEnd'
rs-pointcloud.cpp:(.text+0xa0a): undefined reference to `glPopMatrix'
rs-pointcloud.cpp:(.text+0xa14): undefined reference to `glMatrixMode'
rs-pointcloud.cpp:(.text+0xa19): undefined reference to `glPopMatrix'
rs-pointcloud.cpp:(.text+0xa1e): undefined reference to `glPopAttrib'
CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o: In function `draw_pointcloud_wrt_world(float, float, glfw_state&, rs2::points&, rs2_pose&, float*, std::vector<rs2_vector, std::allocator<rs2_vector> >&)':
rs-pointcloud.cpp:(.text+0xdb7): undefined reference to `glLoadIdentity'
rs-pointcloud.cpp:(.text+0xdc1): undefined reference to `glPushAttrib'
rs-pointcloud.cpp:(.text+0xde6): undefined reference to `glClearColor'
rs-pointcloud.cpp:(.text+0xdf0): undefined reference to `glClear'
rs-pointcloud.cpp:(.text+0xdfa): undefined reference to `glMatrixMode'
rs-pointcloud.cpp:(.text+0xdff): undefined reference to `glPushMatrix'
rs-pointcloud.cpp:(.text+0xe4a): undefined reference to `gluPerspective'
rs-pointcloud.cpp:(.text+0xe54): undefined reference to `glMatrixMode'
rs-pointcloud.cpp:(.text+0xe59): undefined reference to `glPushMatrix'
rs-pointcloud.cpp:(.text+0xe90): undefined reference to `glTranslatef'
rs-pointcloud.cpp:(.text+0xec3): undefined reference to `glRotated'
rs-pointcloud.cpp:(.text+0xef5): undefined reference to `glRotated'
rs-pointcloud.cpp:(.text+0xf0a): undefined reference to `glTranslatef'
rs-pointcloud.cpp:(.text+0xf14): undefined reference to `glEnable'
rs-pointcloud.cpp:(.text+0xf21): undefined reference to `glLineWidth'
rs-pointcloud.cpp:(.text+0xf2b): undefined reference to `glBegin'
rs-pointcloud.cpp:(.text+0xfaa): undefined reference to `glColor3f'
rs-pointcloud.cpp:(.text+0xfdb): undefined reference to `glVertex3f'
rs-pointcloud.cpp:(.text+0xff4): undefined reference to `glEnd'
[... 21 lines cut ...]
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x55): undefined reference to `glGetError'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x11e): undefined reference to `glBindTexture'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x18f): undefined reference to `glTexImage2D'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x1db): undefined reference to `glTexImage2D'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x227): undefined reference to `glTexImage2D'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x270): undefined reference to `glTexImage2D'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x2bf): undefined reference to `glTexParameteri'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x2d3): undefined reference to `glTexParameteri'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x2e7): undefined reference to `glTexParameteri'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x2fb): undefined reference to `glTexParameteri'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x30a): undefined reference to `glPixelStorei'
rs-pointcloud.cpp:(.text._ZN7texture6uploadERKN3rs211video_frameE[_ZN7texture6uploadERKN3rs211video_frameE]+0x319): undefined reference to `glBindTexture'
CMakeFiles/rs-pointcloud.dir/rs-pointcloud.cpp.o: In function `window::operator bool()':
rs-pointcloud.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0xd): undefined reference to `glPopMatrix'
rs-pointcloud.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x76): undefined reference to `glClear'
rs-pointcloud.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x9d): undefined reference to `glViewport'
rs-pointcloud.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0xa2): undefined reference to `glPushMatrix'
rs-pointcloud.cpp:(.text._ZN6windowcvbEv[_ZN6windowcvbEv]+0x117): undefined reference to `glOrtho'
collect2: error: ld returned 1 exit status
CMakeFiles/rs-pointcloud.dir/build.make:95: recipe for target 'rs-pointcloud' failed
make[2]: *** [rs-pointcloud] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/rs-pointcloud.dir/all' failed
make[1]: *** [CMakeFiles/rs-pointcloud.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

MartyG-RealSense, 2020-09-16:
Hi @harshmittal2210   The first thing I would like to check if whether you built librealsense from source code.  If you did, can you confirm whether you performed the installation instruction for the **glfw** graphics packages whilst in the librealsense root directory.  These provide OpenGL support for the example programs that use OpenGL.

For Ubuntu 18, the glfw installation instruction is:

**sudo apt-get install libglfw3-dev libgl1-mesa-dev libglu1-mesa-dev**

Full details of this process can be found in the **Prepare Linux Backend and the Dev. Environment** section of the section of the souce-code installation instuctions linked to below:

https://dev.intelrealsense.com/docs/compiling-librealsense-for-linux-ubuntu-guide#section-prerequisites

harshmittal2210, 2020-09-17:
Hi @MartyG-RealSense I have build librealsense from source code only using instructions mentioned in above link. I had done it again just to be sure. Still the error is there. I have attached my CMakeLists.txt, please have a look and let me know if I need to add something else as well.

[CMakeLists.txt](https://github.com/IntelRealSense/librealsense/files/5236881/CMakeLists.txt)

MartyG-RealSense, 2020-09-17:
Hi @harshmittal2210  Can you confirm please that you are able to run the **RealSense Viewer** program successfully after you have installed librealsense?  That would indicate whether your librealsense installation is working correctly.

Can you clarify for me please what you are trying to achieve.  

-  Are you trying to run the **rs-pointcloud** sample program and it is failing?  
-  Or are you trying to learn how to create your own program that can use OpenGL and it is failing because of a problem in linking to libraries, such as a missing linkage?

harshmittal2210, 2020-09-18:
Hi @MartyG-RealSense Yes, I am able to compile and install librealsense with no issues. All the example, tools and RealSense Viewer is working fine.

Here I am trying to build the same rs-pointcloud project outside the librealsense. I have copied the required folders and files in a separate folder. The folder structure looks like this:

```
test-pointcoud
├── build  
├── src
├── third-party
├── CMakeList.txt
├── example.hpp
└── rs-pointcloud.cpp
``` 
As you said I am trying to create a program that can use OpenGL but the above mentioned errors are coming. I have changes the header file locations also accordingly in the example.hpp

MartyG-RealSense, 2020-09-18:
The advice in the link below looks as though it meets the needs for what you are seeking:

https://github.com/IntelRealSense/librealsense/issues/4388#issuecomment-509727612

harshmittal2210, 2020-09-21:
Thank you so much!

closed 2020-09-21 by harshmittal2210