realsenseai/librealsense #7361: CMake ERROR

chungyutseng, 2020-09-16:
I key in this comment 
./vcpkg install realsense2
and something went wrong
below are the error messages
===========================================
CMake Error at scripts/cmake/vcpkg_execute_required_process.cmake:85 (message):
    Command failed: /bin/bash ./../src/3ae1ed06fa-f8218372d8.clean/configure --disable-silent-rules --verbose --disable-shared --enable-static --prefix=/home/chungyutseng/vcpkg/installed/x64-linux/debug --bindir=${prefix}/../tools/libusb/debug/bin --sbindir=${prefix}/../tools/libusb/debug/sbin --includedir=${prefix}/../include --datarootdir=${prefix}/share/libusb
    Working Directory: /home/chungyutseng/vcpkg/buildtrees/libusb/x64-linux-dbg
    Error code: 1
    See logs for more information:
      /home/chungyutseng/vcpkg/buildtrees/libusb/config-x64-linux-dbg-out.log
      /home/chungyutseng/vcpkg/buildtrees/libusb/config-x64-linux-dbg-err.log

Call Stack (most recent call first):
  scripts/cmake/vcpkg_configure_make.cmake:540 (vcpkg_execute_required_process)
  ports/libusb/portfile.cmake:53 (vcpkg_configure_make)
  scripts/ports.cmake:79 (include)

Error: Building package libusb:x64-linux failed with: BUILD_FAILED
Please ensure you're using the latest portfiles with `./vcpkg update`, then
submit an issue at https://github.com/Microsoft/vcpkg/issues including:
  Package: libusb:x64-linux
  Vcpkg version: 2020.06.15-unknownhash

Additionally, attach any relevant sections from the log files above.

somebody please help me

MartyG-RealSense, 2020-09-16:
Hi @chungyutseng   I should first emphasise that I have not got personal experience of vcpkg on Ubuntu.  Virtually the same error has been documented as occurring when using vcpkg with Windows though.

https://github.com/IntelRealSense/librealsense/issues/6131
https://github.com/IntelRealSense/librealsense/issues/6462

The second of those links provides much more detail about the problem.    In that second case, the information that provided a solution was in the comment linked to below:

https://github.com/IntelRealSense/librealsense/issues/6462#issuecomment-640006475

Bearing in mind that you are using Linux instead of Windows, please let me know if the links above do not solve your particular error with vcpkg.  Thanks!

MartyG-RealSense, 2020-09-26:
Hi @chungyutseng   Do you require further assistance on this case, please?  Thanks!

MartyG-RealSense, 2020-10-05:
Case closed due to no further comments received.

closed 2020-10-05 by MartyG-RealSense

adityag6994, 2021-04-25:
Hi @MartyG-RealSense I'm facing the same issue on Ubuntu 18, couldn't exactly understand how to use instructions given on [#6462](https://github.com/IntelRealSense/librealsense/issues/6462#issuecomment-640006475) , any updates would be really helpful. Thanks!

MartyG-RealSense, 2021-04-25:
Hi @adityag6994   As https://github.com/IntelRealSense/librealsense/issues/6462 relates to installing vcpkg on Windows, the short section of guidelines in the official librealsense documentation may be a better starting point.

https://github.com/IntelRealSense/librealsense#building-librealsense---using-vcpkg

vcpkg is a Microsoft tool, and their own general installation documentation (not specific to librealsense) is structured similarly to the librealsense notes.

It is worth mentioning that the Microsoft version adds the recommendation for Linux users to install **git** if not done so already and also use **g++** version 6 or newer.

https://github.com/Microsoft/vcpkg#quick-start-unix