realsenseai/librealsense #7362: Point cloud from live footage displaying

Xander-Vreeswijk, 2020-09-16:
I am using the intel realsense D455 camera and I have made a point cloud from the live image. I can use the helix toolkit and the command:
`points.ExportToPLY("pointcloud.ply", colorFrame);`
to show the point cloud from the c# cookbook:
```
using (var frames = pipeline.WaitForFrames())
                        {
                            var colorFrame = frames.ColorFrame.DisposeWith(frames);
                            var depthFrame = frames.DepthFrame.DisposeWith(frames);
                            var points = pc.Process(depthFrame).As<Points>();

                            // We colorize the depth frame for visualization purposes
                            var colorizedDepth = colorizer.Process<VideoFrame>(depthFrame).DisposeWith(frames);

                            // CopyVertices is extensible, any of these will do:
                            var vertices = new float[points.Count * 3];
                            // var vertices = new Intel.RealSense.Math.Vertex[points.Count];
                            // var vertices = new UnityEngine.Vector3[points.Count];
                            // var vertices = new System.Numerics.Vector3[points.Count]; // SIMD
                            // var vertices = new GlmSharp.vec3[points.Count];
                            //  var vertices = new byte[points.Count * 3 * sizeof(float)];
                            points.CopyVertices(vertices);
```
helixtoolkit loading the image:
```
            Model3DGroup model1 = import.Load(@Filename);
            model.Content = model1;
```
But this is only an image and not live. Can someone help me and explain how I can make a point cloud of live footage that changes when something changes in the footage.

MartyG-RealSense, 2020-09-16:
Hi @Xander-Vreeswijk   It sounds as though you are exporting a live point cloud to a ply file and then importing that ply file into Helix Toolkit.  But because it is a ply, the imported cloud is static and does not update in real-time.  Also, it is just an image and is non-interactive.  Is that interpretation correct, please?

If that is the case, an approach that may work is to embed a web browser page into a C# window (google **embed web page in c#** for more detais about how to do this).  You could then set it to display a live-updating web page containing a stream from the RealSense camera.

There have been a couple of projects for streaming a RealSense camera to a browser window:

https://github.com/IntelRealSense/librealsense/issues/5941

https://support.intelrealsense.com/hc/en-us/community/posts/360039895474-Highlighting-an-Intel-RealSense-depth-camera-web-browser-demo-program

Xander-Vreeswijk, 2020-09-16:
@MartyG-RealSense You are completely correct with your assumption. I will try what you have proposed. Thank you for your response.

AndreV84, 2020-09-18:
could you extend on how to inteall it on computer, please?
the github repo doesn't seem to contain explicit steps for building

MartyG-RealSense, 2020-09-18:
Hi @AndreV84   The **Depth Capture in HTML5** demo provides a link to a guide article in the link below that is authored by one of the demo's contributer developers:

https://01.org/node/5101

AndreV84, 2020-09-18:
is it likely that we could use d400/ l515/ t265?
I do not have S300

MartyG-RealSense, 2020-09-18:
Their documentation included information for calibrating for new camera models that were not available at the time that the demo was created.

https://github.com/intel/depth-camera-web-demo/blob/master/DEVELOPMENT.md

AndreV84, 2020-09-18:
calibration?
the html web page won't appear to detect cameras when I run the html code locally
is it a matter of calibration of a camera?

MartyG-RealSense, 2020-09-18:
I added info above about how the docs say that you can get calibration info for a particular model with **rs-enumerate-devices -c** and use the indformation to alter the demo's camera file.

AndreV84, 2020-09-18:
this info?
https://github.com/intel/depth-camera-web-demo/blob/master/DEVELOPMENT.md ?
were you able to run with following up it e.g d400 in web browser?

MartyG-RealSense, 2020-09-18:
Yes, that page.  I have not tried it myself, I was just highlighting its existence to the community.

AndreV84, 2020-09-18:
could you confiirm at your side that it works for d400, please?

MartyG-RealSense, 2020-09-18:
I have not got the time available to create the project myself, unfortunately.

MartyG-RealSense, 2020-09-18:
The notes in the code of the javascript file **depth-camera.js** say that it has been updated with D430, D435 and D435i calibration data in March 2019.

https://github.com/intel/depth-camera-web-demo/issues/14

https://github.com/intel/depth-camera-web-demo/blob/master/depth-camera.js

AndreV84, 2020-09-18:
@MartyG-RealSense ,
Thank you for pointing out.
I will try to get folks to connect d435/ d435i to try with remotely

AndreV84, 2020-09-18:
this one I was able to run with d435i
depthdemo.html
Will Point Cloud Virtualization work as is with d435i?
I am having hard times locating a file that executes Point Cloud visualization

MartyG-RealSense, 2020-09-18:
Do you mean this demo please?

https://philipnoonan.github.io/wglFusion/depthColorDemo.html

AndreV84, 2020-09-18:
I shall try with it.
Thank you very much.
However, it doesnt appear to run on aarch64, as aarch64 only supports chromium, but not chrome; therefore seems some issue

MartyG-RealSense, 2020-09-19:
@AndreV84    Does it make any difference if you launch it with **chromium unstable** instead of google-chrome-unstabe?

AndreV84, 2020-09-19:
Thank you for pointingf out!
for arm there see no straightforward way to find chromium unstable branch as per the linux magazine,
However, I will try to sort out how to get it on arm

AndreV84, 2020-09-19:
reference thread https://forums.developer.nvidia.com/t/installing-from-apt-testing/154805

AndreV84, 2020-09-24:
we installed on lin ux x86_64 chromium-broser-unstable
but it shows just white boxinstead of the camera d455 output

AndreV84, 2020-09-25:
@MartyG-RealSense 
Given there is full version Unity editor on linux, could you advise if there will be required any extra arguments for building the realsense sdk in order to integrate the d455 with unity?

MartyG-RealSense, 2020-09-25:
@AndreV84   There are too many unknowns involved for me to able able to give a confident answer.  For example, would the Linux version of Unity recognize the RealSense wrapper assets imported into the project by **realsense.unitypackage** if they were created in the Windows version of Unity.

AndreV84, 2020-09-25:
so I shall start locating unity windows assets & trying to import in Linux unity?

AndreV84, 2020-09-25:
so I first try to import https://github.com/IntelRealSense/librealsense/tree/master/wrappers/unity into unity linux, right?

MartyG-RealSense, 2020-09-25:
@AndreV84   I checked the process with a new project in Unity 2019.4 in Windows.  I followed these steps:

-  Downloaded the Unity package **realsense.unitypackage** 
-  Started a new '3D' template project in Unity
-  Double-leftclicked on the downloaded package file whilst the new project was open in the Unity window to import the contents of the package into Unity.

I experienced no problems with this in Windows and no messsage that the project uses Unity 2017.  Did you also try importing the package into a new empty project please?

AndreV84, 2020-09-25:
Thank you for following up!
trying to set up windows unity to try with d455

AndreV84, 2020-09-25:
will I need to build intel sdk on windows also?

MartyG-RealSense, 2020-09-25:
My recollection is that the Unity package should install the RealSense library file into the Unity assets folder.  So by all means try importing the package first.  If you find that you do need to install the SDK on Windows though, you can do it in a few minutes using the 793 mb Windows installer file on the SDK **Releases** page.

AndreV84, 2020-09-25:
assets import with the unity package without error there.
but I can not find any vlue how to run any camera output to there

AndreV84, 2020-09-25:
[image]

AndreV84, 2020-09-25:
[image]

MartyG-RealSense, 2020-09-25:
@AndreV84   That is the same view and Assets panel contents that I had in my test, so it certainly seems to be correct.

I believe that you are meant to press the Play triangle at the top of the Unity window to run the project and bring up a list of samples, then click on one of the sample options to run it.

[image]

AndreV84, 2020-09-25:
I installed 700mb->2gbv realsense sdk but it won't run, it will throw error;
I shall try playing the play triangle in unity

MartyG-RealSense, 2020-09-25:
Sorry I edited with new instructions (see above)

AndreV84, 2020-09-25:
[image]
after pressing play;
does the camera d455 require to be necessarily be plugged into usb3,0 port? usb 2.0 will work, will it?

AndreV84, 2020-09-25:
it wont show list of samples to me

AndreV84, 2020-09-25:
windows lists the d455 device in the device manager

MartyG-RealSense, 2020-09-25:
I tested the D455 in the Viewer on a USB 2 connection just now and it showed up.  I do not know if the same is true for the Unity wrapper though.

MartyG-RealSense, 2020-09-25:
Ok, I tested the StartHere scene thoroughly with all cameras.  D415 and D435i brought up the sample list.  D455 and L515 did not.

AndreV84, 2020-09-25:
so I am out of luck trying unity with d455?

MartyG-RealSense, 2020-09-25:
I will raise the subject with Intel but there is nothing else to do for now.

AndreV84, 2020-09-25:
I will get guys to conenct d435i at the remote side to test with
Thank you

AndreV84, 2020-09-25:
d435i worked on 2.1 speed in unity;
Thank you very much

MartyG-RealSense, 2020-09-26:
@AndreV84   Thanks for the update about USB 2.1 in Unity.

AndreV84, 2020-09-30:
@MartyG-RealSense 
May I know if this be used in Linux environment?somehow? likely? unlikely?
www.intelrealsense.com/dimensional-weight-software/

AndreV84, 2020-09-30:
the activation doesn't get through

MartyG-RealSense, 2020-09-30:
@AndreV84   The bottom of the information page that you linked to says that it currently works with **Windows**, with Linux and Android support 'coming soon'.  I am not able to answer questions about DWS though because it is an L515-related product and somebody else on the RealSense team is responsible for supporting that camera.   Please create a new GitHub case to ask about activation problems.

AndreV84, 2020-10-07:
could you point out how to retrieve realsense sdk version from terminal, please? @MartyG-RealSense 
we got the activation issue worked out; thanks.

MartyG-RealSense, 2020-10-07:
You could display the value of **RS2_API_VERSION** with scripting.  I do not know a command for displaying the SDK version in the terminal.

AndreV84, 2020-10-07:
the command for displaying sdk version in the terminal has been sorted out:
```
 rs-fw-update --version

rs-fw-update  version: 2.39.0
```

MartyG-RealSense, 2020-10-07:
Ah, that's a good way to do it.  Thanks!

Xander-Vreeswijk, 2020-10-08:
I have a question back to my first question but is it possible to show the points that are made through the generis support of the C# cookbook?

MartyG-RealSense, 2020-10-08:
Hi @Xander-Vreeswijk   I researched your question again but could not find any information about how to add to the C# generics point cloud example, unfortunately.

Xander-Vreeswijk, 2020-10-08:
no problem thank you for looking
If i find something I will share it in this thread

AndreV84, 2020-10-08:
@Xander-Vreeswijk 
we are trying to sort out showing points through https://github.com/fzi-forschungszentrum-informatik/gpu-voxels

AndreV84, 2020-10-08:
also for point cloud visuZlization Intel's Opern3D can be used 
ref:
https://github.com/devshank3/JetScan

Xander-Vreeswijk, 2020-10-08:
Can that be used as code in c#?

AndreV84, 2020-10-08:
I did not try but I can see folks were trying to do so 
ref: https://github.com/intel-isl/Open3D/issues/362#issue-324970517
try asking at open3d issues github about it;

Xander-Vreeswijk, 2020-10-08:
will do thank you

[35 more replies]

closed 2020-11-09 by Xander-Vreeswijk