realsenseai/librealsense #7364: Need an advice for choosing camera model, programming language for particular problem

Danilich1994, 2020-09-16:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC/Raspberry Pi |
| SDK Version                            |  2  | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }  | 
| Segment			|  Robot | 

Hello community!
I'm not sure where to ask such a question, but if I'm in wrong place, can you, please, advice me where can I get necessary info.

So, the problem I'm facing is kinda complicated (it seems to me). There is a large area (3 by 2.5 meters) filled with carton boxes. 
Boxes can be of different sizes, but their linear dimensions can't be bigger/lower then defined max/min dimensions. Boxes can be of different colors, have pictures, text, patterns, etc. on them (examples of boxes are shown in the picture). Boxes can be located within area verticaly or horizontaly (+-5 degrees), filling the whole area, or with gaps between them. I want to locate each box within area, find the coordinates of the center of each box, as well as he direction of the box (vertical or horizontal). The error of all measurements should be less then 5 cm. 

Initially, I was going to use a high resolution digital camera to get an image of the entire layer of boxes at the same time (from the top view). Attached picture is one of the example pictures I used to process (rgb to gray -> some filtering -> thresholding -> binarization -> edge detection -> extracting the necessary information). Unfortunately, my attempts were unsuccessful, due to the color inhomogeneity of the boxes (or, which is more likely, due to lack of necessary knowledge and experience). So I decided to search for another sensing instrument to use it standalone or in conjunction with RGB camera to get better results. So I ended up here in the world of depth cameras :) After a small analysis of depth cameras market I decided to use Intel RealSense cameras. 

I know that actual depth cameras don't have big resolution, so it's unreal to cover whole layer of boxes with one frame, so I consider to:
1. get more then one camera (6-8) and create cameras array, where cameras are placed in specified x,y coordinates within area. So I can capture depth data from each camera, and then unite all frames to get whole layer depth image.
2. take single camera and move it along x, y coordinates while capturing and build depth map of whole layer.

Fuuh, I hope at least something is clear -_-
Let's move on to the questions:

Which Intel depth camera model best suits my needs?
Which programming language I should use for processing? (I got some exp with python, I'm sufficient c++ user in microcontroller programing)
Is it even possible to succeed?
Perhaps there is another approach to solving this problem, your thoughts?

Thanks in advance for your advices and help!!!!!
[image]

MartyG-RealSense, 2020-09-16:
Hi @Danilich1994   I carefully consider your project's needs and the vision computing tasks that might be involved in detecting and analysing multiple boxes of different kinds simultaneously.  It seems to be a task that will involve training a computer AI that can use the camera to observe the scene and apply algorithms to it in order to make decisions (determining the orientation of the box, what specific product it is from an RGB image of its packaging graphics, etc).

In regard to specific RealSense camera technology, the RealSense 400 Series cameras will best meet your needs, as they can be used in multiple numbers without interfering with each other and they work excellently both indoors and outdoors, so can cope with the variable lighting and shadow that may be encountered in a typical warehouse environment.  

They are also well proven in robot arm applications, including those with AI intelligence tasks such as identifying and picking individual objects in an untidy pile of multiple objects.  Just one example of this is LIPS Corp's RealSense-powered **Vision Guided Robotics** system.

https://www.lips-hci.com/lipsarm-vgr-footwear

Up to twenty 400 Series cameras have been used to capture individual scans that can be stitched together into a single scan, so that aspect of your plans is proven if you decide to pursue the multiple camera route.  The scenario of stitching up to 20 camera images was demonstrated by the CONIX Research Center at Carnegie Mellon.

https://github.com/conix-center/pointcloud_stitching

You can certainly also take the approach of using a single camera and moving it around the scene.  Vision computing systems can use AI and object data-sets to identify and classify multiple unique objects whilst in motion.

In regard to software, both C++ and Python are well supported with programming resources for RealSense, created by Intel and by its community of users.

MartyG-RealSense, 2020-09-26:
Hi @Danilich1994   Do you require further assistance with this case, please?  Thanks!

Danilich1994, 2020-09-27:
Hello @MartyG-RealSense. Thank you for your reply. I made a big progress with my project during past 11 days, it was very helpful. And I'm going to post some update with bunch of questions soon. So, If it's possible to keep this topic active, it would be very handy (no need to open new one) ;)

MartyG-RealSense, 2020-09-27:
No problem @Danilich1994  I will keep the case open for a longer period.  Thanks for the update!

mpottinger, 2020-09-30:
I am slowly working on a similar project in a warehouse environment, where I need to count or measure volume of pails and boxes.

For me it is not practical to use a multi camera setup because this is a warehouse with hundreds of small rack spaces that need to be scanned, its physically impractical to get a full birds eye view of the boxes all at once.

A single camera moving around is a much more complicated situation. I have been slowly working at it for over a year. I need a compact handheld setup though, so I use a Huawei phone with built in depth sensor, and use the slam in ARCore or AREngine for position tracking.

This position is used to transform the point cloud to world coordinates, then I render the fused point cloud as a top down view map. 

The Android app is written in Java but the heavy processing of the point cloud images is done in a glsl OpenGL frag shader.

A T265 with a D400 series could achieve something similar but is not quite as convenient.

MartyG-RealSense, 2020-09-30:
Hi @mpottinger   I wonder if an autonomous patrolling robot such as **Tally** may suit your environment.  If the camera was on a mount that could move up and down and alter its height up and down at the start of a patrol sequence then over time all of the racks - high and low - may be able to be scanned over the course of multiple patrols.

https://www.youtube.com/watch?v=ktiQgFPDtr4

Tally uses a combination of a RealSense stereo camera and a lidar camera for navigation.  Conceivably, the two functions could be combined into one 400 Series camera using a package such as Intel's **ros_object_analytics** to provide both SLAM navigation and object detection / avoidance.

https://github.com/intel/ros_object_analytics

mpottinger, 2020-10-03:
@MartyG-RealSense Yes that is something I have considered. That would be a long term solution though. In the meantime I am just finding out if 3D mapping will even be able to achieve the kind of accuracy that I need in a handheld device. A robot would be the next level, but considerably more investment and may not be attractive to some, cost wise. Also it is in a hazmat environment so electronics are frowned upon, though I am able to justify something like a phone, anything with significantly higher power or exposed electronics might be a hazard.

MartyG-RealSense, 2020-10-03:
Given the environment that you are operating in, a  rugged industrial Windows tablet with a 400 Series camera built into it may be a suitable solution.  Examples of such tablets that have a D410 or D415 camera built in are below.

https://www.officer.com/command-hq/technology/computers-software/tablets-handhelds/press-release/21027456/dt-research-the-dt301x-rugged-tablet-is-meant-for-high-accuracy-measurements

https://ruggedtablet.msi.com/product/NF21 

https://business.panasonic.co.uk/mobile-solutions/news-and-events/panasonic-introduces-a-fully-rugged-tablet-with-integrated-intel-realsense-3d-camera-to-capture

Please bear in mind that technical support for such devices will be the responsibility of the manufacturer rather than Intel, as they are not Intel products.

MartyG-RealSense, 2020-10-12:
Hi @Danilich1994   Do you still require assistance with this case, please?   Thanks!

Danilich1994, 2020-10-14:
Hello @MartyG-RealSense. I almost finished writing update post, will try to finish and put it here till the end of this week. Sorry for delay, I'm crazy busy at work right now -_-

MartyG-RealSense, 2020-10-14:
No problem at all, I completely understand @Danilich1994 - thanks for the update!

MartyG-RealSense, 2020-10-22:
Adding a note to keep this case open for a further period of time.

MartyG-RealSense, 2020-11-15:
Hi @Danilich1994   Do you have an update that you can provide yet, please?   Thanks!

Danilich1994, 2020-11-15:
@MartyG-RealSense big thanks for keeping this issue up. Sorry for delay, I'll post today, but a bit later :)

Danilich1994, 2020-11-15:
@MartyG-RealSense  Thank you one more time for keeping this thing alive!

So, here I am with some results. I chose IES D415 Camera coz it looks like it provides best depth image quality for my particular task. D435, as I understood from reviews and videos, more focused on gestures and object recognition in motion. In terms of camera itself, I managed to connect it and write some python code to get this kind of image(1):
[image]
It's 'Hand' camera preset, with following combination of filters (the best I managed to find): 
1. decimation filter with default settings (shrink image 2 times)
2. depth -> disparity.
3. spatial filter (magnitude 2, alpha 1 to beta 30).
4. disparity -> depth.
5. threshold filter (0.4-0.5) to get depth image only about boxes surface.

This way I'm geting enough precise picture of boxes (here I'm using small size boxes placed on desk in front of camera) to clearly see edges and surfaces. But there is some noise on surfaces, as I understood it comes from laser beams, and it bothers me. Is there any way to get rid of the noise? Next picture(2) show performance of canny edge detection in previous picture( max in min values are computed automaticaly using median). It can be seen that it catches all noise from innitial picture.
[image]
To solve problem with noise, I desided to blur image (gaussian) and then threshold it into binary image, followed by erosion + dilation, and finally canny edge detection (next four pictures(3,4,5,6) show the results).
[image]
[image]
[image]
[image]

Next step - grab one image and feed through "box finding" algorithm. The best result I was getting with picture 2 (just canny edge detection of depth image) as an input. Algorithm:

0. flood fill the image to fill areas inside edges
1. scan whole image with "window" (4 pixels x image height/width) vertically and horizontaly to get vertical and horizontal gradients next.
2. threshold gradients to find all vertical and horizontal lines.
3. find all crossing points and pretend that they are boxes corners.
4. each point-corner has four quadrants, so scan them to find out white areas. Here we threshold absence/presence of box in each quadrant marking each point with number (Numbers are generated as deciamal reptresentation of binary code, where each binary digit represent quadrant; 1- quadrant is white, 0 - black). For example: point with '0' means that all it's quadrant are black - empty (such points are deleted from corners pool), but point with number '15'  - white all around, may be point between 4 boxes or point in the middle of box.
5. filter all irrelevant points. Create real corners pool.
6. go through corners pool and "collect" groups of 4 points to form boxes. Here I'm using information about maximal/minimal boxes sizes and aspect ratio to unite points in groups.
7. Find boxes middle points and orientation.

Next pictures(7,8) shows result of algorithm performance (used another boxes configuration and amount) - run time 0,03 sec. Blue circles represents quadrants scaning area. 
[image]
[image]
My algorithm can work only with pictures where boxes are positioned vertically or horizontaly, it is downside, but it is task condition. I tried using Hough Line Transform algorithm but I didn’t manage to get any clear result.
The next step is to buy another camera (or more then one) and stitch their images to increase the frame area without losing resolution. Furthermore, connect them to raspberry and try to process images there.

Thats kinda all :)
Some questions:
1. Is there any way to get rid of the noise? (Just repeating this one)
2. Maybe there are more advanced algorithms for finding shapes (in my case, rectangles) in the image? I was thinking about using some kind of rectangular mask and scan image with it as an option.
3. Can you point out any drastical misstakes in my box finding algorithm? What can be changed/improved?
4. Any advices to improve depth image quality coming from camera? Is my camera selection correct? May be you can suggest another model?

Thanks for your answers in advice! I am working on this project alone, don't even have any person to discuss certain ideas or decisions with!
@MartyG-RealSense  huge apologies for the delay with posting -_-

MartyG-RealSense, 2020-11-16:
Apologies for the delay in responding further.  Thanks so much for providing high detail in your explanations, which is sure to benefit other RealSense community members reading this case in future.

1.  If you believe that the noise is related to the projector dot pattern (noise known as **laser speckle**) then you can reduce this by using an external LED pattern projector (which may be too bulky to be mobile if the camera needs to be moved around) or by using a pattern-removal visual preset called **Left Imager Color w/o IR pattern**.  This preset is supported on D415, and the functionality is coming for the new D455 model too.

https://dev.intelrealsense.com/docs/d400-series-visual-presets#section-preset-table

2 and 3 are not specialist areas of knowledge for me, so I wuld recommend creating a new case on this forum just for those questoins.

4.  The optimal resolution for depth accuracy on the D415 is 1280x720 (higher than the D435 and D455's 848x480 optimal depth resolution), so if you are able to, I wpuld recommend using 1280x720 for optimal accuracy.

Given the warehouse environment that you are using the camera in, the D415 may be your best option of RealSense model if you are scanning from a static position and not moving the camera around quickly whilst sensing (otherwise RGB blur could occur).  Whilst the L515 lidar depth camera can process 23 million accurate depth points per second, it works best in controlled lighting conditions and would not be suitable for outdoor sensing if you need to scan in outdoor areas.

If you have to do outdoor sensing **and** the camera is moving around quickly, the D455 would be a good choice as it has a fast shutter on its RGB sensor to help avoid motion blur on the RGB image if the motion is faster than human walking pace.

Danilich1994, 2020-11-23:
Hello @MartyG-RealSense! Thank you for assistance!  And, I think that for now I no longer need help. If I have any questions, I will open a new issue and provide a link to this one. Thanks again!

MartyG-RealSense, 2020-11-23:
Okay, thanks very much for the response.   I will close this case now as you are satisifed. with the outcome  Thanks again!

closed 2020-11-23 by MartyG-RealSense

Oburshan, 2021-02-21:
Hey @Danilich1994 , Can you share your code with us? 
I'm dealing with similar problem (with window detection & realsense camera) 
Thanks a lot and have a great day!

MartyG-RealSense, 2021-02-21:
Hi @Oburshan  If your window detection problem is related to glare from reflections off glass panes then this negative effect can be significantly reduced by adding a physical **linear polarization** filter over the lenses on the outside of the camera.  Section 4.4 of Intel's white-paper document about optical filters describes these principles.  The improvement that this filter type can provide when observing windows is demonstrated in the image below.

https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400#section-4-the-use-of-optical-filters

[image]

If you would prefer a software-based solution, aligning depth to color can help, as dmeonstrated in Intel's Python **depth_under_water** tutorial about sensing through water or aquarium glass.

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/depth_under_water.ipynb

If you have a row of similar looking windows instead of just one window then the camera may be being confused by a **repetitive pattern** phenomenon that can occur when observing a horizontal or vertical repeating pattern, such as a horizontal row of fence posts or a vertical row of window blinds.  If this might be the case, the link below provides links to resouces for combatting repetitive patterns. 

https://github.com/IntelRealSense/librealsense/issues/6713#issuecomment-651114720

Oburshan, 2021-02-23:
Hey @MartyG-RealSense, thanks for your quick response.
I'm currently trying to use the depth camera (d435i) with my door instead and the depth information i get is quite wrong..

The image attached is after erode & dilate and bilateral filter which made it look better but still not the result i am looking for because of those black values near the door..

If you have any filters / other suggestion it will be great! 
Thanks ! 

[image]

MartyG-RealSense, 2021-02-23:
Hi @Oburshan   Textureless or low-texture surfaces such as doors, walls, desks and chairs can be difficult for the camera to analyze for depth detail.  If you enable the camera's **IR emitter** component then a semi-random dot pattern is projected onto the surfaces in the scene.  The camera can use these dots cast onto surfaces as a texture source to analyze for depth detail.

If you tell me which programming language you are using (C++, C#, Python, etc) then I can direct you to scripting for enabling the IR emitter if you wish to try projecting a dot pattern onto the scene.

Oburshan, 2021-02-24:
Hey, i am using python, it will be great thanks @MartyG-RealSense.

I applied the holes fill in spatial filter and it keeps improving :)

  @Danilich1994 may you share your box finding code?
  I am currently at picture 6 step and it would be amazing in order to keep going.
  
  Thanks and have a great day!

MartyG-RealSense, 2021-02-24:
Hi @Oburshan   The Python script in the link below which contains code for setting both the IR emitter and post-processing filters may be helpful for you.

https://github.com/IntelRealSense/librealsense/issues/4095

The section of the script that enables the IR emitter is highlighted below

[image]

An alternate Python approach for setting the state of the IR emitter is here:

community.intel.com/t5/Items-with-no-label/How-to-enable-disable-emitter-through-python-wrapper/m-p/547902#M8384

Setting the emitter to a value of '0' disables it and setting it to '1' enables it.

Danilich1994, 2021-03-01:
Hello, @Oburshan. I can't provide code itself, it's my job project. But overall algorithm is described, so you ca try to code it yourself. All necessary info can be found in google, but I would be happy to answer your question in terms of coding.
For now I changed algorithm a bit. Step 4 is different now - I created binary corners templates and compare them through XOR with obtained picture corners to filter out "not-a-box-corner" line crossings.
Please notice, that my algorithm works only when boxes sides are orthogonal to picture sides. If you need to find rectangles which are angled on a picture then open cv functions  cv2.findContours -> cv2.minAreaRect will help you!

Oburshan, 2021-03-02:
Hey, @Danilich1994,

Danilich1994, 2021-03-02:
Hi, @Oburshan 
I didn't have this problem, coz I'm working with steady captured images.
If your edge detector give noisy results then you feed him noisy frames. How do you process your frames from moment of camera capture till "frame is ready for edge detection" moment? What filters and image processing do you use?