realsenseai/librealsense #7366: USB Overflow and no RGB frames in Ubuntu 20.04 / Python 3.8

masynthetic, 2020-09-17:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.12.07.100) | 
| Operating System & Version |   Ubuntu 20.04  | 
| Kernel Version (Linux Only)    |  5.4.0-45-generic    | 
| Platform                                 | Dell XPS 13 Laptop AND Lattepanda Delta |
| SDK Version                            |  { legacy / 2.<38>.<0> }                          | 
| Language                            |  {python / realsense viewer}                          | 

### Issue Description
I have successfully built pyrealsense2 from source inside Ubuntu 20 with python 3.8 using:

```
wget https://github.com/IntelRealSense/librealsense/archive/v2.38.0.tar.gz && tar -xvf v2.38.0.tar.gz && cd librealsense-2.38.0/ && mkdir build && cd build &&  cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_WITH_CUDA=true -DBUILD_GRAPHICAL_EXAMPLES=true && make -j4 && sudo make install
```
I must run *sudo* realsense viewer or I get 
```
WARNING [default] Could not open device failed to set power state
``` 

Inside of realsense viewer I am able to start the depth camera but attempting to start the RGB camera I get "no frames received" an inflow of "USB cam overflow" errors. In the terminal is the error:

```
 16/09 21:10:03,763 WARNING [140200743266048] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
```
I have not found any more descriptive errors.

### Some additional info
- under the "info" button in viewer it shows "CAMERA LOCKED: YES" I don't really know what this means 
- more interestingly is that the camera ALWAYS shows as on USB2.1. It shows it on 3 different machines which all have USB 3 ports, regardless of using Ubuntu or windows (where RGB works), I have tried 4 different cables, primarily a 10 foot one which definitely works with USB 3. 
- I have tried rolling back the firmware all the way to 5.10.3 as someone mentioned that as the fix to their similar issue but nothing
- I have tried decreasing framerate and resolution to the minimums

MartyG-RealSense, 2020-09-17:
Hi @masynthetic   It looks as though you are acting on a suggestion in https://github.com/IntelRealSense/librealsense/issues/6436 by a RealSense user to install librealsense through the **RSUSB** method in the way that you have used above.  However, the RealSense user providing that advice states that it is designed for use with **Docker**.

It would certainly be abnormal behavior if you have to run realsense-viewer in order for pyrealsense2 to work, as the two should not have any connection between them and be totally independent.

If you are not using Docker though (as most people installing librealsense will not be) then the method in the link below for installing librealsense with RSUSB should be suitable for you.  

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908

masynthetic, 2020-09-17:
Thank you @MartyG-RealSense I didn't realize I had overlooked this being for docker as well as the viewer version and cuda factors. All that I meant about using sudo realsense-viewer was needing the sudo part, the issue reproduces anytime I try and use the pipeline, was just using realsense-viewer as example really.

I'll give those things a try and report back

MartyG-RealSense, 2020-09-17:
A more conventional version of the RSUSB installation method that splits the instructions into multiple stages instead of a single long instruction is available in the link below.  Again, ignore the CUDA part if you do not have an Nvidia GPU in your computer.  Good luck!

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908

masynthetic, 2020-09-18:
I broke it down a bit and removed the CUDA stuff and everything works great now! No sudo needed for the power state error either.

```
wget https://github.com/IntelRealSense/librealsense/archive/v2.38.1.tar.gz && tar -xvf v2.38.1.tar.gz && cd librealsense-2.38.1/ && mkdir build && cd build
cmake ../ -DFORCE_RSUSB_BACKEND=true -DBUILD_PYTHON_BINDINGS=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true
make -j4 
sudo make install
```

closed 2020-09-18 by masynthetic

MartyG-RealSense, 2020-09-18:
Thanks very much for the update - I'm pleased that you were successful