realsenseai/librealsense #7369: How to develop my own work in realsense demo?

1556260456, 2020-09-17:
I just install the librealsense and realsense-ros,and I want to get deepths of all pixels,but I found the Cmake error after I copy the code "rs-hello-realsense.cpp" and run : cmake ..  ,the error is follow:

CMake Error at CMakeLists.txt:14 (install):
  install TARGETS given no RUNTIME DESTINATION for executable target
  "rs-hello-realsense".

-- Configuring incomplete, errors occurred!
See also "/home/adam/hello-realsense/build/CMakeFiles/CMakeOutput.log".

and the Cmakelists is:
cmake_minimum_required(VERSION 3.1.0)

project(RealsenseExamplesHelloRealSense)

add_executable(rs-hello-realsense rs-hello-realsense.cpp)
set_property(TARGET rs-hello-realsense PROPERTY CXX_STANDARD 11)
target_link_libraries(rs-hello-realsense ${DEPENDENCIES})
include_directories(rs-hello-realsense)
set_target_properties (rs-hello-realsense PROPERTIES FOLDER "Examples")

install(TARGETS rs-hello-realsense RUNTIME DESTINATION ${CMAKE_INSTALL_BINDIR})

MartyG-RealSense, 2020-09-18:
Hi, I researched the error **install TARGETS given no RUNTIME DESTINATION for executable target
"rs-hello-realsense"** and located a similar case.

https://github.com/IntelRealSense/librealsense/issues/2946

When you say that you "copied" the code, could you provide more information about this please.  Did you copy the **hello-realsense** example files to a new folder location and try to build the example from there, please?

MartyG-RealSense, 2020-09-26:
Hi @1556260456   Do you still require assistance, please?  Thanks!

1556260456, 2020-09-26:
Yes,I copied the demo from librealsense to a new   folder on my desktop ,then try to build and run it.And I had do it correctly by rewrite the CMakelists.

MartyG-RealSense, 2020-10-05:
Hi, do you still require assistance with this case please?   Thanks!

1556260456, 2020-10-06:
@MartyG-RealSense thanks for your help,I do not need anymore!

closed 2020-10-06 by 1556260456