realsenseai/librealsense #7370: D435 device can not be detected on ubuntu

RZz96, 2020-09-17:
Hi 
I have installed the SDK2.38 on ubuntu 18.04, but when i run the code , it can not detect the camera . then i tried the viewer and there is no camera (of course because it didnt detect the camera  )

hope someone can suggest something

Regards

MartyG-RealSense, 2020-09-17:
Hi @RZz96   Can you describe the method that you used to build SDK 2.38.1 on Ubunty please?   Thanks!

RZz96, 2020-09-17:
> 
> 
> Hi @RZz96 Can you describe the method that you used to build SDK 2.38.1 on Ubuntu please? Thanks!
i tried pakage and source code

RZz96, 2020-09-17:
I tried both pakage and source code . 
please tell me what to do

MartyG-RealSense, 2020-09-17:
Hi @RZz96   Are you also posting on the Intel Support forum on the link below, please?  If so, I have posted a reply there on that case.

https://support.intelrealsense.com/hc/en-us/community/posts/360048315774/comments/360014081253

RealSenseSupport, 2020-09-17:
Hi @RZz96 

Can you provide a little more details on your system hardware / cpu and what the failure is. Error logs or steps that can help reproduce. We are able to run all realsense cameras on Ubuntu 18.04.

Thanks

MartyG-RealSense, 2020-09-26:
Hi @RZz96   Did you manage to solve your problem please?  If not, can you provide further details requested by @RealSenseSupport  above.  Thanks!

closed 2020-10-05 by MartyG-RealSense