realsenseai/librealsense #7371: D435 get_distance and dept frame pixel value *depth_scale isn't same

asbilgi, 2020-09-17:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415, D435 } | 
| Firmware Version                   | (5.12.6) | 
| Operating System & Version |   {Linux (Ubuntu 16)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.15.0)                                         | 
| Platform                                 | PC / nano Pi M4  |
| SDK Version                            |  {2.34.1}                          | 
| Language                            |  {C++ }                          | 

### Issue Description
I want to obtain the depth of some point in depth frame. But get_distance() and pixel value *get_depth_scale isn't same. Here is sample code: 

` int main(){

    rs2::config m_config;

    m_config.enable_stream(RS2_STREAM_DEPTH, 1280, 720, rs2_format::RS2_FORMAT_Z16, 15);

    rs2::pipeline m_pipeline;

    rs2::pipeline_profile m_pipelineProfile = m_pipeline.start(m_config);

    std::vector<rs2::sensor> m_sensorList = m_pipelineProfile.get_device().query_sensors();

    while(true){

        rs2::frameset frameset = m_pipeline.wait_for_frames();

        rs2::video_frame tIRFrameL = frameset.get_infrared_frame(1);

        rs2::depth_frame tDepthFrame = frameset.get_depth_frame();

        cv::Mat tImageDepth1DTest;

        tImageDepth1DTest = cv::Mat( cv::Size(1280, 720), CV_16U, (uchar *)( tDepthFrame.get_data()), cv::Mat::AUTO_STEP);

        int testtt = tImageDepth1DTest.at<uchar>(500, 500);

        float tscale = 0.;

        for(auto tSensor : m_sensorList){

            std::string tSensorName(tSensor.get_info(RS2_CAMERA_INFO_NAME));

            if( tSensorName == "Stereo Module"){

                tscale = tSensor.as<rs2::depth_sensor>().get_depth_scale();

            }

        }

        std::cout << "Scale: " << tscale << " Pixel value: " << testtt << " distance: " << tDepthFrame.get_distance(500, 500) <<std::endl;

    }}`

and result here:
Scale: 0.001 Pixel value: 246 distance: 3.549
Scale: 0.001 Pixel value: 234 distance: 3.513
Scale: 0.001 Pixel value: 234 distance: 3.489
Scale: 0.001 Pixel value: 40 distance: 3.549
Scale: 0.001 Pixel value: 234 distance: 3.525
Scale: 0.001 Pixel value: 66 distance: 3.562
Scale: 0.001 Pixel value: 185 distance: 3.574

MartyG-RealSense, 2020-09-18:
Hi @asbilgi   I hope that the discussion in the link below will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/7264

asbilgi, 2020-09-18:
Hi @MartyG-RealSense link very helpful to me. 
To obtain depth value you should convert CV_64F also in opencv.
And found cv_helper.hpp it's include opencv conversation.
https://github.com/IntelRealSense/librealsense/blob/025fccf76803ee6a6e60de9f18ac6193b7ff8597/wrappers/opencv/cv-helpers.hpp

closed 2020-09-18 by asbilgi

sandeshk1, 2021-01-13:
Could someone explain how the two values were fixed and same

MartyG-RealSense, 2021-01-13:
Hi @sandeshk1   I think you are asking how @asbilgi achieved getting the same distance values from both get_distance and by multiplying the pixel value by the depth unit scale?

sandeshk1, 2021-01-13:
@MartyG-RealSense Thats right.. Thanks