realsenseai/librealsense #7372: Two L515 HW Sync master/slave

finde, 2020-09-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.05.00.00 | 
| Operating System & Version |   Win 10  | 
| Platform                                 | PC / Surface Pro  |
| SDK Version                            |  2.38.1                          | 
| Language                            |  C                          | 

### Background

My current setup only requires two lidars (L515) + a tracker (T265), attached to a moving object.
I am looking for a Master/Slave settings without an external trigger.
I am using two lidars at the same time to extend the FOV of depth sensors.
The current bottleneck of the setup is the synchronization between two lidars.

### Issue Description

#### 1. External Trigger

In the white paper (https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup), you mentioned:

> For multi-camera case, an external signal generator should be used as the master trigger with all cameras set to slave mode.  The trigger voltage amplitude should be @3.3V, and the first frame is triggered on the positive (rising) pulse edge ...

**Q:** Can it work without using an external trigger? 

#### 2. Multi-Camera Programming

In the white paper, you have `r.get_device().query_sensors().front().set_option(RS2_OPTION_INTER_CAM_SYNC_MODE, 1.f);`

**Q:** Is this also applied the same to all lidar? Does it suppose to make the lidar become a "slave"?

I checked the D400 (`ds-private.h`), it written:
```
        enum inter_cam_sync_mode
        {
            INTERCAM_SYNC_DEFAULT    = 0,
            INTERCAM_SYNC_MASTER     = 1,
            INTERCAM_SYNC_SLAVE      = 2,
            INTERCAM_SYNC_FULL_SLAVE = 3,
            INTERCAM_SYNC_MAX        = 258 // 4-258 are for Genlock with burst count of 1-255 frames for each trigger
```
I guess this is not the same case for L515.

RealSenseSupport, 2020-09-25:
Hi @finde 

That is correct. You must use an external trigger for L515 and each camera must have it's option set (slave) separately.

Thanks

finde, 2020-09-29:
I see, thank you for the answer

closed 2020-09-29 by finde