realsenseai/librealsense #7373: > A .raw image stores depth data, yes.

sayyidabeegam, 2020-09-18:
> A .raw image stores depth data, yes.
> 
> The .csv format stores depth data in a text format that is suitable for import into database and spreadsheet programs.
> 
> [image]

How did you saved this depth data as csv file.rs-save-to-disk is saving metadata only in csv format

_Originally posted by @sayyidabeegam in https://github.com/IntelRealSense/librealsense/issues/3632#issuecomment-694688648_

sayyidabeegam, 2020-09-18:
How to save this depth data as csv file, rs-save-to-disk is saving meta data only as csv format?

MartyG-RealSense, 2020-09-18:
Hi @sayyidabeegam   Yes, a .raw image stores depth data and so is a better option than png for doing so, as saving to png causes depth information to be lost.

The example program **rs-capture** has the ability to save frames to csv.

https://github.com/dorodnic/librealsense/blob/text_sample/examples/capture/rs-capture.cpp#L15

You may also find to be helpful the discussion in the link below about saving depth data to formats such as .csv and .raw.

 https://github.com/IntelRealSense/librealsense/issues/1485#issuecomment-379665281

MartyG-RealSense, 2020-09-26:
Hi @sayyidabeegam   Do you still require assistance, please?  Thanks!

closed 2020-10-05 by MartyG-RealSense