realsenseai/librealsense #7374: CMake file for Librealsense generates Error when built: could not find git for clone of libusb

EvaKrassnig, 2020-09-18:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (x) | 
| Operating System & Version |   {Win (10) | 
| Kernel Version (Linux Only)    |  (-)                                         | 
| Platform                                 | PC, Lenovo |
| SDK Version                            |  { SDK 2.0}                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Surface Plotting }                          | 

### Issue Description
<I want to add PCL libraries and followed your guide for the PCL wrapper. However, CMake does not allow to build the CMakeLists. The Error and Output.log are attached.

[CMakeError.log](https://github.com/IntelRealSense/librealsense/files/5244212/CMakeError.log)
[CMakeOutput.log](https://github.com/IntelRealSense/librealsense/files/5244218/CMakeOutput.log)

EvaKrassnig, 2020-09-18:
Edit: After installing git it worked. Now I am stuck when I try to build the rs-pcl project. It gives me the following error: LNK1104, glfw3.lib cannot be opened (same as this unsolved issue: #4929). I am using Visual Studio 2019.

MartyG-RealSense, 2020-09-18:
Hi @EvaKrassnig   In your CMake GUI settings, did you have **BUILD_EXAMPLES** and **BUILD_GRAPHICAL_EXAMPLES** ticked as the illustration in the PCL guide suggests?  **BUILD_GRAPHICAL_SAMPLES** provides OpenGL support for RealSense SDK programs that make use of OpenGL.

EvaKrassnig, 2020-09-18:
Hi, yes, build_graphical_examples and build_examples is ticked (see screenshot attached). What I find suspicious is that glfw3_DIR is under unropued Entries and "Directory is not found" (see screenshot2 attached).

[image]

[image]

MartyG-RealSense, 2020-09-18:
The link below provdes information about adding glfw to a project that you are creating yourself.

https://github.com/IntelRealSense/librealsense/issues/4388#issuecomment-509727612

EvaKrassnig, 2020-09-18:
[CMakeLists.txt](https://github.com/IntelRealSense/librealsense/files/5248625/CMakeLists.txt)

I downloaded glfw-library and built it. I tried the integration in CMake as described in #4388, however I always get an error message. How do I need to adapt attached CMake file for it to finde the Glfw-library?
Thank you for your help,
BR;
Eva

MartyG-RealSense, 2020-09-19:
I have carefully researched your case from the beginning again but did not find anything that stood out as incorrect.  May I confirm please that you ticked the **BUILD_SHARED_LIBS** option in CMake as the PCL wrapper instructions state as a necessity to ensure that there are not problems with PCL linkage?

I would also advise that when you make your next build attempt, start over from the beginning of the guide instead of modifying an existing attempt .  Thanks very much for your patience.

EvaKrassnig, 2020-09-24:
Build_shared_libs is ticked. I tried from start on a different PC with Windows 10 (x64) installed, but I got exactly the same error. I manually installed the OpenGL library as shown in this video: https://www.youtube.com/watch?v=XiWNi5Gk0zc and the rs-pcl application worked, but the rs-pcl color did not. Could there be something wrong with the wrapper? 

Also, I have the device connected to a USB2 hub (USB3 is not working at all) and the datastream is insufficient, so I only see a few points and not the full picture.

MartyG-RealSense, 2020-09-24:
Are you able to get a point cloud image at all with **rs-pcl-color** please?  And if you can get one, is it like the point cloud in the link below to a case where someone else had problems with the color in this example?

https://github.com/IntelRealSense/librealsense/issues/5635

MartyG-RealSense, 2020-10-02:
Hi @EvaKrassnig   Do you still require assistance with this case, please?  Thanks!

EvaKrassnig, 2020-10-06:
Hi MartyG, 
I did not get the example running unfortunately. It is not the same error as described in #5635, it has to be something with the CMake file. For the time being I will try with Open3D library.

MartyG-RealSense, 2020-10-06:
Thank you very much for the update.  Let us know later how things go with Open3D please.

EvaKrassnig, 2020-10-07:
Hi Marty,
short feedback: Open3D works well with IntelRealSense, very uncomplicated in implementation. I am missing a few features but for fast results Open3D can definitely be recommended.

MartyG-RealSense, 2020-10-07:
Great news that your tests turned out well for you!  Thanks for sharing your experiences with the RealSense community!

MartyG-RealSense, 2020-10-14:
Hi @EvaKrassnig   Do you require further assistance with this case, please?   Thanks!

EvaKrassnig, 2020-10-14:
Hi, no thank you, I will close this issue.

closed 2020-10-14 by EvaKrassnig

MartyG-RealSense, 2020-10-14:
Thanks very much @EvaKrassnig for the update!

nikstp, 2020-10-28:
Hi, 

I'm actually having the same issue on a Dell WIN10 laptop with the OpenGL package missing even with **BUILD_SHARED_LIBS** and **BUILD_GRAPHICAL_EXAMPLES** whenever I try to build the PCL wrapper and examples using CMake.

MartyG-RealSense, 2020-10-29:
Hi @nikstp  Have you got **BUILD_EXAMPLES** selected as well as **BUILD_GRAPHICAL_EXAMPLES**?

[image]