realsenseai/librealsense #7376: 'Frame didn't arrived withing 15000' exception

MikhailUskin, 2020-09-18:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version | Win (8.1/10) | 
| Platform                                 | PC |
| SDK Version                           | 2.38.1| 
| Language                               | C++ | 

Hi,

I have installed the latest SDK v2.38.1 and checked whether everything is correct using 'Intel RealSense Viewer'. Both channels 'Stereo Module' and 'RGB Camera' functioning looked normal but my test app throws an exception when rs2::pipeline::wait_for_frames() is called with message 'Frame didn't arrive within 15000'. The same issue is reproducible if I run 'capture' sample provided by SDK.

MartyG-RealSense, 2020-09-19:
Hi @MikhailUskin   Could you tell us what computer / computing device you are using the camera with please, and whether you are using a USB2 or USB3 connection?  Thanks!

MikhailUskin, 2020-09-19:
The camera is recognized as USB3 device by 'Intel RealSense Viewer'. The app that throws this exception is running on Win10

MartyG-RealSense, 2020-09-19:
Is it your experience with this error that once it occurs then the program is unrecoverable and the error continues generating constantly until the camera is reset by being unplugged and then plugged back in?

MikhailUskin, 2020-09-19:
Everything looks fine with 'Intel RealSense Viewer' but when I run any code to catch frames using rs2::pipeline::wait_for_frames() call it throws. I've also tried to call hardware_reset() before starting a pipeline and after each throw but it didn't help.

MartyG-RealSense, 2020-09-19:
You can use **try_wait_for_frames()** instead of wait_for_frames() as it works similarly but returns a **False** state when timeout occurs instead of throwing an error.   The stream may still be effectively paralysed but removing the error throw may be a step in the right direction.

MikhailUskin, 2020-09-20:
Are there any functions like **hardware_reset()** that could fix the the stream if **try_wait_for_frames()** return **False**?

MartyG-RealSense, 2020-09-20:
It does not always work in these situations but there is no harm in trying hardware_reset() since it does not take long to set up in a script.  Some example C++ code for this is in the link below.

https://github.com/IntelRealSense/librealsense/issues/2161#issuecomment-408779150

MikhailUskin, 2020-09-21:
I put try_wait_for_frames() and hardware_reset() calls instead of wait_for_frames() that throws as you suggested and then during debug session I saw that _aggregator->dequeue(frame, timeout_ms) returns false beacuse underlying queue remained empty. Also, I found that _post_process_callback->on_frame((rs2_frame*)fr) is never called in sensor.cpp. So, what functions I could debug next, to reveal more about the root cause of the issue?

MartyG-RealSense, 2020-09-21:
Given that a pipeline.stop() may not work when the program is in this state, I suggested in another case about the **frame didn't arrive within 15000** recently that **sensor.stop()** could be tried as the stop instruction instead.  Then this could be followed up with a hardware_reset() after the stop if it stops the pipeline successfully.

https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103

MikhailUskin, 2020-09-21:
Seems it doesn't help. I tested it on the following code based on [#4158 (comment)](https://github.com/IntelRealSense/librealsense/issues/4158#issuecomment-505559462):

```
// License: Apache 2.0. See LICENSE file in root directory.
// Copyright(c) 2017 Intel Corporation. All Rights Reserved.

#include <librealsense2/rs.hpp> // Include RealSense Cross Platform API
#include "example.hpp"          // Include short list of convenience functions for rendering

std::pair<std::unique_ptr<rs2::syncer>, std::vector<rs2::sensor>> restart_stream()
{
	rs2::context                ctx;            // Create librealsense context for managing devices

	//std::vector<rs2::pipeline>  pipelines;
	std::unique_ptr<rs2::syncer> sync = std::make_unique<rs2::syncer>();

	std::vector<rs2::sensor> sensors;

	// Start a streaming pipe per each connected device
	for (auto&& dev : ctx.query_devices())
	{
		for (rs2::sensor sensor : dev.query_sensors())
		{
			if (sensor.get_active_streams().size() > 0)
			{
				sensor.stop();
				sensor.close();
			}

			dev.hardware_reset();
			rs2::device_hub hub(ctx);
			dev = hub.wait_for_device();
		}

		for (rs2::sensor sensor : dev.query_sensors()) 
		{
			auto sensor_stream_profiles = sensor.get_stream_profiles();

			if (auto depth_sensor = sensor.as<rs2::depth_sensor>()) {
				depth_sensor.open(sensor_stream_profiles[77]); // 640 x 480 60Hz
				depth_sensor.start(*sync);
				sensors.push_back(sensor);
			}
			else if (sensor_stream_profiles[0].stream_type() == RS2_STREAM_COLOR) { // "color sensor" doesn't have its own class
				sensor.open(sensor_stream_profiles[85]); // 640 x 480 60Hz RGB8

				sensor.start(*sync);
				sensors.push_back(sensor);
			}
		}
	}

	return std::make_pair(std::move(sync), sensors);
}

// Capture Example demonstrates how to
// capture depth and color video streams and render them to the screen
int main(int argc, char * argv[]) try
{
    rs2::log_to_console(RS2_LOG_SEVERITY_ERROR);
    // Create a simple OpenGL window for rendering:
    window app(1280, 720, "RealSense Capture Example");

	rs2::colorizer              colorizer;      // Utility class to convert depth data RGB colorspace

	auto objects = restart_stream();
	auto sync = std::move(objects.first);
	auto sensors = objects.second;

	// We'll keep track for the last frame of each stream available to make the presentation persistent
	std::map<int, rs2::frame> render_frames;

	// Main app loop
	while (app)
	{
		// Collect the new frames from all the connected devices
		std::vector<rs2::frame> new_frames;
		//for (auto &&pipe : pipelines)
		{
			rs2::frameset fs;
			if (sync->try_wait_for_frames(&fs))
			{
				for (const rs2::frame& f : fs)
					new_frames.emplace_back(f);
			}
		}

		if (new_frames.empty())
		{ 
			auto objects = restart_stream();
			auto sync = std::move(objects.first);
			auto sensors = objects.second;
		}

		// Convert the newly-arrived frames to render-firendly format
		for (const auto& frame : new_frames)
		{
			render_frames[frame.get_profile().unique_id()] = colorizer.process(frame);
		}

		// Present all the collected frames with openGl mosaic
		app.show(render_frames);
	}

	for (auto sensor : sensors) {
		sensor.stop();
		sensor.close();
	}

    return EXIT_SUCCESS;
}
catch (const rs2::error & e)
{
    std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
    return EXIT_FAILURE;
}
catch (const std::exception& e)
{
    std::cerr << e.what() << std::endl;
    return EXIT_FAILURE;
}
```

MikhailUskin, 2020-09-21:
I have another app using RealSense cameras that works fine on other machines but if I call sensor.stop() on my machine it throws: `wrong_api_call_sequence_exception("stop_streaming() failed. UVC device is not streaming!");` though 'Intel RealSense Viewer' see this camera on my machine and works fine.

MartyG-RealSense, 2020-09-21:
Are you using the short official USB cable supplied with the camera or a longer cable of your own choice, please?

MikhailUskin, 2020-09-21:
The short one

MartyG-RealSense, 2020-09-21:
Does the **frames didn't arrived within 15000 error** occur as soon as the program starts or some time later?

MikhailUskin, 2020-09-21:
Unfortunately, it occurs every call of rs2::pipeline::wait_for_frames() function. So, my app doesn't receive frames at all.

MartyG-RealSense, 2020-09-21:
I note in your test script earlier that it makes use of **syncer**.  If you are using syncer, could you try **poll_for_frames()** instead of wait_for_frames() or try_wait_for_frames().  The difference between poll and wait is that poll_for_frames() does not wait for frames to arrive, and is especially recommended when creating programs where multiple cameras will be accessed,

MikhailUskin, 2020-09-21:
I replaced wait_for_frames() with poll_for_frames() and removed restart_stream() function call each time new_frames container is empty but this didn't help

MartyG-RealSense, 2020-09-21:
Thank you very much for your patience during this diagnostic process. 

It may be useful to test with a sample program that is simpler than **rs-capture** to eliminate the possibility that the computer is being over-burdened with processing.  Could you run **rs-hello-realsense** in the SDK **Tools** folder of examples please and see whether you experience errors?

MikhailUskin, 2020-09-21:
This sample throws the same exception

MartyG-RealSense, 2020-09-21:
There was a similar case to yours recently, where the RealSense Viewer worked fine but **rs-hello-realsense** and self-created programs generated the **Frame didn't arrived within 15000** error.

https://github.com/IntelRealSense/librealsense/issues/7306

Cases where the camera works fine with some applications (such as the Viewer) but not others are problematic to diagnose, as it is more difficult to narrow the problem down to a specific cause (because the successes suggest that the hardware is not at fault).  

A RealSense user who experienced the error with rs-hello-realsense posted a simple test script:

https://github.com/IntelRealSense/librealsense/issues/4517#issuecomment-536290117

MartyG-RealSense, 2020-09-29:
Hi @MikhailUskin   Do you still require assistance with this problem, please?   Thanks!

MikhailUskin, 2020-09-30:
I'm going to dig deeper inside components responsible for communicating with drivers but a little bit later

MartyG-RealSense, 2020-09-30:
I would think that the simplest way to test the connection on Windows would be to run the simple **rs-hello-realsense** example, which just provides a text readout of measured distance without any graphics.

A pre-built version of the example can be found in the SDK's **Tools** folder.  You can find this easily on Windows by right-clicking on the desktop launch icon for the RealSense Viewer and selecting the **Open file location** option from the menu.

MikhailUskin, 2020-10-06:
Unfortunately, **rs-hello-realsense** example throws the exception we discussed earlier.

MartyG-RealSense, 2020-10-06:
Can you upgrade your camera to firmware **5.12.8.200** (the newest one) please if you have not done so already and see if it makes any difference?

JANGSOONMYUN, 2020-10-08:
@MartyG-RealSense 
Hi, I have the same problem. I followed your comments but it doesn't work.

MartyG-RealSense, 2020-10-08:
Hi @JANGSOONMYUN   I will reply at the case that you have created.   Thanks!

MartyG-RealSense, 2020-10-25:
Case closed due to no further comments received.

closed 2020-10-25 by MartyG-RealSense

Captain299792458, 2023-04-06:
after all these solutions, just changing a power source worked for me. the camera needs enough power to get the frame otherwise it also throws above error

MartyG-RealSense, 2023-04-06:
Thanks so much @Captain299792458 for sharing your experience with the RealSense community!