realsenseai/librealsense #7378: RealSense error calling rs2_pipeline_start_with_config?

Monsterkappa-dev, 2020-09-19:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   |  05.11.01.100 | 
| Operating System & Version |  Win10 64bit  |                            | 
| Platform                                 | PC |
| SDK Version                            |  2.32.1 }                          | 
| Language                            |  {C/C++ }                          | 

### Issue Description
I integreted realsense SDK to my application for grabbing frame stream. Everything works fine when i run my application in single time. But when I do the pressure test, it throw the following rs2::error:

RealSense error calling rs2_pipeline_start_with_config(pipe:09AC0E38, config:09AC0DB8): hr returned: HResult 0x800703e3

I almost start and close the camera more than 100 times in one test loop.

Anyone knows about how to fix this problem? 
Thank you

MartyG-RealSense, 2020-09-19:
Hi @Monsterkappa-dev   The case in the link below provides an example of the problems that can occur when performing rapid on-off toggling for the purposes of stress testing.  

https://github.com/IntelRealSense/librealsense/issues/6343

Aside from setting a time delay inbetween start and stop (e.g 5 seconds), other RealSense users have tried dealing with errors related to start-stop by starting the pipeline with **pipeline.start()** but stopping it with **sensor.stop()** instead of pipeline.stop().

https://github.com/IntelRealSense/librealsense/issues/7252#issuecomment-695194103

Monsterkappa-dev, 2020-09-21:
Thank you for your comment@MartyG-RealSense. I’ll try this solution

MartyG-RealSense, 2020-09-29:
Hi @Monsterkappa-dev   Do you still require assistance with this problem, please?   Thanks!

closed 2020-10-08 by MartyG-RealSense