realsenseai/librealsense #7379: Aligned depth image repeats itself

ItsMeTheBee, 2020-09-19:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |   Ubuntu 18.04 (host) & Docker Container ros:eloquent  | 
| Kernel Version (Linux Only)    |  5.4.0-47-generic                                         | 
| Platform                                 | Desktop x86_64  |
| SDK Version                            |  latest                          | 
| Language                            |  C++                        | 
| Segment			|  Robot                        | 

### Issue Description
Due to low activity on https://github.com/intel/ros2_intel_realsense I´m cross-posting here - I know you guys probably know very little about ROS and such things but maybe you´re still able to help me.
I´ll close the issue if it´s too off-topic.
 
I´m trying to segment objects by detecting them in the rgb image, cropping the depth image and then converting that image to a pcl.
Doing this I´m facing some issues and you might be able to help me with this one:

The depth image seems to repeat itself instead of actually being aligned to the color image:
[image]

This leads to issues when I try to crop and convert the depth image.
Where does this behavior come from? Is there anything I can do about it?

Code to convert the depth image msg to cv::Mat:
`cv::Mat image(m_current_depth_msg.height, m_current_depth_msg.width, CV_8UC3, const_cast<unsigned char *>(m_current_depth_msg.data.data()), m_current_depth_msg.step);`

Something I noticed is that I can´t visualize the topic "/d435/camera/aligned_depth_to_color/image_raw" in Rviz. The messages get dropped due to some unknown error. The topic "/d435/camera/depth/image_rect_raw" can be visualized just fine but here I´m facing the same issue (plus the image is obviously no aligned so I can´t really use it).

Any help would be appreciated!
Thanks in advance!

MartyG-RealSense, 2020-09-19:
Hi @ItsMeTheBee   May I first confirm that you are generating the aligned image in ROS by setting depth to color alignment to **True** (perhaps by using the term **align_depth:=true** in your roslaunch instruction)?

ItsMeTheBee, 2020-09-19:
Hey @MartyG-RealSense !

Yes!
The config file looks like this:

```
/d435:
  camera:
    ros__parameters:
      serial_no: 801212070544 # d435
      base_frame_id: d435_link
      align_depth: true
      enable_pointcloud: true
      dense_pointcloud: true
      color0:
        enabled: true
        resolution: [640,480]
        fps: 6
      depth0:
        enabled: true
        resolution: [640,480]
        fps: 6
      infra1:
        enabled: true
        resolution: [640,480]
        fps: 6
      infra2:
        enabled: true
        resolution: [640,480]
        fps: 6
```

I usually disable both infrared streams and set fps to 6 because I´ll deploy my code to a device that can´t handle more than 6fps (USB 2.0 Port and rather weak processor compared to a normal desktop).
Right now I´m still developing on my desktop so it doesn´t really matter but I´d rather leave the fps low to make deployment easier later on.

MartyG-RealSense, 2020-09-19:
I did some initial research on your case.  I have seen some cases of similar "wraparound" of the image before, where one part of the image is duplicated on another part of the image.  At this time, I believe that the code for converting the depth image to cv::mat may be the most likely cause.  

An example of image wraparound when using cv::mat code can be seen in the case in the link below.

https://github.com/IntelRealSense/librealsense/issues/1381

Within that discussion, suggestions for OpenCV format conversion codes to use are suggested here:

https://github.com/IntelRealSense/librealsense/issues/1381#issuecomment-374226200

If that does not solve your problem, I will be happy to continue the conversation when I return in 7 hours from the time of writing this.  Good luck!

amirreza7876, 2020-09-20:
dsfsdfsdfsfd

MartyG-RealSense, 2020-09-29:
Hi @ItsMeTheBee   Do you still require assistance with this case, please?  Thanks!

ItsMeTheBee, 2020-09-29:
Hey @MartyG-RealSense !
This solved it for me, I just changed CV_8UC3 to CV_16UC1.
Thanks a lot!

closed 2020-09-29 by ItsMeTheBee

MartyG-RealSense, 2020-09-29:
Great news that you found a solution - thanks for the update!