realsenseai/librealsense #7380: r200 depth stream returns zero

amirreza7876, 2020-09-20:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  R200  | 
| Operating System & Version |   Linux Ubuntu 16.04  | 
| Platform                                 | PC  |
| SDK Version                            |  legacy  1.12.1                  | 
| Language                            |  python       | 

im working with r200 realsense and want to get depth of center an aruco, but when aruco rotates, the depth stream returns zero,
here is my code:
```
markerCorners, markerIds, rejectedCandidates = cv2.aruco.detectMarkers(gray, dictionary, parameters=parameters)

            for i, corner in enumerate(markerCorners):
                idd = markerIds[i]
                centerX = (corner[0][0][0] + corner[0][1][0] + corner[0][2][0] + corner[0][3][0]) / 4
                centerY = (corner[0][0][1] + corner[0][1][1] + corner[0][2][1] + corner[0][3][1]) / 4
                cv2.circle(c, tuple(corner[0][0]), 2, (0,0,255), 2)
                cv2.circle(c, tuple(corner[0][1]), 2, (255,25,0), 2)
                cv2.circle(c, tuple(corner[0][2]), 2, (25,255,0), 2)
                cv2.circle(c, tuple(corner[0][3]), 2, (255,255,230), 2)
                
                center = (int(centerX), int(centerY))
                depth = dev.depth[int(center[1])][int(center[0])]/1000
```

MartyG-RealSense, 2020-09-20:
Hi @amirreza7876   I thought about your problem carefully.  It looks as though your project needs a means of detecting the **pose** of the Aruco (its position and rotation) so that when rotation takes place, the Aruco can continue to be tracked.

For example, a research paper in the link below that used the R200 and ROS took the approach quoted below:

******************

We set up two Intel RealSensor R200 camera by 90 degree as a stereo camera.  A marker tag is attached to robot arm (da Vinci Patient Side Manipulators, PSM).  The marker’s pose relative to camera can be detected, and the robot pose is reported by dVRK.  From these two poses, the transformation between cameras and robot base is obtained through hand-eye calibration. 

Then a phantom is set up, and the four corners of the phantom will be automatically detected in the images of two camera. 
 The 3D position can be calculated to register the phantom to the robot.

**************

The paper can be viewed at this link:

https://ciis.lcsr.jhu.edu/lib/exe/fetch.php?media=courses:446:2017:446-2017-12:group12_final_report.pdf

My research indicated that numerous projects, including the research paper above, have made use of a ROS package called **aruco_ros**:

https://index.ros.org/r/aruco_ros/github-pal-robotics-aruco_ros/

MartyG-RealSense, 2020-09-29:
Hi @amirreza7876   Do you still require assistance with this problem, please?   Thanks!

closed 2020-10-08 by MartyG-RealSense