realsenseai/librealsense #7384: Where to buy filters for D400 Intel RealSense Cameras? Problems with depth, IR, and 3D projection mapping AR

Sensebellum, 2020-09-21:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 } | 
| Firmware Version                   | (05.12.07.100) | 
| Operating System & Version |   {Win 10  |                                   | 
| Platform                                 | Intel NUC Hades Canyon  |
| SDK Version                            |  {N/A}                          | 
| Language                            |  { ALL / Processing }                          | 
| Segment			|  { AR }                          | 

### Issue Description

Where can you buy filters for the Intel RealSense D400 cameras?

I read a lot of white papers, a lot of comments, but almost no links to buy.

My problem is with 3D projection mapping. I am projecting on the surface I am reading depth from. For some reason the "IR" stream is not actually IR and lets in visible light. Not sure why that is a good idea... Anyways, I need to block out the visible light from the IR stream so the projections themselves are not interfering with the tracking. I have never experienced a depth camera having this problem.

For instance I can confirm using a true IR USB camera that the projections do not show in the IR spectrum so it's not the projector's fault. The Intel RealSense D435 is showing visible light in the IR. What?!?

Luckily you can put a filter in place, right...? Well, I'll be honest it's kind of hard to find them? Why not put them for sale with the cameras themselves on the store? Why make the customer search through crazy technical product brochures to find some weird filter that takes 10 emails and 2 weeks to arrive? I see listings on other issues about Edmund optics, Astra optics, etc. HUGE catalog. No time. Need to make it easier.

So... I just bought this one. Who knows how I will attach it to the camera... 

[https://www.amazon.com/Quanmin-9-6mm%C3%971-0-Optical-Infrared-Modified/dp/B07KQWKYD3/ref=sr_1_1?dchild=1&keywords=ir+long+pass+filter&qid=1600703244&sr=8-1](url)

I also bought this one

[https://www.amazon.com/Quanmin-6mm%C3%976mm%C3%970-55mm-Infrared-Bandpass-Recognition/dp/B07C6TK3DN/ref=sr_1_5?dchild=1&keywords=ir+long+pass+filter&qid=1600703410&sr=8-5](url)

Is there a way to do this in software? Any settings I can change? Any specific listing for filters to buy?

I truly need to long pass IR as documented in this paper it just has no product listing.
[https://dev.intelrealsense.com/docs/optical-filters-for-intel-realsense-depth-cameras-d400](url)

MartyG-RealSense, 2020-09-21:
Hi @Sensebellum   The Chief Technical Officer of the RealSense Group at Intel (**agrunnet**)  provides suggestions for suppliers for filters for use with RealSense in the link below:

https://github.com/IntelRealSense/librealsense/issues/2875#issuecomment-446793681

Also, here is a source for thin-film linear polarizers, under the category **Linear Polarizers > Thin-Film Polarizers**:

https://www.edmundoptics.co.uk/c/thin-film-polarizers/1338/

agrunnet has said, "Any polarizer will do that is linear.  The circular polarizers used most often in 3D glasses won’t work".

You can also create your own front-covers ("cover materials" for the 400 Series cameras.  The data sheet for the 400 Series cameras contains details of specifications for doing so, and agrunnet comments on the subject here:

https://github.com/IntelRealSense/librealsense/issues/2566#issuecomment-431852571

You may find it easier to 3D-print a custom casing design and embed one of the RealSense **Depth Module Kits** (caseless PCB versions of the cameras) into it, along with your chosen filter. 

In regard to IR: the 400 Series cameras do not have an IR-Cut filter on their IR imager components and so can see all visible light frequencies.  The T265 Tracking Camera in comparison does have an IR-Cut filter (in order to enable it to be paired with a 400 Series camera and its IR dot pattern projector) but does not have depth sensing capability built in (it can perform passive depth-sensing using software and the fisheye imagers).  

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/t265_stereo.py

Sensebellum, 2020-09-21:
Thank you for the thorough review.

I found one work around is to dim the projector to only 25% brightness. It is barely legible but does not interfere as much. Really wish there was a simple way around this.

I have read the comments from to CTO. It is great they responded and are involved however the l;istings were vague. Big catalogs and lots of expensive options.

I appreciate you listing one further. Yet, as you'll note they are over $300 and ship from the UK.

I do believe I will go the 3D printed route. Thank goodness for 1 day shipping as it looks like those filters will be here tomorrow.

Do both IR cameras need the exact same filter?

Looking towards the future are there any Intel RealSense cameras, that work with this library, that have real IR not a greyscale combination of all frequencies?

Thank you for the help!

agrunnet, 2020-09-21:
Please make sure that you make identical changes to left and right images. The stereo algo compares left and right images so we want them to look at identical as possible.

MartyG-RealSense, 2020-09-21:
@agrunnet   Thanks so much for the advice!

You are very welcome @Sensebellum   If you need to reduce the visibility of the projector's dot pattern then you can do so by reducing the value of the **Laser Power** option, as dot visibility reduces as Laser Power reduces and becomes more visible as Laser Power is increased.   On the D415 model, you can also use a Visual Preset called **Left Imager Color w/o IR Pattern** to remove the dot pattern.  This feature is also coming to the new D455 model in an upcoming firmware release.

https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets#preset-table

For inspiration, you can find examples of community created 3D printer blueprints for RealSense camera housings, mounts and other accessories by visiting model library sites such as Thingiverse and GrabCAD and searching for **realsense**.  Intel also provide a page of official RealSense CAD model files.

https://dev.intelrealsense.com/docs/cad-files

The links below may also be useful references.

https://github.com/IntelRealSense/librealsense/issues/3370
https://github.com/IntelRealSense/librealsense/issues/1070

You can find a feature comparison of the full RealSense range in the guide below:

https://www.intelrealsense.com/which-device-is-right-for-you/

Sensebellum, 2020-09-22:
@MartyG-RealSense It's not about reducing the dot pattern it's about being able to see it all. Once again the projections from the real life projector (not your dot pattern) is what is causing the issue since the IR feed isn't really IR.

Even on the cheapest Kinect this isn't an issue. IR means IR there. Not here apparently.

Still makes no sense why IR Feed would be a combination of all frequencies. Isn't that why you have an RGB camera too?

Anyways, I really hope these filters from Amazon fix the issue. $15 and one day shipping.

If I could make a suggestion it would be:

A) Have the option to only look at IR in software (real IR, only, not this color image addition thing).

B) Sell the filters on your website WITH a mount. This is much better than instructing your customers to some random UK website where the filters (who knows if they are even the right ones) cost over $300 and take forever to get here. As Intel you should know this is about rapid prototyping and 2 weeks for a filter to arrive for $300 is anti-rapid to the nth degree.

I will report back if there is a solution.

MartyG-RealSense, 2020-09-22:
@Sensebellum   On the 40 Series cameras, the projector is a separate component from the IR imager components.  There is the option to set the **IR Emitter** option to Off and still have the IR active without the dot pattern and light that the projector provides.

You can test this in the RealSense Viewer by setting the Emitter Enabled drop-down menu in the **Stereo Module** section of the Viewer's options side-panel to 'Off' to turn off the projector or to 'Laser' to have it enabled.

**Off**

[image]

**Laser**

[image]

It should be noted that whilst the 400 Series cameras do not experience interference with each other, they can interfere with non-RealSense IR devices (e.g the built-in IR camera of a laptop).  

The RealSense support forums are not involved in the operation of the RealSense store, so I cannot comment on their stock policies.

Sensebellum, 2020-09-22:
@MartyG-RealSense Thanks for the share on your computer I appreciate the efforts here.

Really I think the only saving grace would be to disable the RGB visible portions of the spectrum from being integrated into the IR feed. Turning off the laser dots pattern only makes things worse as then it is only getting depth from the images the video projector is putting out). However, it does not look like disabling the visible light integration is possible from software.

So, I hope the IR cut lenses arrive today. Else, I was able to set some weird settings with turning off auto exposure, cranking up the dot pattern power, and dimming my video projector Finally working. Yet, I know it can be better!

Thank you for all of the help I will update as the IR cut lenses arrive.

Best,

MartyG-RealSense, 2020-09-22:
If the scene is well lit then the dot pattern is not required, as the 400 Series camera can use the ambient light in the scene instead of analyze surfaces for depth detail.  

Depth frames are constructed inside the camera hardware from the left and right infrared images, which is why it is a Stereo depth camera.  You can stream infrared on its own with Depth and RGB streams turned off though,.  You can also stream Depth and just one of the two Infrared streams, and Depth will not be affected.  Basically, Depth is still generated whether you have just one IR stream active or no IR stream active.

Sensebellum, 2020-09-23:
Lens filters worked!

To anyone with this similar problem here is the product listing and the .STL so you can print this yourself.

Here are pictures showing the problem and solution.
[image]
[image]
[image]
[image]
[image]
[image]

$15 lenses shipped in 1 day, modeled and printed in an afternoon. Problem solved. See pictures.

Intel, please consider selling these on your site. It would be easy. Sending people to a UK lens site with $300 lenses is just not the best.

Download link for .STL

https://drive.google.com/file/d/11wcv_wmW6_j8ZHiLnXbETPAlAMVqIAzl/view?usp=sharing

MartyG-RealSense, 2020-09-23:
Thanks so much for sharing the details of your solution with the RealSense community!

MartyG-RealSense, 2020-10-02:
Hi @Sensebellum   Do you still require assistance with this case, please?   Thanks!

Sensebellum, 2020-10-02:
We are good to go thanks for the great camera!

MartyG-RealSense, 2020-10-02:
Great news!   I will close this case then.  Please feel welcome to open a new case if you have future questions.  Good luck!

closed 2020-10-02 by MartyG-RealSense

Aigric-nwc, 2021-04-15:
Hi @Sensebellum  
The link to the stl file you uploaded has expired. Can you share it again?

debOliveira, 2021-07-16:
> Hi @Sensebellum
> The link to the stl file you uploaded has expired. Can you share it again?
> or Send to my E-mail : [dc3059@gmail.com](mailto:dc3059@gmail.com)

Did you managed to get the .stl? I am searching for a D400 series lens case.
Maybe you still have the file @Sensebellum ? Don't know if you unintentionally deleted it.

Sensebellum, 2021-07-16:
Hello! This should be a more lasting link.

You can find the model on the Sensebellum Sketchfab and download the .stl

[Intel RealSense D435 Infrared Clip 1.0](https://sketchfab.com/3d-models/intel-realsense-d435-infrared-clip-10-4fd4a3ef04d7456a8248b731637d6759)

[image]

taihup, 2021-11-18:
@Sensebellum Do you have an infrared clip for the Intel Realsense D455?

MartyG-RealSense, 2021-11-18:
Hi @taihup  In the absence of availability of a 3D printed holder for external filters on a D455, one RealSense user simply held their filter in place with tight bands.  It is not attractive looking but may be worth considering as a temporary measure until a 3D printing pro such as @Sensebellum creates a D455 3D printing model.

[image]

Sensebellum, 2021-11-18:
> @Sensebellum Do you have an infrared clip for the Intel Realsense D455?

Yes you can download from the link provided. Enjoy!

[https://sketchfab.com/3d-models/intel-realsense-d435-infrared-clip-10-4fd4a3ef04d7456a8248b731637d6759](https://sketchfab.com/3d-models/intel-realsense-d435-infrared-clip-10-4fd4a3ef04d7456a8248b731637d6759)

taihup, 2021-11-18:
> Yes you can download from the link provided. Enjoy!

Nope, that is for D435i. I am looking for D455 model.

Sensebellum, 2021-11-18:
Okay well you may need to make your own.

I used Fusion 360. Pretty easy!

Best of luck!

jamesheatonrdm, 2023-12-05:
Hi @CaseyJScalf you mentioned you used a 'true' IR camera to confirm the output. Would you be able to share the make/model of this particular camera please? Thanks in advance