realsenseai/librealsense #7386: Private member variable in Intrinsic class in Android Wrapper

YichaoW, 2020-09-22:
Hi,

I noticed that all member variables in intrinsic class are private and they don't have getter except for width and height. Is there any other way to get intrinsic value such as ppx and ppy?

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/android/librealsense/src/main/java/com/intel/realsense/librealsense/Intrinsic.java

MartyG-RealSense, 2020-09-22:
Hi @YichaoW    There is a very long and detailed discussion about reading instrinsics in Android, so I will link to the part of the discussion that I think may be of particular interest to you, as it discusses printing the value of PPX and PPY in the Android log.  You can of course return to the start and go through the entire discussion.

https://github.com/IntelRealSense/librealsense/issues/6522#issuecomment-639152764

YichaoW, 2020-09-22:
Thanks @MartyG-RealSense ! Is there any java solution? I am not very familiar with C++ and the NDK wrapper.

MartyG-RealSense, 2020-09-22:
Hi again @YichaoW   Intrinsics and extrinsics were added to the Android wrapper in SDK **2.37.0**.

https://github.com/IntelRealSense/librealsense/issues/4580
https://github.com/IntelRealSense/librealsense/issues/6915

YichaoW, 2020-09-22:
@MartyG-RealSense I knew they were added. But the variables stored in Intrinsics are private and I cannot access them.

MartyG-RealSense, 2020-09-23:
A "pull" was submitted a week ago to add intrinsic private property getters to the Android wrapper.  The submission has passed build checks but it looks as though it is not merged into the SDK yet.

https://github.com/IntelRealSense/librealsense/pull/7342

YichaoW, 2020-09-25:
Got it thanks

closed 2020-09-25 by YichaoW

electro-logic, 2023-04-25:
Hello,
I can't get a StreamProfiles of type VideoStreamProfile using v2.50 on Android 9.

            var sensors = device.QuerySensors();
            foreach (var sensor in sensors)
            {                
                foreach (var sp in sensor.StreamProfiles)
                {
                    // sp can't cast to VideoStreamProfile 
                }
            }

I am trying to get the intrinsic.

electro-logic, 2023-04-26:
Any chance to get a Java compatible API?

MartyG-RealSense, 2023-04-26:
There is not an official Java wrapper or plans to create one, but a RealSense user created an unofficial Java wrapper several years ago.

https://github.com/cansik/librealsense/tree/master/wrappers/java

electro-logic, 2023-04-26:
Hello, I'm on Android and I'm using the official library binding with Java, but I can't access the Intrinsic api

MartyG-RealSense, 2023-04-26:
https://github.com/IntelRealSense/librealsense/pull/6594 states that the API instruction for accessing intrinsics on Android is **rs2_get_video_stream_intrinsics**

electro-logic, 2023-04-26:
Hello,
From my understanding (https://github.com/IntelRealSense/librealsense/blob/b874e42685aed1269bc57a2fe5bf14946deb6ede/src/android/jni/stream_profile.cpp#L151) the API rs2_get_video_stream_intrinsics is not directly exposed, but can be used by calling GetIntrinsic() on a VideoStreamProfile instance. The problem I'm facing is that there is no way to get that instance (see my first post).

MartyG-RealSense, 2023-04-26:
I will tag @remibettan into this discussion in case they can provide insight about your problem described at https://github.com/IntelRealSense/librealsense/issues/7386#issuecomment-1522519745