realsenseai/librealsense #7389: Get intel D4XX ASIC temperature and Projector temperature

dingbaiyin, 2020-09-22:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D430/D435| 
| Firmware Version                   | 5.12.7.1| 
| Operating System & Version |  ubuntu  | 
| Kernel Version (Linux Only)    | 4.19.66-v7                                | 
| Platform                                 |Raspberry Pi4 |
| SDK Version                            |  2.36.0             |
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
< I want to get D430 or D435 Projector temperature and ASIC temprerature,but I failed.
I referred to these two links
https://github.com/IntelRealSense/librealsense/issues/3888
https://github.com/IntelRealSense/librealsense/issues/866#issuecomment-357461253
this is my code:
eg 1:
void intelCameraControl::GetCameraTemperature()       
{
	std::vector<rs2::sensor> sensors = selected_device.query_sensors();
	auto  depth_sensor = selected_device.first<rs2::depth_sensor>();
	if (depth_sensor.supports(RS2_OPTION_ASIC_TEMPERATURE))
	{
		float tmpASIC = depth_sensor.get_option(RS2_OPTION_ASIC_TEMPERATURE);

		cout << "tmpASIC=" << tmpASIC << endl;
	}
	else
		cout << "error" << endl;
	return;
}
eg 2:
void intelCameraControl::GetCameraTemperature()     
{
	
	auto dbg = selected_device.as<rs2::debug_protocol>();
	std::vector<uint8_t> cmd = { 0x14, 0, 0xab, 0xcd, 0x2a, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0 };

	auto res = dbg.send_and_receive_raw_data(cmd);
	int temp = res[4];
	std::cout << "Projector 温度为  "<<temp << std::endl;
	
	return;
}
Application method eg 1, I cannot get the ASIC temperature and Projector temperature. Using method eg2, I can get the Projector temperature. But how to get the ASIC temperature. ASIC temperature is very important for me to test the camera, please give some suggestions, thank you.

>

MartyG-RealSense, 2020-09-22:
Hi @dingbaiyin  The link below provides information for reading both ASIC and projector temperatures with C++.

https://github.com/IntelRealSense/librealsense/issues/6735

I see that you have already viewed the https://github.com/IntelRealSense/librealsense/issues/866 case about reading the temperatures in Python.  You can also find Python information for temperature reading in the official Python documentation for the SDK:

https://intelrealsense.github.io/librealsense/python_docs/search.html?q=temperature&check_keywords=yes&area=default

dingbaiyin, 2020-09-23:
Sorry I submitted the question repeatedly.
I still cannot read the ASIC temperature successfully. 
I read the this link.I saw that this user successfully read the ASIC temperature. I increased the delay after obtaining the frame according to his method, and then read the ASIC temperature, but it still failed.
https://github.com/IntelRealSense/librealsense/issues/6735
Can you provide a C++ method to read the ASIC temperature?

MartyG-RealSense, 2020-09-23:
All of the information sources that I know of about the subject of reading ASIC temperature have already been provided above, unfortunately and I could find no other useful references.

dingbaiyin, 2020-09-24:
Thank you very much for your help, I hope this problem can be solved in the fu help, I hope this problem can be solved in the future update.
In my test, I found that an error occurred when getting the Projector temperature.
The error message is "terminate called after throwing an instance of'rs2::backend_error' what(): set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
"An error occurred when collecting 700-900 images, and the frequency was about 4 seconds.
Get the Projector temperature every time a picture is collected.
 Is The reason that the acquisition frequency is too high？. I want to acquire the Projector temperature in real time. How can I avoid this problem?

MartyG-RealSense, 2020-09-24:
I researched your latest question but could not find any useful additional information again unfortunately.

If you are reading the temperature for the purposes of testing the camera then there is a simpler test.  Touch the camera casing with your hand and if it is hot to the touch then the temperature of the camera is too hot.

Also, the SDK has a laser safety mechanism.  If the projector temperature exceeds 60 degrees C for several seconds then the Laser Power will be halved.  If the projector temperature continues to be excessive for several seconds after that then the laser is shut off.

dingbaiyin, 2020-09-27:
Yes, my camera did experience a situation what the temperature was too high and the laster shut off, so I need to monitor the Projector and ASIC temperature.If there is a new way to get the ASIC temperature, I will try it. I will close this issue

closed 2020-09-27 by dingbaiyin

MartyG-RealSense, 2020-09-27:
Thank you very much for the update.