realsenseai/librealsense #7390: Intel D455 notdetected on Raspberry Pi 4

sub400, 2020-09-22:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.12.7.100| 
| Operating System & Version |  Linux (Raspbian Buster)| 
| Kernel Version (Linux Only)    |  5.4.65-v7l+ #1341                                        | 
| Platform                                 | Raspberry Pi|
| SDK Version                            |  2.38.1 | 
| Language                            |  C                          | 
| Segment			|  others                          | 

### Issue Description
On Raspbery Pi the D455 Sensor is not detected. I built latest SDK and Realsense Viewer is running. It can detect/stream D415 and D435 sensors on same machine but no D455. D455 running fine on Windows 10.

Tried to update raspi firmware as suggested in issue #7312 

Any ideas?

Thx

MartyG-RealSense, 2020-09-22:
Hi @sub400  Can you confirm to me please that the D455 camera is not detected in the RealSense Viewer but D415 and D435 are detected.  Also, could you tell me whether or not you are using the SDK's Python wrapper?

sub400, 2020-09-22:
Excatly, D415 and D435 are working fine in Realsense Viewer on this Raspberry Pi. D455 ist not detected. 

I'm not using Python in this project, but C++. In my custom solution, the same issue. D415 and D435 OK, D455 not detected.

MartyG-RealSense, 2020-09-22:
Thank you very much.  Which method did you use to install the SDK please:

- Packages
- Source code with kernel patching
- Source code without patching using the RSUSB method

sub400, 2020-09-22:
I built the SDK using the following instructions: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

I have to mention that I built the SDK on a different Raspi than the one I'm using for testing. I then copied the fresh libraries to the second machine by hand, which always worked until I tried the D455.

MartyG-RealSense, 2020-09-22:
There was a case a couple of months ago where a Raspberry Pi 4 was set up using a pre-made SD card image of a RealSense build provided by Intel.  It worked fine, but when the SD card was copied to test the image on another Pi, the copied image performed much worse.  

https://support.intelrealsense.com/hc/en-us/community/posts/360048495493--Intel-Realsense-D435-with-RaspberryPi-Best-practice-installation-guide

If you are using the SDK in Raspbian Buster on your Pi 4 though then I would recommend a specialist installation guide.  If you are not going to be installing additional wrappers such as Python or ROS, the **Acrobotic** guide is a popular one that is similar in content to Intel's official installation guide for Pi 3.

https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435

If you do plan to use wrappers later on then if you use the SDK with the Ubuntu Server OS on the Pi 4 instead of Raspbian Buster then the installation process should be more straightforward.

RealSenseSupport, 2020-09-23:
@sub400 I just tried D455 on my Raspberry Pi setup and it can be detected and start streaming successfully. I suspect your issue was caused by just copied the libraries and you might need apply udev rules by ./scripts/setup_udev_rules.sh just as described in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md#prerequisites . Or you can also refer to MartG's suggestion to fully install librealsense on it. Thanks!

sub400, 2020-09-23:
Thanks, but of cource I deployed the udev rules, but no success.

But what is the best way to deploy the realsense libraries to a new device, in particluar for raspi (as there are no prebuilt binaries to install via apt)? I don't want to build it on every new device from source.

MartyG-RealSense, 2020-10-05:
Hi @sub400   Do you still require assistance with this case, please?  Thanks!

sub400, 2020-10-05:
I rebuilt everything on a fresh raspberry pi and then d455 sensor was detected correctly.

But still I need a way to deploy everithing to another device without rebuilding.

MartyG-RealSense, 2020-10-05:
There was a case in June 2020 where a RealSense user installed librealsense on their Pi 4 using the Acrobotic guide mentioned earlier in this discussion and then copied the SD card to test the image on another computer.  

The RealSense user reported that for some reason the copied image executed code 5 times slower than the original.  This may not necessarily occur with you too if you attempted to copy an SD card imae of your Pi for use on another computer, but it is something to bear in mind.

https://support.intelrealsense.com/hc/en-us/community/posts/360048495493--Intel-Realsense-D435-with-RaspberryPi-Best-practice-installation-guide

closed 2020-10-20 by MartyG-RealSense