realsenseai/librealsense #7392: RealSense D435 Getting started Windows

karlita101, 2020-09-22:
Hi there, apologies for the simplicity of the question.

I am quite new to all of this and not clear on what the difference is between the viewer and the SDK?

I would like to start with the examples, but not sure what I need to get set up: I have the SDK WIN 10, the viewer and running visual studio.

When I attempted the Hello Real Sense example

- it says to use vcpkg for the #include **<librealsense2/rs.hpp> // Include RealSense Cross Platform API**

is this what should be used?

MartyG-RealSense, 2020-09-22:
Hi @karlita101  The RealSense SDK is the development platform that RealSense applications can be created in.  The RealSense Viewer is an application created with the RealSense SDK that can be used as a user-friendly tool to access, visualize, alter and export camera data without requiring any programming knowledge.

If you are using Windows and have installed the full RealSense SDK from the **Releases** web-page using the 793 mb Windows installer file (the easiest way to install the SDK on Windows) then you can immediately access a set of pre-built launchable .exe versions of SDK examples.  The installer file for Windows 10 is in the **Assets** file-list at the bottom of the information listing for each SDK version number.

https://github.com/IntelRealSense/librealsense/releases/

If you installed the SDK with the installer file and had launch icons for the **RealSense Viewer** and the **Depth Quality Tool** placed on your Windows desktop by the installer then you can right-click on one of those launch icons and select **Open file location** from the menu to open the folder window where the executable sample programs are located.

The full address on Windows of the Tools folder location is: 
**C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools**

If you would like to access the examples through Visual Studio, Intel have a guide for this process in the link below:

https://www.intel.co.uk/content/www/uk/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html

karlita101, 2020-09-22:
> Hi @karlita101 The RealSense SDK is the development platform that RealSense applications can be created in. The RealSense Viewer is an application created with the RealSense SDK that can be used as a user-friendly tool to access, visualize, alter and export camera data without requiring any programming knowledge.
> 
> If you are using Windows and have installed the full RealSense SDK from the **Releases** web-page using the Windows installer file (the easiest way to install the SDK on Windows) then you can immediately access a set of pre-built launchable .exe versions of SDK examples. The installer file for Windows 10 is in the **Assets** file-list at the bottom of the information listing for each SDK version number.
> 
> https://github.com/IntelRealSense/librealsense/releases/
> 
> [image]
> 
> If you installed the SDK with the installer file and had launch icons for the **RealSense Viewer** and the **Depth Quality Tool** placed on your Windows desktop by the installer then you can right-click on one of those launch icons and select **Open file location** from the menu to open the folder window where the executable sample programs are located.
> 
> The full address on Windows of the Tools folder location is:
> **C: > Program Files (x86) > Intel RealSense SDK 2.0 > Tools**
> 
> If you would like to access the source-code of the examples through Visual Studio, Intel have a guide for this process in the link below:
> 
> https://www.intel.co.uk/content/www/uk/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html

Perfect, thanks!

**Question1: ** is that the same as described here: 
https://dev.intelrealsense.com/docs/compiling-librealsense-for-windows-guide

**Question 2**: Would it work for VS 2019?

***Question 3** Is there a more in-depth (step by step) user manual/guide that I can refer to, with what it takes to set up the camera.

Thank you

MartyG-RealSense, 2020-09-22:
1.  Compiling on Windows is a longer way of achieving the same goal as using the installer file.  for example, selecting the **BUILD_EXAMPLES** and **BUILD_GRAPHICAL_EXAMPLES** tick-boxes in the CMake interface builds the sample programs and tools.  BUILD_GRAPHICAL_EXAMPLES builds programs that make use of OpenGL graphics, such as the Viewer and Depth Quality Tool.  

In regard to the section of the instructions that discusses enabling **metadata**, you can now do this from within the RealSense Viewer instead.

2.  VS2019 has been used to compile librealsense successfully, yes.  Here is an example report of success:

https://github.com/IntelRealSense/librealsense/issues/6219

3.  Intel have an excellent camera tuning guide as the first link on their list of **white paper** documents that provide expert guidance on a wide range of RealSense subjects.

https://dev.intelrealsense.com/docs/whitepapers

karlita101, 2020-09-22:
Hi Marty,

What is the difference then between the method you [noted above](https://www.intel.co.uk/content/www/uk/en/support/articles/000030463/emerging-technologies/intel-realsense-technology.html)

and this [one ](https://github.com/IntelRealSense/librealsense).

Why is it that it isn't registered in the first place? I can the **librealsense2** under the SDK folder

[image]

MartyG-RealSense, 2020-09-22:
The second link represents the source code of librealsense in a GitHub **repository**.

The first link explains that you can **clone** a repository (make a copy of it). You can then **build** the code in that copied repository into **binaries**.

Regarding the question about registering, can you explain more about your question please?

karlita101, 2020-09-22:
For the Hello Real Sense example: https://dev.intelrealsense.com/docs/rs-hello-realsense

[image]
 
We include _librealsense2_ however when I try to do this one Visual Studio it does not work ( even though I do have the folder present).

MartyG-RealSense, 2020-09-22:
I would personally recommend using the librealsense GitHub version of the examples, because then you have the source code at the top of the page of each example.  The tutorials pick out sections from the code to highlight instead of going through the script line by line, so you can't make a working script by pasting the lines from the tutorial together.  You should refer to the .cpp file at the top of the page (e.g **rs-hello-realsense.cpp**).

https://github.com/IntelRealSense/librealsense/tree/master/examples

I am off-shift now.  If you leave a comment below, I will be happy to reply when I return in 8 hours from the time of writing this.

MartyG-RealSense, 2020-10-02:
Hi @karlita101   Do you still require assistance with this case, please?   Thanks!

karlita101, 2020-10-10:
No, it worked out, thank you !

MartyG-RealSense, 2020-10-10:
Great news - thanks for the update!

closed 2020-10-10 by MartyG-RealSense