realsenseai/librealsense #7395: get.distance RuntimeError - "out of range value for argument "x/y""

coopawhite, 2020-09-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415} | 
| Firmware Version                   | RSV v2.38.1, PRS2 v2.38.1 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 5.3.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.38.1 }                          | 
| Language                            |  Python                          | 

### Issue Description
Hi everyone!

I am currently having issues with my x and y pixel co-ordinates that are being parsed to my get_distance function. The x and y pixel co-ords are the co-ords of the center of a bounding box so from my understanding as long as those co-ords are within the frame dimensions it should be ok?

Here is a the snippet of code:
```

       if len(resultant)>0:
            focusedIndex = resultant.index(min(resultant)) #this returns the index of the object that we want to focus
            focusX, focusY = posVal[focusedIndex] #get the centre co-ords for the focused object
            rho, phi = cart2pol(focusX,focusY) #convert to polar co-ords

            #draw a line on the image from the centre of the frame to the centre of the focused box
            cv2.line(focus, (focusX+640,focusY+360), (640,360), (0,0,0)) 

            #draw the focused box on the image
            x, y, w, h = boxes[focusedIndex]
            label = str(LABELS[class_ids[focusedIndex]])
            color = colors[1]
            cv2.rectangle(focus, (x, y), (x + w, y + h), color, 2)

            try:
                xpixel = int(x + w/2)
                ypixel = int(y + h/2)
                depth = aligned_depth_frame.get_distance(xpixel, ypixel)
                depth_point = rs.rs2_deproject_pixel_to_point(depth_intrin ,[xpixel, ypixel], depth)
                print(depth_point)
                pass
            except RuntimeError as err:
                print(err)
                pass
            
```

This issue mainly affects the left hand side of the frame (in regards to the frame viewer). 

I have also attached a GIF of the program running to give you an idea of where the error occurs.

[image]

MartyG-RealSense, 2020-09-23:
Hi @coopawhite   My recollection of such object detection examples is that the value highlighted on the bounding box refers to the confidence of the correct  identification of the object (recognizing as a banana in this case), since the XYZ values are provided in the side-panel.

So as you say, the main issue seems to be with the **out of range value for argument** errors.  I note on your GIF that there seems to be some duplication of the upper part of the image on the bottom of it, so that there appears to be two hands moving.  Is this present on the original image please or is it just on the GIF?  If it is on the original image then I can imagine how it could disrupt the reading of the coordinates from the image.

coopawhite, 2020-09-23:
Hi Marty, Thanks for the response!

You are right in stating that the value is the confidence of said
detection. However the values that I am parsing to the get.distance
function are the x and y center co-ordinates of the bounding box.

Also, in regards to the second image. This can be disregarded as I run many
different algorithms on the frame separately however to view their outputs
I concate them into a single window for easy viewing. So the only frame
data that is parsed to the function is the single depth frame.

MartyG-RealSense, 2020-09-23:
Once the error appears, I would suggest immediately letting go of the banana and withdrawing the hand from the image to see whether the tracking recovers.  This may suggest whether the hand is causing disruption, such as the bunch of fingers being mis-identified as a bunch of bananas and causing the program confusion.

coopawhite, 2020-09-23:
I have tested your hypothesis - I do not believe that this is the issue.

The reason for my certainty is that if an object is not detected, the rest of the program does not run.

I have also made sure that the depth and colour streams are the same dims and that they are aligned. 

From my understanding of the function, as long as the parsed pixels meet the criteria that 0 < x < w and 0 < y < h the function should work? However it seems that the only valid inputs are in the top right hand quadrant of the image frame.

Do you have any other ideas of what could be causing this issue? I have attached another image where I print the pixel values before it attempts the function to validate that the pixels are in fact valid.

[image]
[image]

coopawhite, 2020-09-23:
**UPDATE: IF YOU HAVE THIS PROBLEM PLEASE READ THE WHOLE COMMENT BELOW**

So as a test I disabled my alignment of the depth and colour frames (see the code I was running below)
```
# Create an align object

align_to = rs.stream.depth
align = rs.align(align_to)

dec_filter = rs.decimation_filter()   # Decimation - reduces depth frame density
spat_filter = rs.spatial_filter()          # Spatial    - edge-preserving spatial smoothing
temp_filter = rs.temporal_filter()    # Temporal   - reduces temporal noise
try:
    while True:

        start = time.time() #We want to see how long each iteration takes

        frames = pipeline.wait_for_frames()
        #aligned_frames = align.process(frames)
       
        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        aligned_depth_frame = dec_filter.process(aligned_depth_frame)
        aligned_depth_frame = spat_filter.process(aligned_depth_frame)
        aligned_depth_frame = temp_filter.process(aligned_depth_frame)
        aligned_depth_frame = aligned_depth_frame.as_depth_frame()
        depth_image = np.asanyarray(aligned_depth_frame.get_data())

        color_frame = frames.get_color_frame()
        color_image = np.asanyarray(color_frame.get_data())
```

and replaced it with a basic non-aligned stream:

```
# Start streaming
profile = pipeline.start(config)

# Getting the depth sensor's depth scale (see rs-align example for explanation)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

try:
    while True:

        start = time.time() #We want to see how long each iteration takes

        frames = pipeline.wait_for_frames()
      
        color_frame = frames.get_color_frame()
        color_image = np.asanyarray(color_frame.get_data())

        depth_frame = frames.get_depth_frame()
        depth_image = np.asanyarray(depth_frame.get_data()) 

```

and the program worked correctly (get.distance is able to return a value for the whole range of pixels).

So i tried to apply the filters and I started to receive the same problem as I was experiencing. So this leads me to believe that the error is actually my application of the filters onto the depth frame!

Please see below my final stream config code that makes me not receive this error.

```
#set up stream
pipeline = rs.pipeline()
config = rs.config()

config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
config.enable_stream(rs.stream.depth, 1280, 720, rs.format.z16, 30)

# Start streaming
profile = pipeline.start(config)

# Getting the depth sensor's depth scale (see rs-align example for explanation)
depth_sensor = profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()
print("Depth Scale is: " , depth_scale)

# Create an align object
# rs.align allows us to perform alignment of depth frames to others frames
# The "align_to" is the stream type to which we plan to align depth frames.
align_to = rs.stream.depth
align = rs.align(align_to)

try:
    while True:

        start = time.time() #We want to see how long each iteration takes

        frames = pipeline.wait_for_frames()
        aligned_frames = align.process(frames)

        color_frame = frames.get_color_frame()
        color_image = np.asanyarray(color_frame.get_data())

        depth_frame = aligned_frames.get_depth_frame()
        depth_image = np.asanyarray(depth_frame.get_data())

        if not depth_frame or not color_frame:
            continue
```

The alignment still makes a little of the left hand of the screen a bit unusable however the rest of the pixel are now valid!

Thanks for your help @MartyG-RealSense.

MartyG-RealSense, 2020-09-24:
Awesome news @coopawhite - thanks so much to you too for sharing your solution with the RealSense community   :)

MartyG-RealSense, 2020-10-02:
Case closed due to no further comments received

closed 2020-10-02 by MartyG-RealSense

arhitiron, 2023-11-22:
In my case it was a shameful error - I mixed up x and y :)