realsenseai/librealsense #7396: Can l515 be used in industrial buildings with fog？

Zhang-Qinghui, 2020-09-23:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

RealSenseSupport, 2020-09-23:
Hi @Zhang-Qinghui 

I want to point you to some documentation around the L515 camera in order for you to determine if the camera fits the environment you are looking at:

https://www.intelrealsense.com/download/13061/

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

Have a look at both of these locations and read through the material.  This should give you a better understanding of the L515 camera.

Zhang-Qinghui, 2020-09-24:
Thanks for your reply. In the document you posted, I didn't see any information about whether L515 can be used in a foggy environment. I just want to know whether the L515 can be used in a foggy environment. Looking forward to your reply.

RealSenseSupport, 2020-10-02:
Hi @Zhang-Qinghui 

We believe that a foggy environment wouldn't be ideal for the L515 camera.  Anything that can block or refract the laser pattern that the L515 camera uses to determine depth will cause issues in determining depth.  We suggested reading the document so that you have an understanding of the technology behind the camera.  We also suggest you validate in the environment you plan on operating the camera in.  

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-10-19 by dorodnic