realsenseai/librealsense #7398: Build still failing with -DBUILD_EASYLOGGINGPP=OFF

dbolkensteyn, 2020-09-23:
The build still fails without easylogging on the current `dev` branch (7ebea7b2918f703d3da89b442d596d5372ed7041), even after #6304 got merged, because of this line:
https://github.com/IntelRealSense/librealsense/blame/development/src/l500/ac-trigger.cpp#L590

Error message: `cout` is not a member of `std`.

dbolkensteyn, 2020-09-23:
Same issue here: https://github.com/IntelRealSense/librealsense/blame/development/src/terminal-parser.cpp#L32-L34

With those lines commented out, I am able to build librealsense without easylogging.

RealSenseSupport, 2020-10-09:
Thank you so much for informing us. We will take a look.

RealSenseSupport, 2020-10-13:
Sorry this was closed by mistake. Reopen and link to PR#7554.

RealSenseSupport, 2020-10-23:
https://github.com/IntelRealSense/librealsense/pull/7554 approved and merged. 
Closing this ticket now.
Thank you.

closed 2020-10-23 by RealSenseSupport