realsenseai/librealsense #7401: Raspberry Pi 4: Segmentation faults upon errors such as timeouts

dbolkensteyn, 2020-09-23:
Context: While building an app for the Raspberry Pi 4 and D415 / D435 cameras, I noticed that the app was either working well (the vast majority of the times), or occasionally crashing with a segmentation fault within a few seconds after startup.

I've created a minimal reproducer which seems to indicate the the issue might be in the way librealsense handles its errors (I suspect the `HANDLE_EXCEPTIONS_AND_RETURN` macro).

I could reproduce the problem with:
* Multiple D415 / D435 cameras
* Multiple versions of librealsense (`v2.36.0`, `v2.38.1` as well as today's `master` and `dev` branches)
* All combinations of `-DFORCE_RSUSB_BACKEND` and `-DBUILD_EASYLOGGINGPP` (`ON`/`OFF`)
* Multiple camera firmware versions (latest `05.12.06.00` and `05.12.05.00`)

Note that I am using uclibc-ng and not glibc.

Reproducer 1, segfault occurs in `rs2_pipeline_wait_for_frames()` (`NULL` is passed instead of an `rs_error*` for brevity):
```
#include <stdio.h>
#include <stdlib.h>

#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>

int main(int argc, char* argv[]) {
    rs2_config* config = rs2_create_config(NULL);
    rs2_config_disable_all_streams(config, NULL);
    rs2_config_enable_stream(config, RS2_STREAM_DEPTH, 0, 1280, 720, RS2_FORMAT_Z16, 15, NULL);
    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, NULL);
    rs2_pipeline* pipeline = rs2_create_pipeline(ctx, NULL);
    rs2_pipeline_profile* profile = rs2_pipeline_start_with_config(pipeline, config, NULL);

    printf("Main loop starting...\n");
    while (1) {
        printf("Tick0!\n");
        rs2_frame* composite = rs2_pipeline_wait_for_frames(pipeline, 50, NULL);
        printf("Tick1!\n");

    }

    rs2_pipeline_stop(pipeline, NULL);

    rs2_delete_pipeline(pipeline);
    rs2_delete_context(ctx);
    rs2_delete_config(config);

    return 0;
}
```
which outputs:
```
Main loop starting...
Tick0!
Segmentation fault
```

Reproducer 2, segfault occurs in `rs2_pipeline_start()`:
```
#include <stdio.h>
#include <stdlib.h>

#include <librealsense2/rs.h>
#include <librealsense2/h/rs_pipeline.h>

int main(int argc, char* argv[]) {
    rs2_context* ctx = rs2_create_context(RS2_API_VERSION, NULL);
    rs2_pipeline* pipeline = rs2_create_pipeline(ctx, NULL);

    printf("Tick0\n");
    rs2_pipeline_profile* profile = rs2_pipeline_start(pipeline, NULL);
    printf("Tick1\n");

    rs2_pipeline_stop(pipeline, NULL);

    rs2_delete_pipeline(pipeline);
    rs2_delete_context(ctx);

    return 0;
}
```
which outputs:
```
Tick0
Segmentation fault
```

As you see, I lack proper debugging knowledge 😄 

Do you have any clue as to what might be the issue or how to proceed to further investigate this?`Thanks a lot!

MartyG-RealSense, 2020-09-23:
Hi @dbolkensteyn   Could you first try increasing the size of the **swapfile** on your Pi 4 please to create more virtual memory for the Pi to use if the physical memory somehow gets used up?

dbolkensteyn, 2020-09-24:
Thanks for the reply @MartyG-RealSense : The Raspberry Pi is only running this one process (no X terminal nor anything else). It has no configured swap, but has 4 GB of RAM out of which >3.7 GB are free according to `top`.

I dug a bit deeper here, and switched from my `printf()` debugging styles to a proper GDB instance, which seems to indicate that the segfault occurs in libstdc++ whenever librealsense throws an exception, such as the `throw std::runtime_error("No device connected");` in [`device_hub.cpp:145`](https://github.com/IntelRealSense/librealsense/blob/7ebea7b29/src/device_hub.cpp#L145).

I will do some more investigation, but I am now suspecting some libstdc++/compiler/compiler flags incompatibilities.

dbolkensteyn, 2020-09-24:
I've recompiled everything using glibc instead of uclibc-ng, and the issue vanished.

So this issue is most likely caused by some kind of incompatibility between uclibc-ng and libstdc++ which is in any case too advanced for me to understand or investigate further.

I'll simply stick to glibc where exception handling is working as expected. Closing this issue.

closed 2020-09-24 by dbolkensteyn

MartyG-RealSense, 2020-09-24:
Thanks so much for the update and sharing the details of what worked for you!