realsenseai/librealsense #7402: Very slow build libusb

Xander-Vreeswijk, 2020-09-23:
Not sure if this is the right place for this issue but when I start my program it takes very long to build, like 3.5 minutes. 
I get 1>Performing update step for 'libusb'
2>Performing configuration step for 'libusb'
3>Performing building step for 'libusb'
Does anyone have an idea what is going on and how I can speed up the building?

MartyG-RealSense, 2020-09-23:
Hi @Xander-Vreeswijk   If you are using CMake to build a program that you have created then instead of just using the term **make** to build it, you can use **make -jx** to utilize multiple cores in a computer's processor for the build process instead of just using a single core.  

'x' refers to the number of processor cores that you will make available for the build process. - for example, the librealsense documentation for building from source code suggests using **make -j8**.  With modern top-end CPUs with a large number of cores (e.g CPUs with core numbers ranging from 10 to 18) you could use a higher value, such as -j16.  The librealsense installation documentation points out though that using the -jx method can cause low-end computers to hang randomly.

Xander-Vreeswijk, 2020-09-23:
Do I need to use that in CMake or somewhere else?

MartyG-RealSense, 2020-09-23:
It is used in CMake.

Xander-Vreeswijk, 2020-09-23:
So you say I need to add entry **make -jx** as a bool and set it true?

MartyG-RealSense, 2020-09-23:
make -jx does not involve a bool.  It is just the standard **make** term with an additional **-jx** term placed next to it.  

The section of the librealsense instructions that discuss use of -jx is pasted below.

[image]