realsenseai/librealsense #7407: Facing some issues on install jit.realsense

faiahmed, 2020-09-24:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   |  | 
| Operating System & Version |   ( MacOS ) 
| Platform                                 | Max 8  |

### Issue Description

hello, i just got Realsense D415 and I'm facing some issues on install with Max 8.
I got this message "CMake Error: The source directory "/Users/Fai/build/path/to/jit.realsense" does not exist."
anyone can help me?

MartyG-RealSense, 2020-09-24:
Hi @faiahmed   I believe that you are referring to the **jit.realsense** program that provides RealSense compatibility for **Max**.  Is that corrrect, please?

https://github.com/blueyeti/jit.realsense
https://cycling74.com/products/max-features

The instructions give me the impression that the appropriate course of action is to download or clone the jit.realsense GitHub using the **Code** button and then go to the parent folder of the downloaded repository, create the **build** directory and use the **cmake path/to/jit.realsense -DCMAKE_BUILD_TYPE=Release** command to point CMake to the CMakeLists.txt file in the parent folder.  That is how I personally interpret the process (I have not encounted Max before).

[image]

[image]

faiahmed, 2020-09-27:
thank you your reply but i did everything you mentioned & still got the same issues.

MartyG-RealSense, 2020-09-27:
It is very hard to work out the correct instructions from the official documentation available.  The best I can do I think is link you to another case related to MacOS installation problems with jit.realsense and see if it makes sense to you.

https://github.com/blueyeti/jit.realsense/issues/2

There does not seem to be anywhere else to go if that link does not prove useful unfortunately.  I do apologize.

MartyG-RealSense, 2020-09-27:
It is very hard to work out the correct instructions from the official documentation available.  The best I can do I think is link you to another case related to MacOS installation problems with jit.realsense and see if it makes sense to you.

https://github.com/blueyeti/jit.realsense/issues/2

Unfortunately there does not seem to be anywhere else to research this particular question if that link does not lead to the solution.  I do apologize.

MartyG-RealSense, 2020-10-05:
Case closed due to no further comments received and no other solutions available.

closed 2020-10-05 by MartyG-RealSense