realsenseai/librealsense #7410: D455 realsense viewer missing frames when recording

gongyeted, 2020-09-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16  | 
| Kernel Version (Linux Only)    |  4.15.0                                         | 
| Platform                                 | PC |
| SDK Version                            |  librealsense-2.38.1 }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
D455. When using realsense-viewer record button and moving the camera around for a while with color camera and depth camera on, sometimes the viewer will be static for one second or two, which makes the data recorded inconsecutive, some frames missing. This situation happens like around every 30 seconds, and everytime it happens, I will lose dozens of frames consecutively. Please help me to fix this, thank you so much.

MartyG-RealSense, 2020-09-24:
Hi @gongyeted   I ran tests with a D455 with the record button in the RealSense Viewer but was unable to reproduce the frame drop that you experienced.    Does it only occur when recording two streams please, or does it happen when recording a single stream too?

gongyeted, 2020-09-24:
> Hi @gongyeted I ran tests with a D455 with the record button in the RealSense Viewer but was unable to reproduce the frame drop that you experienced. Does it only occur when recording two streams please, or does it happen when recording a single stream too?

I just tried again and when using single stream it didn't occur. However, it happened when using two streams again.

MartyG-RealSense, 2020-09-24:
You could eliminate whether it is related to a momentary drop in FPS speed by disabling the **Auto-Exposure Priority** setting in the **RGB** section of the Viewer's options side-panel.  Having auto-exposure enabled but Auto-Exposure Priority disabled forces the camera to try to maintain a constant FPS.

[image]

gongyeted, 2020-09-25:
> You could eliminate whether it is related to a momentary drop in FPS speed by disabling the **Auto-Exposure Priority** setting in the **RGB Camera** section of the Viewer's options side-panel. Having auto-exposure enabled but Auto-Exposure Priority disabled forces the camera to try to maintain a constant FPS.
> 
> [image]

I tried as you said disabling Auto-Exposure Priority, but the frames dropping occurred again.

MartyG-RealSense, 2020-09-25:
I will seek advice from Intel on this case.  Thank you very much for your patience.

MartyG-RealSense, 2020-09-25:
May I ask the following questions please:

1.  Are you able to try the camera on a different PC?

2.  Could you check whether the recorded bag file has the pauses, in order to confirm whether or not the frame drop is only occurring in the on-screen rendering of the image and not the recording of it.

You can play back the bag file by drag-and-dropping the file into the center panel of the Viewer, or by using the **Add Source** button at the top of the Viewer's options side-panel to load a recording.

gongyeted, 2020-09-29:
> May I ask the following questions please:
>
> 1. Are you able to try the camera on a different PC?
> 2. Could you check whether the recorded bag file has the pauses, in order to confirm whether or not [...]

1、This situation occurred on my laptop with ubuntu 1604. Then I tried on the same laptop with another windows 10 system, it didn't occur. Then I tried on another PC with ubuntu 1804, it didn't occur. I think it worked on my laptop on Windows system indicates that the USB connection of my computer isn't the cause of dropping frames. 
2、The bag file I recorded has the pauses too.
For now, I can use windows system to record files temporarily fullfilling my need. But the problem still there unsolved.

MartyG-RealSense, 2020-09-29:
@gongyeted   I will feed your answers back to Intel.  Thank you very much.

MartyG-RealSense, 2020-09-30:
Hi @gongyeted   We were unable to replicate the problem that you have been experiencing.  The recommendation is therefore to close this case.

closed 2020-10-08 by MartyG-RealSense