realsenseai/librealsense #7416: Cannot view RGB images, but depth works

snowface3, 2020-09-24:
------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |  Linaro | 
| Kernel Version (Linux Only)    | don't know (custom)                                       | 
| Platform                                 | HiKey RB3 (Snapdragon) |
| Segment			|  Robot                          | 

### Issue Description
I recently installed librealsense2 using the following steps:
` sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic main" -u`
` sudo apt-get install librealsense2-utils`
` sudo apt-get install librealsense2-dev`

When I open realsense-viewer, depth works fine but I am getting errors:
`synthetic-stream.cpp:48, exception was thrown during user processing callback!` and the screen display 'no frames found'.

Following is the dmesg -wT log:
`[Thu Sep 24 17:06:36 2020] usb 3-1: reset SuperSpeed Gen 1 USB device number 4 using xhci-hcd
[Thu Sep 24 17:07:40 2020] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[Thu Sep 24 17:07:46 2020] xhci-hcd xhci-hcd.0.auto: ERROR mismatched command completion event
[Thu Sep 24 17:07:48 2020] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[Thu Sep 24 17:07:48 2020] usb 3-1: device not accepting address 4, error -62
[Thu Sep 24 17:07:58 2020] xhci-hcd xhci-hcd.0.auto: ERROR mismatched command completion event
[Thu Sep 24 17:08:00 2020] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for stop endpoint command
[Thu Sep 24 17:08:11 2020] xhci-hcd xhci-hcd.0.auto: ERROR mismatched command completion event
[Thu Sep 24 17:08:13 2020] xhci-hcd xhci-hcd.0.auto: Timeout waiting for reset device command
[Thu Sep 24 17:08:23 2020] xhci-hcd xhci-hcd.0.auto: ERROR mismatched command completion event
[Thu Sep 24 17:08:25 2020] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[Thu Sep 24 17:08:36 2020] xhci-hcd xhci-hcd.0.auto: ERROR mismatched command completion event
[Thu Sep 24 17:08:38 2020] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for setup device command
[Thu Sep 24 17:08:38 2020] usb 3-1: device not accepting address 4, error -62
[Thu Sep 24 17:08:39 2020] usb 2-1: USB disconnect, device number 17
[Thu Sep 24 17:08:43 2020] usb 2-1: new full-speed USB device number 18 using xhci_hcd
[Thu Sep 24 17:08:43 2020] input: Logitech USB Receiver as /devices/platform/soc@0/1c00000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.0/0003:046D:C52B.0033/input/input86
[Thu Sep 24 17:08:43 2020] hid-generic 0003:046D:C52B.0033: input: USB HID v1.11 Keyboard [Logitech USB Receiver] on usb-0000:01:00.0-1/input0
[Thu Sep 24 17:08:43 2020] input: Logitech USB Receiver Mouse as /devices/platform/soc@0/1c00000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.1/0003:046D:C52B.0034/input/input87
[Thu Sep 24 17:08:43 2020] input: Logitech USB Receiver Consumer Control as /devices/platform/soc@0/1c00000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.1/0003:046D:C52B.0034/input/input88
[Thu Sep 24 17:08:43 2020] input: Logitech USB Receiver System Control as /devices/platform/soc@0/1c00000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/usb2/2-1/2-1:1.1/0003:046D:C52B.0034/input/input89
[Thu Sep 24 17:08:43 2020] hid-generic 0003:046D:C52B.0034: input: USB HID v1.11 Mouse [Logitech USB Receiver] on usb-0000:01:00.0-1/input1
[Thu Sep 24 17:08:43 2020] hid-generic 0003:046D:C52B.0035: device has no listeners, quitting
[Thu Sep 24 17:08:48 2020] xhci-hcd xhci-hcd.0.auto: ERROR mismatched command completion event
[Thu Sep 24 17:08:50 2020] xhci-hcd xhci-hcd.0.auto: Timeout while waiting for stop endpoint command
[Thu Sep 24 17:09:01 2020] xhci-hcd xhci-hcd.0.auto: ERROR mismatched command completion event
`
Following is the error when I run RGB in relsense-viewer:
[image]

Following is the warning when I run depth in relsense-viewer, but it still displays the depth:
[image]

Following is the settings info:
[image]

Following is another set of dmesg output:
[image]

Hopefully, you can provide some helpful tips. Thanks

MartyG-RealSense, 2020-09-25:
Hi @snowface3   The non-RealSense errors seem to be mainly related to wireless USB communication.  **ath10k** apparently refers to a wireless driver, whilst the log indicates that a Logitech USB receiver mouse and keyboard is being used.   If it is a Logitech Unifying receiver, the receiver is based on wifi-like 2.4 GHz radio signals rather than infrared.  So it is probably not being disrupted by the D435i's infrared.

The **RealSense Viewer** log error **control_transfer returned** can indicate a serious communication problem with the device if it is continuously generating instead of just appearing a few times at the start of streaming.  The **dmesg -wT** log contents also give me the impression of a serious USB communication problem occurring.

Do any of the problems appear in the log if the D435i camera is not plugged in, please?

snowface3, 2020-09-25:
Following is the dmesg log when I use wired keyboard/mouse and USB3 to plug in the RealSense camera. I have also tried using a powered external hub for the camera, no difference.
Also, `synthetic-stream.cpp:48,` exception was thrown during user processing callback!` error is continuous when enable RGB image.

What kind of USB communication problems/ how should I debug? Also, thank you for responding promptly.

[Fri Sep 25 12:25:50 2020] usb 3-1: new SuperSpeed Gen 1 USB device number 9 using xhci_hcd
[Fri Sep 25 12:25:50 2020] hid-generic 0003:8086:0B3A.0004: device has no listeners, quitting
[Fri Sep 25 12:25:50 2020] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[Fri Sep 25 12:25:50 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[Fri Sep 25 12:25:50 2020] uvcvideo 3-1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[Fri Sep 25 12:25:50 2020] uvcvideo 3-1:1.0: Entity type for entity Processing 2 was not initialized!
[Fri Sep 25 12:25:50 2020] uvcvideo 3-1:1.0: Entity type for entity Camera 1 was not initialized!
[Fri Sep 25 12:25:50 2020] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc@0/1c00000.pci/pci0000:00/0000:00:00.0/0000:01:00.0/usb3/3-1/3-1:1.0/input/input7
[Fri Sep 25 12:25:50 2020] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
[Fri Sep 25 12:25:50 2020] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
[Fri Sep 25 12:25:50 2020] uvcvideo 3-1:1.3: Entity type for entity Processing 7 was not initialized!
[Fri Sep 25 12:25:50 2020] uvcvideo 3-1:1.3: Entity type for entity Extension 8 was not initialized!
[Fri Sep 25 12:25:50 2020] uvcvideo 3-1:1.3: Entity type for entity Camera 6 was not initialized!
[Fri Sep 25 12:25:50 2020] usbcore: registered new interface driver uvcvideo
[Fri Sep 25 12:25:50 2020] USB Video Class driver (1.1.1)
[Fri Sep 25 12:25:51 2020] ath10k_snoc 18800000.wifi: received chan info event without a scan request, ignoring
[Fri Sep 25 12:25:52 2020] ath10k_snoc 18800000.wifi: received chan info event without a scan request, ignoring
[Fri Sep 25 12:26:50 2020] xhci_hcd 0000:01:00.0: Timeout while waiting for setup device command
[Fri Sep 25 12:26:54 2020] ath10k_snoc 18800000.wifi: received chan info event without a scan request, ignoring
[Fri Sep 25 12:26:55 2020] xhci_hcd 0000:01:00.0: Timeout while waiting for setup device command
[Fri Sep 25 12:26:55 2020] usb 3-1: device not accepting address 9, error -62
[Fri Sep 25 12:26:55 2020] ath10k_snoc 18800000.wifi: received chan info event without a scan request, ignoring
[Fri Sep 25 12:27:01 2020] xhci_hcd 0000:01:00.0: Timeout while waiting for setup device command
[Fri Sep 25 12:27:01 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:01 2020] usb 3-1: device not accepting address 9, error -71
[Fri Sep 25 12:27:01 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:01 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:02 2020] usb 3-1: device not accepting address 9, error -71
[Fri Sep 25 12:27:02 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:02 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:02 2020] usb 3-1: device not accepting address 9, error -71
[Fri Sep 25 12:27:02 2020] usb 3-1: USB disconnect, device number 9
[Fri Sep 25 12:27:03 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:03 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:03 2020] usb 3-1: device not accepting address 10, error -71
[Fri Sep 25 12:27:03 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:03 2020] usb 3-1: Device not responding to setup address.
[Fri Sep 25 12:27:04 2020] usb 3-1: device not accepting address 11, error -71
[Fri Sep 25 12:27:04 2020] usb usb3-port1: attempt power cycle

MartyG-RealSense, 2020-09-25:
Could you try re-applying the **udev** device handling rules please by inputting the commands below:

sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
sudo udevadm control --reload-rules && udevadm trigger

snowface3, 2020-09-25:
Hi, So I seem to already have the following conflicting file /lib/udev/rules.d/60-librealsense.....-rules.rules

MartyG-RealSense, 2020-09-25:
If you have multiple rules present, it is ok to delete the rule in the **etc** folder but deleting the other one can potentually break linkages to Debian packages.

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

snowface3, 2020-09-25:
@MartyG-RealSense can you confirm the following steps work for installation on ARM? Or is there another set of instructions I should follow for an ARM build?
`I recently installed librealsense2 using the following steps:

sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo **bionic** main" -u
sudo apt-get install librealsense2-utils
sudo apt-get install librealsense2-dev`

MartyG-RealSense, 2020-09-26:
Apologies for the delay in responding further.  Given that your OS is Linaro and the Linux package distribution method is aimed towards Ubuntu, I'm not sure that it is the best installation method for your particular device.  

The **RSUSB** method of building librealsense from source code may work better.   It requires an internet connection to perform the installation but it is not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/librealsense/issues/6845#issuecomment-659540316

closed 2020-10-12 by MartyG-RealSense