realsenseai/librealsense #7419: C++  read D435 Firmware Version

dingbaiyin, 2020-09-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (5.12.6 | 
| Operating System & Version |   {Ubuntu 16 | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | Raspberry Pi |
| SDK Version                            |  { 2.36.0 }                          | 
| Language                            |  C++                         | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<We have used a lot of IntelD4XX devices. Because of the different purchase batches, we now want to use C++ programs to read the Firmware Version. How to implement this function?>

MartyG-RealSense, 2020-09-25:
Hi @dingbaiyin   **RS2_CAMERA_INFO_FIRMWARE_VERSION** can provide the camera's currently installed firmware version.  

There are a couple of example scripts in the link below that demonstrate the printing on-screen of the value of RS2_CAMERA_INFO_FIRMWARE_VERSION with an **std::cout** instruction:

https://github.com/IntelRealSense/librealsense/issues/2219#issuecomment-412015600

https://community.intel.com/t5/Items-with-no-label/D435-streams-work-once-then-crash/m-p/499732/highlight/true#M6053

dingbaiyin, 2020-09-26:
I got it. Thank you.

dingbaiyin, 2020-09-26:
Thank you very much，I will close it

closed 2020-09-26 by dingbaiyin