realsenseai/librealsense #7421: SR300 IR capabilities

lghasemzadeh, 2020-09-25:
Hello intel IR specialists,

I was wondering if IR of SR300 can make sunglasses transparent.
I want the camera to transparent the sunglasses for me to be able to see the eyes. So I run both RGB and IR streams of camera at the same time to see how it works.

some of the sunglasses I tested became transparent but some of them didn't. I am confused.
I have attached two screenshots of two samples down.
IR stream at the left, RGB stream at the right.
In each screenshot same person with same sunglasses.
fırst: I want to know is SR300 IR stream can transparent all real (original) sunglasses.
second: how can you explain the difference between two screenshots? why in the first one's IR output the sunglasses is transparent but in the second one it isn't?  
@agrunnet 
[image]
[image]

RealSenseCustomerSupport, 2020-09-28:
----------------------------------------------

Hi @lghasemzadeh,
the SR300/SR305 feature a IR imager (camera sensor) and projector that operated @850 nm wavelength (IR light)
so to make it work properly, the target object need to be able to reflect the 850nm light to the receiver (IR imager)

roughly, we can classify the light into different category
(1) UV 100-400 nm
(2) Visible 400-800nm
(3) IR >800nm
basically, most sunglass is designed to block the UV light
it seems some of glass you have also attenuate the IR light  as well

lghasemzadeh, 2020-09-28:
So you mean that for the second case (second screenshot) that the sunglasses didn’t get transparent the reason is that it wasn’t successful to reflect the 850nm light to the receiver. If what I say is correct then what is the reason of that, why the second sunglasses wasn’t able to reflect the 850nm light? Is it because that sunglasses might be fake for example, or what is the exact scientific reason?

agrunnet, 2020-09-28:
@lghasemzadeh 
I think there are two possible explanations (as you are also indicating with your question 2):

A. Some sunglasses are linearly polarized and block light that way to reduce glare from the road - To test this, try to turn the SR305 camera 90 degrees (or the sunglasses) and see if that helps.  The SR305 uses polarized IR light. 

B. Sunglass manufacturers dont spec how they work at 850nm. They rely of dye in the plastic that absorbs light of certain colors. So it is entirely possible that some manufacturer chose a material that blocks IR light too, but we do not know.   

I would put my bet (and hope) on A first, and it is also very easy to test. 
Also if this is the problem, it would suggest two possible fixes: 1. You could rotate your camera, or 2. It is actually possible to buy a half-wave-plate material that can rotate the polarization of the laser. Our receiver does not care about the polarization, so the camera would continue to work as normal. 
  
If it is B, then there is very little we can do. One could try to see if 940nm works any better, as it would inform a future design decision, but unfortunately the SR305 has an 850nm bandpass filter and only sees that light, so there would not be an easy fix.

lghasemzadeh, 2020-09-29:
@agrunnet 
Actually my camera is SR300, not SR305 (I know they are vary similar in functions).

**First Problem:**
I tested the issue A and nothing changed, you can see the screenshots below.
The sunglasses that is not getting transparent:
normal angle
[image]
90 degree
[image]
The sunglasses that became transparent:
normal
[image]
90 degree
[image]
I also check all different angles not only 90, and nothing changed

1) did I rotate the camera correctly?

If it is B then I need to purchase a new camera which is capable of emitting 940nm.

2) But how can I be sure that new camera would work for ALL sunglasses and can make all types of sunglasses transparent? 

for my research I need a camera that has a technology which can make sunglasses transparent. 

3) Is there any ray which will be capable of making sunglasses transparent?

**Second Problem**
while I am recording videos with sunglasses there are two or more glares appear on the sunglasses' lenses. you can see the glare also in screenshots I attached.

4) how can emit them? 

Thank you

lghasemzadeh, 2020-09-30:
I am waiting for your response :)

lghasemzadeh, 2020-10-06:
@RealSenseCustomerSupport @agrunnet 
would you please answer my question?

agrunnet, 2020-10-06:
Your testing is sufficient to draw conclusions, but you wont like the answer. It seems like some sunglasses do NOT transmit 850nm.  It is unlikely that they will transmit in 940nm but that would be last test. Unfortunately, even if the sunglasses would be transparent in 940nm, it would not work with your SR300 camera, as it rejects all light outside of 850nm band. 
 
I am sorry to say that there does not appear to be anything we can do about this. 

Regarding glare, I believe this was asked in another post as well. As Sergey said, the only way to fix that is to somehow have a diffuse source illuminating the face, and ideally off at an angle.

RealSenseCustomerSupport, 2020-10-07:
----------------------------------------------

Dear @lghasemzadeh
could we close this ticket then use #7503 SR300 IR Glare for the remaining question?
thanks

lghasemzadeh, 2020-10-09:
@RealSense-Customer-Engineering @RealSenseCustomerSupport  @MartyG-RealSense 
Thank you.
In our company we need to purchase suitable camera for line production.
I tried to find a contact to arrange a meeting but I wasn't successful.
would you please contact me via my email?

MartyG-RealSense, 2020-10-09:
The above request has been replied to at https://github.com/IntelRealSense/librealsense/issues/7503#issuecomment-706117158

As requested by @RealSenseCustomerSupport  I will close this ticket so that the discussion can be focused at https://github.com/IntelRealSense/librealsense/issues/7503

closed 2020-10-09 by MartyG-RealSense