realsenseai/librealsense #7422: ir channel windows

dodler, 2020-09-25:
Hi,

I have realsense camera d415 with latest drivers on windows 10. I'm using python wrapper. 

I installed latest pyrealsense2, but it doesn't provide an ir channel. Also viewer from the sdk doesn't show ir channel.

However I can acquire ir channel in ubuntu 18 with the same pyrealsense2. 

Can I acquire ir channel in windows?

MartyG-RealSense, 2020-09-25:
Hi @dodler   Do you mean that you have selected an IR channel but it does not display, showing an error in the infrared panel such as "No frames received" please?

dodler, 2020-09-25:
sorry, i didn't clarified this, 
i'm using code similar to this one

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py

but I also add rs.stream.infrared
but it fails on 
pipeline.start(config)

with runtime error - cannot resolve request

MartyG-RealSense, 2020-09-25:
**Cannot resolve request** indicates that the SDK has been asked to do something that either it does not understand (e.g because of an error in the code) or because the requested mode is not available (e.g because the camera is currently being detected - or misdetected- as being on a USB 2 connection).

Does your IR stream definition statement look something like this:

**config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)**

'1' is put after rs.stream.infrared to inform the SDK that you are requesting the left IR stream.  If you were requesting both the left and right IR streams, the code would look something like this:

**config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)**
**config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)**

dodler, 2020-09-25:
actually i'm not specifying exact frame, i'm just using like this

config.enable_stream(rs.stream.infrared, 640,480,rs.format.y8,30)

I also tried your options, but still i'm getting the same error

MartyG-RealSense, 2020-09-25:
I looked again at **opencv_viewer_example.py** and wondered how closely your project was based on that script, because I noticed that it seems to wait for a pair of depth and color frames before it is allowed to proceed.  So i wonder if adding in an infrared stream would disrupt that checking process just after the pipeline starts.

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/opencv_viewer_example.py#L28

It may be worth testing the script in the link below that uses all three streams (depth, color, IR) and see whether the same problem with  **cannot resolve request** occurs:

https://github.com/IntelRealSense/librealsense/issues/6334

MartyG-RealSense, 2020-10-05:
Hi @dodler   Do you still require assistance with this case, please?  Thanks!

dodler, 2020-10-06:
Hi, problem is not still solved, I will test on another machine and report later
thanks

MartyG-RealSense, 2020-10-06:
Thanks very much for the update - I look forward to your report later about testing on another machine.

MartyG-RealSense, 2020-10-14:
Hi @dodler  Do you have an update please regarding testing your project on another machine?   Thanks!

dodler, 2020-10-20:
Hi, no updates yet, i'll close the issue

closed 2020-10-20 by dodler