realsenseai/librealsense #7423: DWS License Activation

siccscha, 2020-09-25:
We just downloaded the DWS Trial. The RealSenseActivationTool_1_8_0 seems to be broken. It seems like it never successfully is able to connect to the Sentinel server. Any idea how to obtain a trial license?

RealSenseSupport, 2020-09-30:
Sorry that you're running into this issue.  We will have a look at it to see what is going wrong.

RealSenseSupport, 2020-10-07:
Hi @siccscha 

We will need some information with regards to the camera serial number you are trying to activate as well as email address used for activation and potentially more.

In order to make sure we maintain your privacy, it would be better for this particular issue for you to submit a ticket to the following location: https://support.intelrealsense.com/hc/en-us/requests/new

This will provide us an avenue to discuss with you those aspects without the information being on this public github ticket.

RealSenseSupport, 2020-11-12:
Hi @siccscha 

I believe this has been taken care of.  Let us know if anything else is needed.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-12-01 by RealSenseSupport