realsenseai/librealsense #7424: RK3399 pyrealsense2 install "ImportError: No module named 'pyrealsense2'" + realsense-viewer run failure

Kosta404, 2020-09-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300   | 
| Firmware Version                   | 2.36 | 
| Operating System & Version |   Ubuntu aarch64 | 
| Kernel Version (Linux Only)    |    4.4.179     | 
| Platform                                 | RK3399  |
| SDK Version                            |  legacy / 2.36.0                         | 
| Language                            |  python                          | 
| Segment			|  others                           | 

### Issue Description
I was installing pyrealsense2 and realsense-viewer according to this guide https://github.com/acrobotic/Ai_Demos_RPi/wiki/Raspberry-Pi-4-and-Intel-RealSense-D435 .
1. But if I issue the comand 
_import pyrealsense2 as rs_  
I get Error ImportError: No module named 'pyrealsense2'

2. After running realsense-viewer i get this output into comand line:

libGL error: unable to load driver: rockchip_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: rockchip
libGL error: unable to load driver: rockchip_dri.so
libGL error: driver pointer missing
libGL error: failed to load driver: rockchip
Segmentation fault (core dumped)

MartyG-RealSense, 2020-09-25:
Hi @Kosta404   Are you using a board with a brand-name that has the RK3399 chipset on it please, such as an **Odroid** or an **Orange Pi** please?

Kosta404, 2020-09-25:
@MartyG-RealSense yes, but it is **FriendlyElec SOM-RK3399 Dev Kit**.

MartyG-RealSense, 2020-09-25:
Although the RK3399  uses an Arm processor like the Raspberry Pi, a Pi-specific installation guide such as Acrobotic may not be the ideal choice for a non-Pi board.   Given the unusual nature of your hardware and OS combination, a good first step may be to install librealsense and the Python wrapper together using a method called **RSUSB**.

This method requires an internet connection to perform the installation but is not dependent on Linux versions or kernel versions and does not require patching.  It can therefore be useful for use on devices that have problems with the usual installation process.

https://github.com/IntelRealSense/librealsense/issues/6964#issuecomment-665650908

Kosta404, 2020-09-25:
@MartyG-RealSense I`ll try and text you back on Monday, Don`t close this please! Thank You.

MartyG-RealSense, 2020-09-25:
It will still be open for at least the next 7 days, so don't worry.  Good luck!

Kosta404, 2020-09-25:
@MartyG-RealSense 
**cmake ../ -DFORCE_RSUSB_BACKEND=ON -DBUILD_PYTHON_BINDINGS:bool=true -DPYTHON_EXECUTABLE=...** 
I`m sorry to tell this, but this build comand didn`t finish successfully. I can send the last part of CMakeOutput.log

MartyG-RealSense, 2020-09-25:
Okay, please send you log.

Kosta404, 2020-09-25:
The system is: Linux - 4.4.179 - aarch64
Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: /usr/bin/c++ 
Build flags: 
Id flags: 

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.out"

The CXX compiler identification is GNU, found in "/home/pi/librealsense-2.36.0/build/CMakeFiles/3.5.1/CompilerIdCXX/a.out"

Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: /usr/bin/cc 
Build flags: 
Id flags: 

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.out"

The C compiler identification is GNU, found in "/home/pi/librealsense-2.36.0/build/CMakeFiles/3.5.1/CompilerIdC/a.out"

Determining if the CXX compiler works passed with the following output:
Change Dir: /home/pi/librealsense-2.36.0/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_c1ce7/fast"
/usr/bin/make -f CMakeFiles/cmTC_c1ce7.dir/build.make CMakeFiles/cmTC_c1ce7.dir/build
make[1]: Entering directory '/home/pi/librealsense-2.36.0/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_c1ce7.dir/testCXXCompiler.cxx.o
/usr/bin/c++      -o CMakeFiles/cmTC_c1ce7.dir/testCXXCompiler.cxx.o -c /home/pi/librealsense-2.36.0/build/CMakeFiles/CMakeTmp/testCXXCompiler.cxx
Linking CXX executable cmTC_c1ce7
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_c1ce7.dir/link.txt --verbose=1
/usr/bin/c++        CMakeFiles/cmTC_c1ce7.dir/testCXXCompiler.cxx.o  -o cmTC_c1ce7 -rdynamic 
make[1]: Leaving directory '/home/pi/librealsense-2.36.0/build/CMakeFiles/CMakeTmp'

Detecting CXX compiler ABI info compiled with the following output:
Change Dir: /home/pi/librealsense-2.36.0/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_fdef9/fast"
/usr/bin/make -f CMakeFiles/cmTC_fdef9.dir/build.make CMakeFiles/cmTC_fdef9.dir/build
make[1]: Entering directory '/home/pi/librealsense-2.36.0/build/CMakeFiles/CMakeTmp'
Building CXX object CMakeFiles/cmTC_fdef9.dir/CMakeCXXCompilerABI.cpp.o
/usr/bin/c++      -o CMakeFiles/cmTC_fdef9.dir/CMakeCXXCompilerABI.cpp.o -c /usr/share/cmake-3.5/Modules/CMakeCXXCompilerABI.cpp
Linking CXX executable cmTC_fdef9
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_fdef9.dir/link.txt --verbose=1
/usr/bin/c++       -v CMakeFiles/cmTC_fdef9.dir/CMakeCXXCompilerABI.cpp.o  -o cmTC_fdef9 -rdynamic  
Using built-in specs.
COLLECT_GCC=/usr/bin/c++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/5/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.12' --with-bugurl=file:///usr/share/doc/gcc-5/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-5 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-5-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-5-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-5-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --enable-multilib --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --disable-werror --enable-multilib --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 5.4.0 20160609 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.12) 
COMPILER_PATH=/usr/lib/gcc/arm-linux-gnueabihf/5/:/usr/lib/gcc/arm-linux-gnueabihf/5/:/usr/lib/gcc/arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/5/:/usr/lib/gcc/arm-linux-gnueabihf/
LIBRARY_PATH=/usr/lib/gcc/arm-linux-gnueabihf/5/:/usr/lib/gcc/arm-linux-gnueabihf/5/../../../arm-linux-gnueabihf/:/usr/lib/gcc/arm-linux-gnueabihf/5/../../../../lib/:/lib/arm-linux-gnueabihf/:/lib/../lib/:/usr/lib/arm-linux-gnueabihf/:/usr/lib/../lib/:/usr/lib/gcc/arm-linux-gnueabihf/5/../../../:/lib/:/usr/lib/
COLLECT_GCC_OPTIONS='-v' '-o' 'cmTC_fdef9' '-rdynamic' '-shared-libgcc' '-march=armv7-a' '-mfloat-abi=hard' '-mfpu=vfpv3-d16' '-mthumb' '-mtls-dialect=gnu'
 /usr/lib/gcc/arm-linux-gnueabihf/5/collect2 -plugin /usr/lib/gcc/arm-linux-gnueabihf/5/liblto_plugin.so -plugin-opt=/usr/lib/gcc/arm-linux-gnueabihf/5/lto-wrapper -plugin-opt=-fresolution=/tmp/ccaoLi1w.res -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc -plugin-opt=-pass-through=-lc -plugin-opt=-pass-through=-lgcc_s -plugin-opt=-pass-through=-lgcc --sysroot=/ --build-id --eh-frame-hdr -export-dynamic -dynamic-linker /lib/ld-linux-armhf.so.3 -X --hash-style=gnu --as-needed -m armelf_linux_eabi -z relro -o cmTC_fdef9 /usr/lib/gcc/arm-linux-gnueabihf/5/../../../arm-linux-gnueabihf/crt1.o /usr/lib/gcc/arm-linux-gnueabihf/5/../../../arm-linux-gnueabihf/crti.o /usr/lib/gcc/arm-linux-gnueabihf/5/crtbegin.o -L/usr/lib/gcc/arm-linux-gnueabihf/5 -L/usr/lib/gcc/arm-linux-gnueabihf/5/../../../arm-linux-gnueabihf -L/usr/lib/gcc/arm-linux-gnueabihf/5/../../../../lib -L/lib/arm-linux-gnueabihf -L/lib/../lib -L [... cut ...]
/usr/bin/c++     -DCOMPILER_SUPPORTS_CXX0X   -std=c++0x -o CMakeFiles/cmTC_6a42c.dir/src.cxx.o -c /home/pi/librealsense-2.36.0/build/CMakeFiles/CMakeTmp/src.cxx
Linking CXX executable cmTC_6a42c
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_6a42c.dir/link.txt --verbose=1
/usr/bin/c++    -DCOMPILER_SUPPORTS_CXX0X    CMakeFiles/cmTC_6a42c.dir/src.cxx.o  -o cmTC_6a42c -rdynamic 
make[1]: Leaving directory '/home/pi/librealsense-2.36.0/build/CMakeFiles/CMakeTmp'

Source file was:
int main() { return 0; }

Kosta404, 2020-09-25:
@MartyG-RealSense The output is above

MartyG-RealSense, 2020-09-25:
I didn't learn much from the CMake log unfortunately.  I was thinking of the on-screen build log that goes through the % stages of completion as librealsense is built.  

Let's forget about the Python wrapper for the moment.  Assuming that you have an active internet connection, could you try the CMake instruction below that builds librealsense with the RSUSB method without the Python wrapper please:

**cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true**

Kosta404, 2020-09-28:
@MartyG-RealSense the command executed successfully, how many cores should i put in **make -j**?

Kosta404, 2020-09-28:
@MartyG-RealSense the build and install was successful, but there`s still **"No module name pyrealsense2"**

MartyG-RealSense, 2020-09-28:
The librealsense documentation recommends **-j8**.  

The link below has a discussion about the calculating of j-values.

https://stackoverflow.com/questions/15289250/make-j4-or-j8

Kosta404, 2020-09-28:
@MartyG-RealSense  the build and install was successful, but there`s still "No module name pyrealsense2"

MartyG-RealSense, 2020-09-28:
After the librealsense build without pyrealsense2 had completed successfully, what method did you use to try to build the Python bindings please?

Kosta404, 2020-09-28:
> I didn't learn much from the CMake log unfortunately. I was thinking of the on-screen build log that goes through the % stages of completion as librealsense is built.
>
> Let's forget about the Python [...]

this one

MartyG-RealSense, 2020-09-28:
The quoted command does not include the instruction to build the Python bindings, as we took the Python command out earlier in this discussion to try to get librealsense working correctly first before adding Python.

Once librealsense has been installed successfully once, can you try going to the **build** directory of your librealsense source folder and using this short CMake command:

**cmake ../ - DBUILD_PYTHON_BINDINGS:bool=true**

Kosta404, 2020-09-28:
cmake ../ -DFORCE_RSUSB_BACKEND=true -DCMAKE_BUILD_TYPE=release -DBUILD_EXAMPLES=true -DBUILD_GRAPHICAL_EXAMPLES=true

cmake ../ - DBUILD_PYTHON_BINDINGS:bool=true

Do I have to run them both from build?

MartyG-RealSense, 2020-09-28:
Just do this one from **build** please:

cmake ../ - DBUILD_PYTHON_BINDINGS:bool=true

Kosta404, 2020-09-28:
@MartyG-RealSense I did what you suggested. Failure again, unfortunately. Is there something more to try?

MartyG-RealSense, 2020-09-28:
cmake ../ -DBUILD_PYTHON_BINDINGS=true
make -j8

Kosta404, 2020-09-28:
> cmake ../ -DBUILD_PYTHON_BINDINGS=true
> make -j8

@MartyG-RealSense that`s what i did and then "sudo make install"

MartyG-RealSense, 2020-09-28:
Thank you very much for your patience.  What version of Python is your computer using please? (e.g Python 3.7, Python 3.8, etc)

Kosta404, 2020-09-28:
@MartyG-RealSense Python 3.5.2 installed by default

MartyG-RealSense, 2020-09-28:
There was a case where somebody using this Python version successfully built the wrapper using the command below:

cmake ../ -DBUILD_PYTHON_BINDINGS=bool:true -DPYTHON_EXECUTABLE=/usr/bin/python3

Kosta404, 2020-09-28:
@MartyG-RealSense ok, I`ll try one more time than and reply shortly or tomorrow

Kosta404, 2020-09-28:
@MartyG-RealSense didn`t work again. Maybe you can suggest some other OS version for this type of board (**RK3399**)?

MartyG-RealSense, 2020-09-28:
Most people who are using Python 3.5.2 with librealsense seem to be using the OS **Ubuntu 16.04**

Kosta404, 2020-09-29:
@MartyG-RealSense along with the **RK3399** board?

MartyG-RealSense, 2020-09-29:
No, nearly all the cases in my search results were using the **Intel NUC** mini-PC with Python 3.5.2.  These are very small PCs available in cased form or as board kits.

Kosta404, 2020-09-29:
@MartyG-RealSense But my employer wants it running on **RK3399**, is there some system you can reccomend? Or it`s useless?

MartyG-RealSense, 2020-09-29:
I believe it should be possible but the Rockchip nature of the board may mean that kernel patching will be involved.  I will consult with Intel about this.  Thanks again for your patience.

Kosta404, 2020-09-29:
@MartyG-RealSense Ok, will wait for your reply.

RealSenseSupport, 2020-10-09:
@Kosta404 Could you please check if you follow the steps in https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python#building-from-source ? Especially the following steps.
3. Run the top level CMake command with the following additional flag 
4. update your PYTHONPATH environment variable to add the path to the pyrealsense library

RealSenseSupport, 2020-10-15:
@Kosta404 Any update from your side? Thanks!

RealSenseSupport, 2020-10-21:
@Kosta404 Could you please update if you still need support on this? Thanks!

RealSenseSupport, 2020-11-05:
@Kosta404 Sorry that we didn't get update from you for weeks. Will close it at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-11-05 by RealSenseSupport