realsenseai/librealsense #7425: Unity3D not working with driver version 6.1.160.21

Vangos, 2020-09-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.07.100 | 
| Operating System & Version |  Windows 10 | 
| Kernel Version (Linux Only)    |  -                                         | 
| Platform                                 | PC |
| SDK Version                            |  SDK 2.0 (v2.38.1) }                          | 
| Language                            |  Unity3D                          | 
| Segment			|  Desktop application                         | 

### Issue Description

I am using a RealSense D400 series camera (D435) on Unity3D, with the latest RealSense SDK version (2.38.1). 

When running with driver version **6.1.160.14**, everything works fine:

[image]

When running with driver version **6.1.160.21**, Unity throws an exception:

[image]

```
ExternalException: rs2_pipeline_start_with_config(...)
Rethrow as Exception: Couldn't resolve requests
```

I'm using the official Unity package and I have not modified the C# code. I'm running the demos as-is.

Notes:
- The issue can be replicated on various Windows 10 computers.
- I'm rebooting the PC whenever I switch drivers.

MartyG-RealSense, 2020-09-25:
Hi @Vangos   The RealSense UWP (Universal Windows Platform) driver is not required to use the 400 Series camera with Windows.  It is intended for the purpose of providing compatibility with products that support interfacing through the UWP platform, such as the **Microsoft Hololens** headset.  If you are not attempting to interface the RealSense camera with such a product then the UWP driver does not need to be installed.

Vangos, 2020-09-25:
Thank you very much for your quick reply!

closed 2020-10-05 by MartyG-RealSense