realsenseai/librealsense #7426: How to uninstall RealSense SDK 2.0

milan-r-shah, 2020-09-25:
| Required Info                 |                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                    | 
| Firmware Version                   | 05.12.06.00          | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)   | 5.3.0-28-generic     | 
| Platform                                 | PC                           |
| SDK Version                          |  2.0 (v2.36.0)           | 
| Language                              |  C++                         | 
| Segment			        |  others                      | 

### Issue Description
I had installed RealSense SDK 2.0 using **building-from-source** method by following [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). However, as I have been facing an issue #7313 I want to update the current SDK v2.36.0 to the latest one v2.38.1. So, for that, first, I want to uninstall the current SDK.

I found steps for uninstalling the SDK in [distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) but those are for the pre-built packages. I couldn't find similar steps for **building-from-source** installation method.

MartyG-RealSense, 2020-09-26:
Hi @milan-r-shah   Could you go to the **build** folder of your source-code installation of librealsense and run the instruction below, please:

**sudo make uninstall** 

https://github.com/IntelRealSense/librealsense/issues/5773#issuecomment-581477785

milan-r-shah, 2020-09-28:
Hi @MartyG-RealSense ,

Thanks for your reply. I was also thinking about this command but got confused because in [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md), before building the SDK, we are also running some scripts e.g. ` ./scripts/patch-realsense-ubuntu-lts.sh` before running `sudo make install`. So, just to confirm, running only `sudo make uninstall` command would be enough?

Also, the current kernel in my Ubuntu 18.04 LTS is **5.3.0-28-generic** and I couldn't find any corresponding *script* in the `scripts` directory. So, which script should I run if I want to reinstall the SDK?

MartyG-RealSense, 2020-09-28:
My understanding is that the **sudo make uninstall** performs the uninstall, and the **&& make clean** instruction that appears in the librealsense build instructions after that is for the purpose of cleaning up old files.  So doing **sudo make uninstall && make clean** would likely be a good idea as an uninstall method.

I would think that if you were using Ubuntu 18.04 then according to the installation instructions for building the SDK from source, you would use this LTS script, even though you have a **generic:** kernel:

**./scripts/patch-realsense-ubuntu-lts.sh**

milan-r-shah, 2020-09-29:
Hi @MartyG-RealSense ,
Thanks a lot for the clarification. I appreciate your help and time.

One more thing, after uninstalling the SDK, if I decide to install the new version using **backend** method then I won't have to run the script you mentioned above, right? I have to just follow the steps you mentioned in https://github.com/IntelRealSense/librealsense/issues/6940#issuecomment-665713929, isn't it?

It would be really helpful if the RealSense team can add these instructions in [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md).

MartyG-RealSense, 2020-09-29:
Yes, once you have cleaned up, you should be able to perform the **backend** installation method with the instructions in that link.

Though the RSUSB backend method has advantages (especially if you have problems with installing using the normal instructions), it also has some disadvantages compared to kernel patching that mean that it may not be suitable for use in commercial products.    Building with DKMS packages (the **distribution** installation method) therefore has the strongest official recommendation.

The advantages and disadvantages are compared in the link below.  Scroll down to the section of the linked-to comment that is headed with **What are the advantages and disadvantages of using libuvc vs patched kernel modules?**

https://github.com/IntelRealSense/librealsense/issues/5212#issuecomment-552184604

milan-r-shah, 2020-09-29:
Thanks again @MartyG-RealSense for a detailed response as well as sharing important links. Also sorry, but now I have a few more confusions/queries:

1.

> Though the RSUSB backend method has advantages (especially if you have problems with installing using the normal instructions), it also has some disadvantages compared to kernel patching that mean [...]

So here, by kernel patching or DKMS packages or Distribution installation method, you meant the **building-from-source** method, right? And that method does the kernel patching, right? On the other hand, **backend** method does not, correct?

2. In https://github.com/IntelRealSense/librealsense/issues/6940#issuecomment-665713929, you have mentioned that 

> There is another installation method that I refer to as backend, and it is performed over an internet connection. 

This **backend** method looks almost as same as **building-from-source** method, the only difference is `-DFORCE_RSUSB_BACKEND=true` flag in `cmake` command. So, this flag will create some make files that would ask `make` to install the required dependencies while building? On the other hand, in building-from-source method, dependencies would be handled through kernel-patching?

MartyG-RealSense, 2020-09-29:
1.  The two main installation methods recommended in the official librealsense documentation are:

**Distribution (building librealsense from pre-built packages)**
https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md

**Manual (building from source code)**
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

The **RSUSB backend** method belongs to the source-code category.  
https://github.com/IntelRealSense/librealsense/issues/6940#issuecomment-665713929

A key difference between the Manual source-code method in the official documentation and the RSUSB process is that with RSUSB, dependencies are installed over an internet connection and so the build does not need to be patched.  It therefore is not dependent on Linux versions or kernel versions and should be able to work with any flavor of Linux.  As the advantage / disadvantage comparison makes clear though, it has various drawbacks compared to a patch-based build.

2.  Basically yes, leaving out the **DFORCE_RSUSB_BACKEND=true** flag will mean that it is a manual build that requires kernel patching.

MartyG-RealSense, 2020-10-17:
Case closed due to no further comments received.

closed 2020-10-17 by MartyG-RealSense

milan-r-shah, 2020-10-29:
Hi @MartyG-RealSense,
Sincere apology for not responding soon but I was quite occupied with some other tasks. Thank you so much for addressing my issue. 

Just wanted to add/request if you or RealSense team can add the instruction for Uninstalling & Upgrading the SDK (built from source) on the instruction page on installing the SDK by building from the source i.e. [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) then it would great!

Thanks again :)

MartyG-RealSense, 2020-10-30:
I would speculate that the source-code uninstall instruction (uninstall old version, clean up old files, install new version) is the way that the developers want it to be done, otherwise there would likely be a dedicated **Uninstall** heading like there is on the packages version of the instructions.

**sudo make uninstall && make clean && make && sudo make install**

HartmannSa, 2021-02-04:
Hi,
I also needed to uninstall Realsense SDK 2 (which was build from source, but one time also with apt). 
I followed all your suggestions and used 
`sudo make uninstall && make clean` and 
`dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge`

However, `sudo updatedb` followed by `locate librealsense2.so` returns:
```
/usr/local/lib/librealsense2.so.2.40
/usr/local/lib/librealsense2.so.2.40.0
```

Can/ Should I just remove/delete this files?
Do I need to pay attention on something else?
Thanks!

MartyG-RealSense, 2021-02-04:
Hi @HartmannSa   In the link below, a RealSense team member advises not to have multiple librealsense **.so** files installed.

https://github.com/IntelRealSense/librealsense/issues/4410#issuecomment-510509469

HartmannSa, 2021-02-05:
Hi, thanks for the advise.
I was already thinking that it's not right to have several of them. 
I have now manually deleted all librealsense2 files via `sudo rm librealsense2*` that I could locate (after the other uninstall steps didn't delete those files on their own).
But I wonder if there are any other files/ folders/ ... on my computer that I also need to uninstall manually?
Indeed, the reinstallation of librealsense throws errors...

MartyG-RealSense, 2021-02-05:
Since you have apparently installed both source code and package versions of librealsense, do the errors that you are experiencing include **Multiple udev-rules were found**?

If so, the link below provides advice about deleting the rule in the **/etc** folder to remove this error.

https://github.com/IntelRealSense/librealsense/issues/6153#issuecomment-605687547

HartmannSa, 2021-02-05:
Thanks for the tip. 
I was now able to build/install everything without errors using 
`sudo make uninstall && make clean && make -j8 && make install`.
But my error had nothing to do with the udev-rules. 
By the way, after my installation, I can't find a _librealsense2-udev-rules.rule_ either under _/lib/udev/rules.d/_ or under _etc/udecv_. Is this correct?

However, I get the following warnings when starting the realsense-viewer: 
```
 05/02 12:03:43,797 WARNING [140316942813056] (backend-v4l2.cpp:1357) Pixel format 36315752-1a66-a242-9065-d01814a likely requires patch for fourcc code RW16!
 05/02 12:03:54,454 WARNING [140316598310656] (hdr-merge.cpp:38) HDR Merge filter cannot process frames because relevant metadata params are missing
 05/02 12:03:58,119 WARNING [140316598310656] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 05/02 12:03:58,119 WARNING [140316598310656] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 05/02 12:03:58,119 WARNING [140316598310656] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
 05/02 12:03:58,119 WARNING [140316598310656] (backend-v4l2.cpp:1775) Invalid md size: bytes used =  0 ,start offset=10
```
I found a similar issue #4971, however I use the following system (so not Kernel 4.16):
```
:~$ uname -r
5.4.0-65-generic

:~$ modinfo uvcvideo | grep version
version:        1.1.1
srcversion:     32217DB1DE9B0BC725EDF5D

:~$ rs-fw-update -l
1) Name: Intel RealSense D435, firmware version: 05.12.10.00

:~$ apt list librealsense2
librealsense2/bionic 2.41.0-0~realsense0.3962 amd64

:~$ lsb_release -a
LSB Version:	core-9.20170808ubuntu1-noarch:security-9.20170808ubuntu1-noarch
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic
```