realsenseai/librealsense #7427: How to upgrade RealSense SDK 2.0 (which was installed "building-from-source" method)

milan-r-shah, 2020-09-25:
| Required Info                 |                                                    |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i                    | 
| Firmware Version                   | 05.12.06.00          | 
| Operating System & Version |   Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)   | 5.3.0-28-generic     | 
| Platform                                 | PC                           |
| SDK Version                          |  2.0 (v2.36.0)           | 
| Language                              |  C++                         | 
| Segment			        |  others                      | 

### Issue Description
I had installed RealSense SDK 2.0 using **building-from-source** method by following [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md). However, as I have been facing an issue #7313 I want to update the current SDK v2.36.0 to the latest one v2.38.1. So, how can I directly upgrade this SDK to the latest version without uninstalling it?

I found steps for upgrading the SDK in [distribution_linux.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md) but those are for the pre-built packages. I couldn't find similar steps for **building-from-source** installation method.

MartyG-RealSense, 2020-09-26:
Hi @milan-r-shah   I believe that upgrading on the **distribution** method is practical because it is compartmentalized into a very small number of packages.  Whereas overwriting just some of the files on a source-code build with newer versions may lead to problems.  

The standard procedure with the source code method of installation is, of course, to do an uninstall, clean and install with **sudo make uninstall && make clean && make && sudo make install**

If you are seeking to upgrade an existing version because of concern about the time that it takes to build the entire SDK, you can speed up the build process by using multiple cores of your computer's processor.  

This is done by adding a **jx** term to the **make** command, with 'x' representing the number of processor cores that can be used.  librealsense recommends **-j8**, though more powerful high-end processors may be able to make use of higher values such as **-j16**.  Bear in mind that on low-end devices, using -jx may cause the build to randomly hanf.

Googling for **how many cores do i have** provides information about how to determine how many cores your particular computer has.

Below is the -j8 version of the build instruction from the librealsense source-code install documentation:

**sudo make uninstall && make clean && make **-j8** && sudo make install**

MartyG-RealSense, 2020-10-12:
Case closed due to no further comments received.

closed 2020-10-12 by MartyG-RealSense

milan-r-shah, 2020-10-29:
Hi @MartyG-RealSense,
Sincere apology for not responding soon but I was quite occupied with some other tasks. Thank you so much for addressing my issue. 

Just wanted to add/request if you or RealSense team can add the instruction for Uninstalling & Upgrading the SDK (built from source) on the instruction page on installing the SDK by building from the source i.e. [installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md) then it would great!

Thanks again :)