realsenseai/librealsense #7429: get_depth_frame() problem

m10507323, 2020-09-26:
Hi, I build librealsense in arm base system and try to run demo python program, here is my program.

```python
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
pipeline.start()
try:
    while True:
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        depth_data = depth.as_frame().get_data()
        np_image = np.asanyarray(depth_data)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(np_image, alpha=0.03), cv2.COLORMAP_JET)
        cv2.imshow('depth_colormap', depth_colormap)
        cv2.waitKey(1)
finally:
    pipeline.stop()
```

But I got error message like this:

Traceback (most recent call last):
  File "realsense_back.py", line 10, in <module>
    depth = frames.get_depth_frame()
TypeError: get_depth_frame(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.composite_frame, arg0: int) -> pyrealsense2.depth_frame

Invoked with: <pyrealsense2.frameset Z16 BGR8 #1>

Is anyone know how to solve it?
Any suggestion with be appreciated.

MartyG-RealSense, 2020-09-26:
Could you try inserting this line directly below **depth = frames.get_depth_frame()** please:

**if not depth: continue**

So that it looks like this:

`try:`
`while True:`
`frames = pipeline.wait_for_frames()`
`depth = frames.get_depth_frame()`
`if not depth: continue`
`colorizer = rs.colorizer()`

ev-mp, 2020-09-27:
@m10507323 , it seem that you're building from source from `development` and not the `master` branch.

If that's the case you'd need to pass in the stream index 0 so it would look
```python
depth = frames.get_depth_frame(0)
```
Note that  `development`  is used for experiments and in this case the change will be reverted.

m10507323, 2020-09-28:
@MartyG-RealSense , @ev-mp 
Hi, sorry for my late reply, 
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md
I follow this toturial to build librealsense, and I've already try both of your suggestion, here is my update code.

'''python
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)
profile = pipeline.start(config)
try:
    while True:
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame(0)
        if not depth: 
            continue
        colorizer = rs.colorizer()
        colorized_depth = np.asanyarray(colorizer.colorize(depth).get_data())
        colorized_depth_rgb = cv2.cvtColor(colorized_depth, cv2.COLOR_BGR2RGB)
        cv2.imshow('depth_colormap_rgb', colorized_depth_rgb)
        cv2.waitKey(1)
finally:
    pipeline.stop()
'''
But I got new error:
'''python
No protocol specified
(depth_colormap_rgb:17537): Gtk-WARNING **: cannot open display: :0.0
'''
In this case, should I change other setting or install any package?
Thanks.

m10507323, 2020-09-28:
I find out that the problem isn't come from realsense but from Xserver.
I'll try other way to solve it, thanks for your help.

MartyG-RealSense, 2020-09-28:
Thanks very much for the update @m10507323  - good luck!

closed 2020-09-30 by m10507323