realsenseai/librealsense #7430: Error while installing librealsense

RayhanHaqi, 2020-09-26:
I was trying to study VIO (Visual Inertial Odometry) on ROS. And when I was installing librealsense using vcpkg from https://github.com/IntelRealSense/librealsense, I got this…

~/vcpkg$ ./vcpkg integrate install
Applied user-wide integration for this vcpkg root.
CMake projects should use: “-DCMAKE_TOOLCHAIN_FILE=/home/rayhan/vcpkg/scripts/buildsystems/vcpkg.cmake”

How do I solve this?

MartyG-RealSense, 2020-09-26:
Hi @RayhanHaqi   I have not seen this error reported on the RealSense GitHub forum before, though an occurrence of it relating to use of librealsense was posted on the vcpkg forum in June 2020.

https://github.com/microsoft/vcpkg/issues/12030

It looks as though the vcpkg forum support member's advice was to manually apply kernel patches.

https://github.com/microsoft/vcpkg/issues/12030#issuecomment-663360973

MartyG-RealSense, 2020-10-05:
Hi @RayhanHaqi   Do you still require assistance with this case, please?   Thanks!

RayhanHaqi, 2020-10-05:
No. I already installed it but not with vcpkg. Thanks.

MartyG-RealSense, 2020-10-05:
Thanks so much for the update!  I will close the case as you do not require further assistance.   Thanks again.

closed 2020-10-05 by MartyG-RealSense